tinygo-org/tinygo #1756: cmd.exe and bash.exe have different build results on Windows

sago35, 2021-03-29:
bash: (GNU bash, version 4.4.23(1)-release-(x86_64-pc-msys) + git version 2.30.2.windows.1)
```
sago3@B450GT3 MINGW64 /c/tinygo/tinygo (dev)
$ tinygo version
tinygo version 0.18.0-dev-90b42799 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)

sago3@B450GT3 MINGW64 /c/tinygo/tinygo (dev)
$ c:/tinygo/tinygo/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  69100   11336    2324 |   80436   13660

sago3@B450GT3 MINGW64 /c/tinygo/tinygo (dev)
$ /c/tinygo/tinygo/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  69100   11336    2324 |   80436   13660
```

cmd.exe:
```
C:\tinygo\tinygo>tinygo version
tinygo version 0.18.0-dev-90b42799 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)

C:\tinygo\tinygo>c:/tinygo/tinygo/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  88604   11760    2432 |  100364   14192
```

The difference in `make smoketest` is in the following commands.

```
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=arduino              examples/blinky1
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=hifive1b             examples/blinky1
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=reelboard            examples/blinky1
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=pca10040             examples/blinky2
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=pca10056             examples/blinky2
c:/tinygo/tinygo/build/tinygo build             -o test.wasm -tags=circuitplay_express  examples/blinky1
c:/tinygo/tinygo/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
```

sago35, 2021-03-29:
Please let me know if there is anything I should do in order to find out.

sago35, 2021-03-29:
On ubuntu20.04, the results did not match either of the windows results.

```
sago35@M75q2  ~/tinygo/tinygo (dev)
$ tinygo version
tinygo version 0.18.0-dev-8d93b9e linux/amd64 (using go version go1.15.10 and LLVM version 11.0.0)

sago35@M75q2  ~/tinygo/tinygo (dev)
$ tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  87676   11692    2432 |   99368   14124
```

aykevl, 2021-03-29:
You can check the `env` subcommand, some environment variables might be different between the shells.

Also, I see that the `version` command does not use the correct command. For example, here:

```
$ tinygo version
tinygo version 0.18.0-dev-90b42799 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)

$ c:/tinygo/tinygo/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
```

In the first you are running `tinygo`, in the second you are running `c:/tinygo/tinygo/build/tinygo`. These might be different. Try running `c:/tinygo/tinygo/build/tinygo version`, for example.

aykevl, 2021-03-29:
> On ubuntu20.04, the results did not match either of the windows results.

You are using a different Go version there: go1.15.10 instead of go1.16.

sago35, 2021-03-30:
It seems that my environment was not good. I tried again, and the files matched.

```
C:\tinygo\tinygo>C:\tinygo\tinygo\build\tinygo.exe version
tinygo version 0.18.0-dev-90b42799 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)

C:\tinygo\tinygo>C:\tinygo\tinygo\build\tinygo.exe build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  69100   11336    2324 |   80436   13660

C:\tinygo\tinygo>C:\tinygo\tinygo\build\tinygo.exe env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Program Files\\Go"
GOPATH="C:\\Users\\sago3\\dev"
GOCACHE="C:\\Users\\sago3\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\tinygo\\tinygo"
```

```
sago3@B450GT3 MINGW64 /c/tinygo/tinygo ((90b42799...))
$ C:/tinygo/tinyto/build/tinygo build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
bash: C:/tinygo/tinyto/build/tinygo: No such file or directory

sago3@B450GT3 MINGW64 /c/tinygo/tinygo ((90b42799...))
$ /c/tinygo/tinygo/build/tinygo.exe build -size short -o test.hex -target=pca10040 -opt=0     ./testdata/stdlib.go
   code    data     bss |   flash     ram
  69100   11336    2324 |   80436   13660

sago3@B450GT3 MINGW64 /c/tinygo/tinygo ((90b42799...))
$ /c/tinygo/tinygo/build/tinygo.exe env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Program Files\\Go"
GOPATH="C:\\Users\\sago3\\dev"
GOCACHE="C:\\Users\\sago3\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\tinygo\\tinygo"
```

closed 2021-03-30 by sago35