tinygo-org/tinygo #1757: Cannot flash pca10059 in windows environment

sago35, 2021-03-29:
windows does not use `/bin/sh`, so the flash-command `&&` in pca10059.json is not processed.
The following are some ideas for solutions, but basically, I think 1 is better.

1. Set `flash-command` to `[]string`
2. Split commands with `&&`
3. Handle it with `sh.exe`
    * However, `sh.exe` is not available for windows, so it will need to be installed separately.
    * Also, in order to be handled by `sh.exe`, `{hex}` needs to be converted to a form that can be handled by sh

https://github.com/tinygo-org/tinygo/blob/8d93b9e54592a0069b7ea3a993b5fa0a0ed3fde3/main.go#L285
https://github.com/tinygo-org/tinygo/blob/8d93b9e54592a0069b7ea3a993b5fa0a0ed3fde3/targets/pca10059.json#L5

```
sago3@B450GT3 MINGW64 /c/tinygo/tinygo (flash-commands)
$ tinygo version
tinygo version 0.18.0-dev-8d93b9e5 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)

sago3@B450GT3 MINGW64 /c/tinygo/tinygo (flash-commands)
$ tinygo flash --target pca10059 --size short -x ./src/examples/blinky1
   code    data     bss |   flash     ram
   7368      32    4080 |    7400    4112
nrfutil pkg generate --hw-version 52 --sd-req 0x0 --application C:\Users\sago3\AppData\Local\Temp\tinygo042226627\main.hex --application-version 1 /tmp/tinygo_$$.zip && nrfutil dfu usb-serial -pkg /tmp/tinygo_$$.zip -p COM1 -b 115200 && rm -f /tmp/tinygo_$$.zip
 Usage: nrfutil pkg generate [OPTIONS] ZIPFILE
Try "nrfutil pkg generate --help" for help.

Error: no such option: -p
error: failed to flash C:\Users\sago3\AppData\Local\Temp\tinygo042226627\main.hex: exit status 2
```

aykevl, 2021-03-29:
Yes, using `flash-command` like this is a bad idea. It is very non-portable.

Instead, I would suggest adding a new binary format. Take a look here:

https://github.com/tinygo-org/tinygo/blob/8d93b9e54592a0069b7ea3a993b5fa0a0ed3fde3/builder/build.go#L553-L560

You can add a new binary format (for example, `"nrf-dfu"`) that is created by running `nrfutil`. This also avoids writing to `/tmp` as it can use the temporary build directory instead which is always removed after tinygo finishes.

oflebbe, 2021-03-30:
thanks @sago35 for reporting, using a shell concat for flashing was a bad idea (tm). Thanks @aykevl for the suggestion, since I didn't respond in time asking myself how to fix the problem. There will be some questions on the way.

oflebbe, 2021-06-14:
sorry for not mentioning it in commit message of #1931 @sago35 . we can close this now, i guess

deadprogram, 2021-06-14:
Labelling to close on next release.

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

closed 2021-07-01 by deadprogram