tinygo-org/tinygo #1762: nodemcu (Amica with ESP8266MOD) examples/blinky1 not working

soypat, 2021-03-31:
Tried following the instructions to flash blinky1 at [https://tinygo.org/microcontrollers/esp8266-nodemcu/](https://tinygo.org/microcontrollers/esp8266-nodemcu/). 

Light was initially blinking before flashing tinygo program (I believe they come with a factory loaded blinky). After flashing the light stayed off

## Output
```
$ tinygo flash -port=/dev/ttyUSB0 -target=nodemcu -size=short examples/blinky1
   code    data     bss |   flash     ram
   2863       0      64 |    2863      64
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:51:bb:5e
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 3072 bytes to 2426...
Wrote 3072 bytes (2426 compressed) at 0x00000000 in 0.2 seconds (effective 113.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

sago35, 2021-04-06:
If you have a schematic, please let me know the URL.

Is the result the same if I try using other pins?
You can set the other pins to output settings as follows

```
// https://github.com/tinygo-org/tinygo/blob/release/src/examples/blinky1/blinky1.go#L11

	led := machine.D0
```

A list of pins can be found below
https://github.com/tinygo-org/tinygo/blob/release/src/machine/board_nodemcu.go#L9-L17

soypat, 2021-04-07:
I'll check as soon as I get my hands back on a nodemcu. This is the board https://www.luisllamas.es/esp8266-nodemcu/

wuhanstudio, 2021-04-10:
I have the same issue, but it has been fixed by changing:

https://github.com/tinygo-org/tinygo/blob/138befd8a25d2a5c65e2e740038a8c3a44e95222/targets/esp8266.json#L19

to
 
```
"flash-command": "esptool --chip=esp8266 --port {port} write_flash 0x00000 {bin} -fm dio"
```

Maybe it's better to add an argument for flashing mode.

soypat, 2021-04-12:
@sago35 Apparently my board is not flashing correctly. Trying out other pins and measuring voltage it seems the blinky program does nothing, including for other pins.

I'd try out wuhanstudio's solution but I'm unsure how to go about doing that modification... do I have to compile from source after modifying?

wuhanstudio, 2021-04-12:
> @sago35 Apparently my board is not flashing correctly. Trying out other pins and measuring voltage it seems the blinky program does nothing, including for other pins.
>
> I'd try out wuhanstudio's [...]

It's not required to compile from source.

In Windows, it works by changing **tinygo/targets/esp8266.json** in the installation directory of pre-compiled tinygo.

[image]

soypat, 2021-04-12:
Yes! that worked, thanks wuhanstudio. I'm on linux so it was in my `/usr/local/lib/tinygo` directory. Changing this file solves the issue. However I seem to have found a new bug in the process. The following blinky program does not work:

```go
package main

import (
	"machine"
	"time"
)

var cfg = machine.PinConfig{Mode: machine.PinOutput}
// The problem seems to lie with using D0 and D4 together. not sure though
var x = [...]machine.Pin{machine.LED, machine.D0}

func main() {
	for i := range x {
		x[i].Configure(cfg)
	}

	for {
		for i := range x {
			x[i].Low()
		}
		time.Sleep(time.Millisecond * 1000)
		for i := range x {
			x[i].High()
		}
		time.Sleep(time.Millisecond * 1000)
	}
}
```

The result should be blinky on `machine.LED` and `machine.D0` but it seems to crash leaving the LED on indefinitely. If I change `machine.D0` to `machine.D1` the program works as expected.

wuhanstudio, 2021-04-12:
[image]

For esp8266, the GPIO0 can not be controlled because it's defines the boot mode.

The ESP8266 will enter the serial bootloader when GPIO0 is held low on reset. Otherwise it will run the program in flash.

GPIO0 Input | Mode
-- | --
Low/GND | ROM serial bootloader for esptool.py
High/VCC | Normal execution mode

soypat, 2021-04-12:
Alright, that solves my issue.thanks @wuhanstudio

closed 2021-04-12 by soypat