tinygo-org/tinygo #1764: Exported WASM function can't use time package effectively

jsimnz, 2021-04-02:
Do to the limitation on async functionality in exported functions (as discuessed [here](https://github.com/tinygo-org/tinygo/issues/1651)), much of the standard library `time` package can't be used.

Here's as basic example
```
package main

import (
    "time"
}

//export myFunc
func myFunc() int {
    return time.Now().Year()
}

func main() {}
```

Trying to build this program with the following: `tinygo build -o main.wasm main.go`
returns the following error:
```
blocking operation in exported function: someFunc

traceback:
(time.Time).Year
        /usr/local/go/src/time/time.go:505:25
(time.Time).date
        /usr/local/go/src/time/time.go:974:22
(time.Time).abs
        /usr/local/go/src/time/time.go:459:12
(*time.Location).get
        /usr/local/go/src/time/zoneinfo.go:85:15
(*sync.Once).Do
        /usr/local/lib/tinygo/src/sync/once.go:9:10
(*sync.Mutex).Lock
        /usr/local/lib/tinygo/src/sync/mutex.go:23:13
```

Much of this stems from the time package trying to get the local system time, which does a lookup [here](https://github.com/golang/go/blob/db8142fb8631df3ee56983cbc13db997c16f2f6f/src/time/time.go#L432), which calls this function:
```
func (l *Location) get() *Location {
	if l == nil {
		return &utcLoc
	}
	if l == &localLoc {
		localOnce.Do(initLocal)
	}
	return l
}
```

We can see that there's a branch where it tries to run `localOnce.Do`, where `localOnce` is a `sync.Once` object which implicity calls a `sync.Mutex.Lock()`.

Unfortunately, do to the tinygo restrictions on exported functions using async functions, this causes most of the `time` package unusable, since one way or another it will call this problematic `get` function.

What's even more strange is the `get` call is just used to get the `time.Location` info, which you can manually set using like so:
```
t := time.Now()
t.In(time.UTC)
```

If you manually set the location like this, it avoids the branch of code which would cause a `sync.Once.Do()` call. However, the compiler still complains and returns the same stack trace.

Not sure what the workaround is here, perhaps using a native time package that avoids the async calls, or to update the compiler to understand when an `async` call actually happens vs just the possibility of it happening.

Thanks!

OneOfOne, 2021-04-07:
I just use `time.Now().UTC().XXX()` mostly

jsimnz, 2021-04-07:
Unfortunately, that doesn't fix the issue.

LIke I mentioned, you can manually set the timezone location using your indicated method (`time.Now().UTC().XXX`), or using `time.Now().In(Time.UTC)` as I posted, and the result is the same.

Even if the runtime doesn't _actually_ need to do the timezone location lookup, its a _potential_ code branch, and the tinygo compiler complains all the same.

jsimnz, 2021-04-07:
As seen here, trying to compile this program (with the suggested change), still produces the exported async compiler error
```
package main

import (
    "time"
}

//export myFunc
func myFunc() int {
    return time.Now().UTC().Year()
}

func main() {}
```

codyohl, 2021-09-10:
any update here?

nakiner, 2021-10-05:
same here, will have to re-implement blocking operations by overriding package. @codyohl maybe any updates here?

AlexGustafsson, 2022-05-14:
This seem to largely be fixed now, right? Everything seems to work fine for me, except timezone handling (`time.Now().Local()` does not have the same behavior as regular Go and `time.LoadLocation(timezone)` panics).

codefromthecrypt, 2022-09-07:
@jsimnz using latest tinygo I was able to compile your original code without a compiler error.

I agree that there are some inconsistencies about timezone use in practice (ex what the sandbox defines as the current timezone and also tz data).

That said, I'm trying to help folks focus attention on WebAssembly issues and unclosed issues distract from that, as people will keep looking to see if it still is broke or not. Mind closing this out?

jsimnz, 2022-09-08:
> @jsimnz using latest tinygo I was able to compile your original code without a compiler error.

Awesome, thanks for checking in and pinging me, happy to close! Didn't notice which fix wouldve changed the behavior on this, but happy it's in.

closed 2022-09-08 by jsimnz