tinygo-org/tinygo #1778: Executable not declared by package os

rogchap, 2021-04-06:
`os.Execuatable()` is missing from the `os` package

https://golang.org/src/os/executable.go

My particular error is caused by the `protobuf` package here: https://github.com/protocolbuffers/protobuf-go/blob/v1.26.0/internal/detrand/rand.go#L42

sillyY, 2021-08-06:
Is the problem solved？

mvdan, 2021-08-12:
I think this missing API should be added, but I've also created an issue over at Protobuf to make them play nicer with `purego`: https://github.com/golang/protobuf/issues/1354

mvdan, 2021-08-13:
Err, they do play nice with `purego` - they will simply fall back to a default if `os.Executable` errors out, just like it does for `js/wasm` on Go's upstream port. So I think it would be really valuable for tinygo to offer a similar stub API declaration.

For a bit of context, this os package API was added all the way back in 1.8.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram