tinygo-org/tinygo #1784: Can I use TinyGo for the ATmega328 chip?

ntthang-dev, 2021-04-10:
I see TinyGo has support for *microcontroller programming* with *ATmega328p* chip for *Ardunio Uno*. 😀
Currently if I want to use TinyGo for *Ardunio Uno R3* with *ATmega328* 👨‍💻 chip is it okay and if it makes any difference in performance or way it works 🤔 and how will I have to do that 😣 I would like to thank 😋 !

soypat, 2021-04-10:
Yes! There should be no problem. You can find some examples here:
* At the [tinygo repo](https://github.com/tinygo-org/tinygo/tree/release/src/examples)
* I've been working on some examples specifically tailored for Arduino users [here](https://github.com/soypat/tinygo-arduino-examples)

ntthang-dev, 2021-04-11:
Oh, that's it! Thanks for taking the time to help me so much. I will try that. 😄

closed 2021-04-18 by deadprogram