tinygo-org/tinygo #1790: wasm: musttail call must precede a ret with an optional bitcast

FrankReh, 2021-04-11:
```
% tinygo build -o try.wasm -target wasm
musttail call must precede a ret with an optional bitcast
  musttail call fastcc void @runtime.runqueuePushBack(%"internal/task.Task"* nonnull %parentHandle.reload1), !dbg !1172
error: GC pass caused a verification failure
```
Version reports
```
tinygo version 0.17.0 darwin/amd64 (using go version go1.16.3 and LLVM version 11.1.0)
```
Problem does not manifest when -opt=1 is used. Problem also reproduced from the latest dev branch.

Ran into this when building for wasm where a js.FuncOf was being called. Reduced the problem to find it is some combination of calls from the syscall/js package which I reduced further to a stand alone main.go file below. But the problem goes away if the sync.Lock call is removed so that sync package I felt I had to keep in the reduction. Many attempts at reducing the main.go further also made the problem go away so I could not reduce it down to just one line that makes a difference. Sorry there are so many functions in this reproduction.

Removing the sync.Lock makes the problem go away. Also just removing the one call to panic makes the problem go away. The great job LLVM does at removing dead code though makes it hard to guess why a line does or does not affect tail call logic. Even just removing the *int field from the struct effects the outcome, even though the code doesn't use the field - guess it is the size of the struct.

```
// +build js,wasm

package main

import "sync"

const (
	T0 int = iota
	T1
	T2
)

var (
	mu sync.Mutex
	v1 = make1(1) // change 1 to 0, get another kind of crash
	v2 = v1.Method1()
)

func main() {
	_ = LockAndReturn()
}
func LockAndReturn() error {
	mu.Lock()
	mu.Unlock()
	return nil
}

type S1 struct {
	f1 int
	f2 *int
}

func make1(b byte) S1 {
	return S1{f1: (int(b))}
}

func (v S1) Method1() S1 {
	if t := v.Type(); !isObject(t) {
		panic(&ValueError{})
	}
	return makeValue(valueGet(0, ""))
}

func makeValue(r int) S1 {
	return S1{}
}

func valueGet(v int, p string) int

type ValueError struct {
	Method string
}

func (e *ValueError) Error() string {
	return "something" + e.Method
}

func isObject(t int) bool {
	return t == T1 || t == T2
}

func (v S1) Type() int {
	if v.f1 == 0 {
		return T0
	}
	return T2
}
```

FrankReh, 2021-04-30:
This is the IR. I believe it comes from the coroutine lowering. WebAssembly can't do tail calls yet so it's odd a musttail is even making it into the generated code.

```
"(*internal/task.Task).returnTo.exit":            ; preds = %"(*sync.Mutex).Unlock.exit"
  %task.state.parent.reload.addr = getelementptr inbounds %main.LockAndReturn.Frame, %main.LockAndReturn.Frame* %FramePtr, i32 0, i32 3, !dbg !1199
  %8 = bitcast i8** %task.state.parent.reload.addr to i32*, !dbg !1199
  %task.state.parent.reload2 = load i32, i32* %8, align 4, !dbg !1199
  %9 = getelementptr inbounds %"internal/task.Task", %"internal/task.Task"* %parentHandle.reload1, i32 0, i32 3, i32 0, !dbg !1247
  call void @llvm.dbg.value(metadata i8* undef, metadata !1242, metadata !DIExpression()), !dbg !1248
  %10 = bitcast i8** %9 to i32*, !dbg !1248
  store i32 %task.state.parent.reload2, i32* %10, align 4, !dbg !1248
  call void @llvm.dbg.value(metadata %"internal/task.Task"* undef, metadata !1239, metadata !DIExpression()), !dbg !1249
  call void @llvm.dbg.value(metadata %"internal/task.Task"* undef, metadata !1191, metadata !DIExpression()), !dbg !1250
  musttail call fastcc void @runtime.runqueuePushBack(%"internal/task.Task"* nonnull %parentHandle.reload1), !dbg !1252
  ret void, !dbg !1199
}
```

FrankReh, 2021-04-30:
Oh I see the problem. There is a pass that adds a store instruction between the musttail and the ret. I had been printing the IR out too early. With hindsight, this seems like a silly error on my part - not outputting the IR right before the error was reported. I had assumed the mistake was having a musttail in the first place. Now it is likely the problem is with the store being inserted there - perhaps the step that added the store could have put it before the last tail call or the pass could have removed the musttail marker itself. Here is the offending IR and now it is clear why there is a complaint.

```
"(*internal/task.Task).returnTo.exit":            ; preds = %"(*sync.Mutex).Unlock.exit"
  %task.state.parent.reload.addr = getelementptr inbounds %main.LockAndReturn.Frame, %main.LockAndReturn.Frame* %FramePtr, i32 0, i32 3, !dbg !1171
  %16 = bitcast i8** %task.state.parent.reload.addr to i32*, !dbg !1171
  %task.state.parent.reload2 = load i32, i32* %16, align 4, !dbg !1171
  %17 = getelementptr inbounds %"internal/task.Task", %"internal/task.Task"* %parentHandle.reload1, i32 0, i32 3, i32 0, !dbg !1217
  call void @llvm.dbg.value(metadata i8* undef, metadata !1212, metadata !DIExpression()), !dbg !1218
  %18 = bitcast i8** %17 to i32*, !dbg !1218
  store i32 %task.state.parent.reload2, i32* %18, align 4, !dbg !1218
  call void @llvm.dbg.value(metadata %"internal/task.Task"* undef, metadata !1209, metadata !DIExpression()), !dbg !1219
  call void @llvm.dbg.value(metadata %"internal/task.Task"* undef, metadata !1163, metadata !DIExpression()), !dbg !1220
  musttail call fastcc void @runtime.runqueuePushBack(%"internal/task.Task"* nonnull %parentHandle.reload1), !dbg !1222
  store %runtime.stackChainObject* %0, %runtime.stackChainObject** @runtime.stackChainStart, align 4, !dbg !1171
  ret void, !dbg !1171
}
```

FrankReh, 2021-04-30:
```
		// Make sure this stack object is popped from the linked list of stack
		// objects at return.
		for _, ret := range returns {
			builder.SetInsertPointBefore(ret)
			builder.CreateStore(parent, stackChainStart)
		}
```

FrankReh, 2021-04-30:
The musttail marker can be turned into a tail marker to solve this. Or the insertion point could be put before the call that has the musttail on it. Or if it is known that this is only used for targets that don't support tail calls any, all tail and musttail markers can be removed. I've tried all three approaches and they all solve the problem from my perspective.

fgsch, 2021-05-12:
Also running into this with the following test case:

```
package main

import (
	"time"
	"fmt"
)

func main() {
	c := make(chan string, 1)
	for i := 0; i < cap(c); i++ {
		go func(i int) {
			time.Sleep(time.Millisecond)
			fmt.Printf("goroutine %d\n", i)
		}(i)
	}
}
```

It builds fine with the leaking gc.

fgsch, 2021-07-29:
Moved from #1979 :

```
package main

import "sync"

var once sync.Once

func bug() *int {
	once.Do(func() {})
	i := 0
	return &i
}

func main() {
	bug()
}
```

aykevl, 2021-08-01:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/2022

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

PR #2022 (2021-08-04): transform: improve GC stack slot pass to work around a bug
Bug 1790 ("musttail call must precede a ret with an optional bitcast")
is caused by the GC stack slot pass inserting a store instruction
between a musttail call and a return instruction. This is not allowed in
LLVM IR.

One solution would be to remove the musttail. That would probably work,
but 1) the go-llvm API doesn't support this and 2) this might have
unforeseen consequences. What I've done in this commit is to move the
store instruction to a position earlier in the basic block, just after
the last access to the GC stack slot allo [... cut]
closed 2021-09-21 by deadprogram