tinygo-org/tinygo #1792: Arduino Uno R3: complete wrong sleep time with time.Sleep values < 17 milliseconds

Polarbaer50, 2021-04-12:
Hello all
in my Arduino Uno R3 project I use multiplexing to light up different LEDs. The multiplexed ON time for an LED should be around 1ms. However whenever I lower the sleep time below 17ms the sleep time is much to short. 
Is this a know issue?
Could there be a problem in tinygo/src/runtime/runtime_avr.go?
Thanks for someone looking into this issue

$ tinygo version
tinygo version 0.17.0 linux/amd64 (using go version go1.16.3 and LLVM version 11.0.0)

package main
import (
	"machine"
	"time"
)
var gpio = []machine.Pin{
        machine.ADC0,
        machine.ADC1,
        machine.D2,
        machine.D3,
    }

func main() { 
	// define gpio 0 .. 3 as output and set them to LOW
	for i := 0; i < len(gpio)-1; i++ {
        gpio[i].Configure(machine.PinConfig{Mode: machine.PinOutput})
	}
 for {  // endless loop
	gpio[0].High()					 //	pulse appr. 4us		  
	gpio[0].Low()

	gpio[1].High()
	time.Sleep(time.Millisecond*50)  // sleeps for appr.35ms
	gpio[1].Low()
	
	gpio[2].High()
	time.Sleep(16*time.Millisecond)  // sleeps for appr.20us
	gpio[2].Low()
	
	gpio[3].High()
	time.Sleep(17*time.Millisecond)  // sleeps for appr.17ms
	gpio[3].Low()
	}
}

cweagans, 2022-01-02:
Fixed by https://github.com/tinygo-org/tinygo/pull/2428

Polarbaer50, 2022-01-10:
I can confirm that the changes applied with #2428 solve this issue.
time.sleep with values < 17ms now work as expected.
**Thank you to all of you how worked on solving this timing issue!**

closed 2022-01-10 by Polarbaer50