tinygo-org/tinygo #1795: Tinygo test fails

Zagidziran, 2021-04-13:
Hi I've just installed go and tinygo with the scoop

scoop install go
scoop install tinygo 

then I've took an example from here https://github.com/tetratelabs/proxy-wasm-go-sdk/tree/main/examples/http_headers

and powered it with

### go.mod 
`module main

go 1.15

require (
	github.com/stretchr/testify v1.7.0 // indirect
	github.com/tetratelabs/proxy-wasm-go-sdk v0.1.1
) `
and

### go.sum

`github.com/davecgh/go-spew v1.1.0 h1:ZDRjVQ15GmhC3fiQ8ni8+OwkZQO4DARzQgrnXU1Liz8=
github.com/davecgh/go-spew v1.1.0/go.mod h1:J7Y8YcW2NihsgmVo/mv3lAwl/skON4iLHjSsI+c5H38=
github.com/pmezard/go-difflib v1.0.0 h1:4DBwDE0NGyQoBHbLQYPwSUPoCMWR5BEzIk/f1lZbAQM=
github.com/pmezard/go-difflib v1.0.0/go.mod h1:iKH77koFhYxTK1pcRnkKkqfTogsbg7gZNVY4sRDYZ/4=
github.com/stretchr/objx v0.1.0/go.mod h1:HFkY916IF+rwdDfMAkV7OtwuqBVzrE8GR6GFx+wExME=
github.com/stretchr/testify v1.6.1/go.mod h1:6Fq8oRcR53rry900zMqJjRRixrwX3KX962/h/Wwjteg=
github.com/stretchr/testify v1.7.0 h1:nwc3DEeHmmLAfoZucVR881uASk0Mfjw8xYJ99tb5CcY=
github.com/stretchr/testify v1.7.0/go.mod h1:6Fq8oRcR53rry900zMqJjRRixrwX3KX962/h/Wwjteg=
github.com/tetratelabs/proxy-wasm-go-sdk v0.0.8/go.mod h1:y1ZQT4bQEBnR8Do4nSOzb3roczzPvcAp8UrF6NEYWNY=
github.com/tetratelabs/proxy-wasm-go-sdk v0.1.1 h1:m8O4nWCyb+8VlAxVxppobGNnC1N9CoAQBmMfRjTq/hU=
github.com/tetratelabs/proxy-wasm-go-sdk v0.1.1/go.mod h1:y1ZQT4bQEBnR8Do4nSOzb3roczzPvcAp8UrF6NEYWNY=
gopkg.in/check.v1 v0.0.0-20161208181325-20d25e280405/go.mod h1:Co6ibVJAznAaIkqp8huTwlJQCZ016jof/cbN4VW5Yz0=
gopkg.in/yaml.v3 v3.0.0-20200313102051-9f266ea9e77c h1:dUUwHk2QECo/6vqA44rthZ8ie2QXMNeKRTHCNY2nXvo=
gopkg.in/yaml.v3 v3.0.0-20200313102051-9f266ea9e77c/go.mod h1:K4uyk7z7BCEPqu6E+C64Yfv1cQ7kz7rIZviUmN+EgEM= `

and then when running **tinygo test**
I'm receiving

`# runtime
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:29:21: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:80:42: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\gc_extalloc.go:412:4: undeclared name: extfree
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\gc_extalloc.go:577:10: undeclared name: extalloc
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\gc_stack_raw.go:19:38: undeclared name: stackTop
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\gc_stack_raw.go:34:17: undeclared name: stackTop
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\panic.go:13:2: undeclared name: abort
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\panic.go:20:2: undeclared name: abort
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:14:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:26:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:35:4: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:58:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:68:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:89:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:95:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:183:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:263:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:268:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:275:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:282:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:286:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:287:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:333:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:342:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:344:3: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:355:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:363:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:364:2: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:368:4: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\print.go:370:4: undeclared name: putchar
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\runtime.go:63:9: undeclared name: ticksToNanoseconds
C:\Users\dantipenko\scoop\apps\go\current\src\runtime\runtime.go:63:28: undeclared name: ticks
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:88:9: undeclared name: ticks
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:118:10: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:123:10: undeclared name: ticks
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:128:53: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:131:26: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:140:8: undeclared name: asyncScheduler
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:146:16: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:150:39: undeclared name: timeUnit
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:153:4: undeclared name: sleepTicks
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler.go:154:7: undeclared name: asyncScheduler
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler_any.go:10:31: undeclared name: nanosecondsToTicks
..\..\Users\dantipenko\AppData\Local\tinygo\goroot-go1.16.3-ffe24d7cee92675718c75850eada46b3a359a1f846cf7f30f299ac7a4eeaa8ee\src\runtime\scheduler_any.go:20:3: undeclared name: postinit`

which makes me think it uses go runtime instead the tingo runtime. How can i fix it?

aykevl, 2021-04-13:
Please post code samples with triple back ticks, like so:

    ```
    some code
    ```

That makes the code a lot easier to read.

aykevl, 2021-04-13:
Regarding your issue: the problem is that you are not specifying a target and it defaults to the host (GOOS=windows). Windows is not a supported target at the moment. So if you want to run tests on WebAssembly, you need to run them using `-target=wasm` or the appropriate environment variables.

    GOOS=js GOARCH=wasm tinygo test

(I'm not sure how this works in Windows).

aykevl, 2021-04-13:
> which makes me think it uses go runtime instead the tingo runtime. How can i fix it?

It also appears there is a bug in how TinyGo presents these error messages: the paths appear to be wrong. The errors are certainly coming from the TinyGo runtime (which doesn't support Windows) but it shows the wrong paths for them.

QuLogic, 2022-02-06:
Might this be fixed by #2231?

aykevl, 2022-02-06:
Yes, this should be fixed now.

closed 2022-02-06 by aykevl

asabya, 2022-02-17:
I'm not sure if this relates to this issue.
If I run `GOOS=js GOARCH=wasm tinygo build -o asd.wasm` on `version: 0.22.0` I get the following

```
# os/user
/usr/local/go/src/os/user/lookup.go:15:41: undeclared name: current
/usr/local/go/src/os/user/lookup.go:36:9: undeclared name: lookupUser
/usr/local/go/src/os/user/lookup.go:45:9: undeclared name: lookupUserId
/usr/local/go/src/os/user/lookup.go:51:9: undeclared name: lookupGroup
/usr/local/go/src/os/user/lookup.go:57:9: undeclared name: lookupGroupId
/usr/local/go/src/os/user/lookup.go:62:9: undeclared name: listGroups
```