tinygo-org/tinygo #1796: Speed up llvm-source step and llvm-build step in ci

sago35, 2021-04-14:
note: If I make a Pull-Request, a cache will probably be created in circleci, so I'll write it in the issue first.

I have improved the following two points.
I think it will speed up the CI in most cases.

However, since we are using a cache-reducing method, the cache may need to be updated in future dev versions.
Typically, this happens when you try to use a file that is not currently in use.
I would especially like to know if you have any opinions on this disadvantage.

* llvm-source-*
    * Change the cache to remove all files except *.h and *.def under the llvm-project directory.
    * This change will reduce the restore-cache on linux from 21 sec to 1 sec
* llvm-build-*
    * Change the cache to remove *.o
    * This change will reduce the restore-cache on linux from 11 sec to 7 sec

The changes can be found below.
For testing purposes, the CI targets have been narrowed down to build-linux, build-macos and assert-test-linux, but will eventually be restored.
I will also rename the cache to an appropriate name.

* changes
    * https://github.com/tinygo-org/tinygo/compare/dev...sago35:ci-llvm-build
* CI result
    * old
        * https://app.circleci.com/pipelines/github/tinygo-org/tinygo/4727/workflows/20480dd8-d147-4818-bd0c-0a7b598e44a2
    * new
        * https://app.circleci.com/pipelines/github/sago35/tinygo/38/workflows/a4108973-ce2e-4a34-930c-90031977eede

[image]

aykevl, 2021-04-14:
> * llvm-source-*
> 
>     * Change the cache to remove all files except *.h and *.def under the llvm-project directory.
> 
>     * This change will reduce the restore-cache on linux from 21 sec to 1 sec

Yes, this is a good idea. I wanted to make this change but haven't done it yet. However, I think it is better to cache specific directories instead of specific file extensions. For example, the directory that is currently needed is `llvm-project/clang/lib/Headers`. Like this:

```yaml
       - save_cache:
          key: llvm-source-11-v1
          paths:
            - llvm-project/clang/lib/Headers
```

Note that the whole llvm-directory is needed when building LLVM, so in the `make llvm-build` step this directory needs to be cleaned entirely and recreated using `make llvm-source`.

> * llvm-build-*
> 
>     * Change the cache to remove *.o
> 
>     * This change will reduce the restore-cache on linux from 11 sec to 7 sec

Again, I think it is better to remove (or keep) specific directories instead of file extensions. I think all `CMakeFiles` directories do not need to be cached: those directories contain all .o files. Or maybe it's enough to only save `llvm-build/bin` and `llvm-build/lib/*.a`.

sago35, 2021-04-15:
> Yes, this is a good idea. I wanted to make this change but haven't done it yet. However, I think it is better to cache specific directories instead of specific file extensions. For example, the directory that is currently needed is llvm-project/clang/lib/Headers. Like this:

It worked well.

```
          paths:
            - llvm-project/clang/lib/Headers
            - llvm-project/clang/include
            - llvm-project/lld/include
            - llvm-project/llvm/include
```

> Again, I think it is better to remove (or keep) specific directories instead of file extensions. I think all CMakeFiles directories do not need to be cached: those directories contain all .o files. Or maybe it's enough to only save llvm-build/bin and llvm-build/lib/*.a.

It seems that wildcard cannot be written in the paths of circleci.
I think it is necessary to delete all files except llvm-build/lib*.a.
Is there a better way?

sago35, 2021-04-16:
@aykevl 
I tried deleting everything except `lib/lib*.a`, but the build failed.
I think it's easier to delete the object (*.o / *.obj) in llvm-build.

Test TinyGo:
```
ulvti3.c'
error: no input files
error: no such file or directory: '/home/circleci/project/lib/compiler-rt/lib/builtins/negvsi2.c'
error: no input files
error: no such file or directory: '/home/circleci/project/lib/compiler-rt/lib/builtins/negvdi2.c'
...
```

* sourcecode
    * https://github.com/sago35/tinygo/commit/535f88369534ebcc725102fa2049163fd377a021
* circleci
    * https://app.circleci.com/pipelines/github/sago35/tinygo/44/workflows/c3c0559f-30b4-4122-a522-4cf374bbd104/jobs/197

aykevl, 2021-04-18:
>  I think it's easier to delete the object (*.o / *.obj) in llvm-build.

You can use `find llvm-build -name CMakeFiles -prune -exec rm -r '{}' \;`. This removes all object files and even some other unnecessary files (1MB more removed).

sago35, 2021-04-18:
> You can use find llvm-build -name CMakeFiles -prune -exec rm -r '{}' \;. This removes all object files and even some other unnecessary files (1MB more removed).

It worked fine.
After #1234 is merged, I' ll create a PR.

https://app.circleci.com/pipelines/github/sago35/tinygo/49/workflows/74d49b91-b46d-4e39-b53d-f3c9248e3649

deadprogram, 2021-04-21:
@sago35 can this issue now be closed?

sago35, 2021-04-21:
@deadprogram 
Yes, this issue has been resolved.

* merged
    * #1819 
    * #1822

deadprogram, 2021-04-21:
Thanks @sago35!

closed 2021-04-21 by deadprogram