tinygo-org/tinygo #1801: Output of "tinygo test -target=wasi -c" not working with wasmtime or wasmer?

dkegel-fastly, 2021-04-14:
A no-op test passes with tinygo test and the default target, but I'm not sure how to get it to pass with wasm and the wasmtime or wasmer runtimes.

```
$ tinygo version
tinygo version 0.17.0 darwin/amd64 (using go version go1.16.3 and LLVM version 11.1.0)
$ tinygo test -target=wasi -scheduler=none -wasm-abi=generic -c main_test.go
$ wasmtime --version
wasmtime 0.23.0
$ wasmtime run tt.test -- -test.run .
Error: failed to run main module `tt.test`
Caused by:
    0: failed to instantiate "tt.test"
    1: unknown import: `env::syscall.Exit` has not been defined
$ wasmer --version
wasmer 1.0.2
$ wasmer run tt.test
error: failed to run `tt.test`
╰─> 1: Error while importing "wasi_unstable"."fd_write": unknown import. Expected Function(FunctionType { params: [I32, I32, I32, I32], results: [I32] })
```
Here's the no-op test case:
```
$ pwd
/Users/dank/tt
$ cat main_test.go
package foo

import "testing"

func TestFoo(t *testing.T) {
	;
}
$ cat go.mod
module foo

go 1.15
$ tinygo test
=== RUN   TestFoo
--- PASS: TestFoo
PASS
```

I probably missed a memo somewhere, but it'd be nice if it were clearer how to do this.

aykevl, 2021-04-14:
See https://github.com/tinygo-org/tinygo/pull/1804 for a fix.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!