tinygo-org/tinygo #1805: regression at 312f5d3 "builder: run interp per package": "panic: interface conversion"?

dkegel-fastly, 2021-04-15:
An app here built ok with tinygo 0.17 (either from release branch using go install or using "brew install tinygo").

The same app now fails with tinygo built from dev branch ( v0.17.0-99-g0d66475 ) using go install:
```
$ tinygo build -target=wasi -scheduler=none -wasm-abi=generic -o bin/main.wasm
panic: interface conversion: interp.value is interp.rawValue, not *interp.mapValue

goroutine 52 [running]:
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc00447e1b0, 0xc003d98be0, 0xc000929190, 0x6, 0x7, 0xc00073d8c0, 0xc0053554c8, 0x8, 0x0, 0x4257d65, ...)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/interp/interpreter.go:467 +0xc695
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc00447e1b0, 0xc003d98b90, 0x0, 0x0, 0x0, 0x0, 0x43f4c7b, 0x4, 0x0, 0x0, ...)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/interp/interpreter.go:522 +0x5ed3
github.com/tinygo-org/tinygo/interp.Run(0x3a7c4c00, 0xc005a7bd00, 0x0, 0x0)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/interp/interp.go:112 +0x69f
github.com/tinygo-org/tinygo/builder.optimizeProgram(0x3a7c4c00, 0xc0002721e0, 0xc0051c4780, 0x2)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/builder/build.go:649 +0x3f
github.com/tinygo-org/tinygo/builder.Build.func2(0xc00414bf20, 0x0, 0x0)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/builder/build.go:384 +0x7c6
github.com/tinygo-org/tinygo/builder.jobWorker(0xc0012b5260, 0xc0012b5200)
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/builder/jobs.go:165 +0xe2
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/dkegel/go/src/github.com/tinygo-org/tinygo/builder/jobs.go:84 +0xc5
```

Bisecting was a little hard, so I built all 99-odd commits from release to dev.
To pierce the veil of vanished go-llvm versions, I had to apply https://github.com/tinygo-org/tinygo/commit/1e9a41d

The first bad commit was

v0.17.0-63-g312f5d3  builder: run interp per package

https://github.com/tinygo-org/tinygo/commit/312f5d3

Paging @aykevl !

aykevl, 2021-04-15:
Can you share how to reproduce this issue? The bisected commit is a big change (even if it might look small) and so it's hard to see what exactly is going wrong.

dkegel-fastly, 2021-04-15:
I was afraid you'd ask that :-) I will try to come up with a minimal reproducer.

fgsch, 2021-04-15:
Here is a minimal repo:

```
package main
type foo map[byte]int
func (f foo) bug() {
        for a := range "a" {
                f[byte(a)] = 0 // bug!
        }
}
func xxx() *string {
        foo{}.bug()
        return nil
}
var a = xxx()
func main() {
}
```

dkegel-fastly, 2021-04-15:
fgsch is a rockstar :-)

aykevl, 2021-04-20:
Here is a bugfix: https://github.com/tinygo-org/tinygo/pull/1827

deadprogram, 2021-04-21:
Actually, since this regression appeared in-cycle, I am going to close it now. Thank you very much @dkegel-fastly for the report and to @fgsch for the repro case!

closed 2021-04-21 by deadprogram