tinygo-org/tinygo #1807: Error building wasm with -scheduler=tasks

fgsch, 2021-04-15:
Version: 

`tinygo version 0.18.0-dev-6dd5666 linux/amd64 (using go version go1.16.3 and LLVM version 11.0.0)`

Command line:

`tinygo build -target=wasi -scheduler=tasks  -o main.wasm`

Test case:

```go
package main

func main() {
        go func() {}()
}
```

Error:

```
# internal/task
/opt/tinygo/src/internal/task/task_stack.go:45:20: currentTask.state.pause undefined (type state has no field or method pause)
/opt/tinygo/src/internal/task/task_stack.go:57:10: t.state.resume undefined (type state has no field or method resume)
/opt/tinygo/src/internal/task/task_stack.go:77:9: undeclared name: calleeSavedRegs
/opt/tinygo/src/internal/task/task_stack.go:77:83: undeclared name: calleeSavedRegs
/opt/tinygo/src/internal/task/task_stack.go:80:4: s.archInit undefined (type *state has no field or method archInit)
```

fgsch, 2021-07-07:
@aykevl is the tasks scheduler supposed to work with wasm or this combination should not be allowed?

aykevl, 2021-07-07:
No. The tasks scheduler only works on non-WebAssembly architectures because it relies on directly modifying the stack pointer (something that is not allowed in WebAssembly). The only options are `-scheduler=none` and `-scheduler=coroutines`.

aykevl, 2021-07-07:
Closing as I don't think there is a lot that can be done here.

That said, I'm curious why you wanted to try `-scheduler=tasks`?

closed 2021-07-07 by aykevl

fgsch, 2021-07-08:
Can we fail with this combination then? I'd rather get an error saying you the scheduler tasks is not available for the wasm/wasi target.

I didn't want to use the tasks scheduler per se, I was trying different combinations when I saw this error.