tinygo-org/tinygo #1811: tinygo-drivers not building after PWM refactor

kenbell, 2021-04-16:
build error...
```shell
tinygo build -size short -o ./build/test.hex -target=itsybitsy-m0 ./examples/apa102/itsybitsy-m0/main.go
# command-line-arguments
examples/apa102/itsybitsy-m0/main.go:18:18: PWM not declared by package machine
examples/apa102/itsybitsy-m0/main.go:30:10: InitPWM not declared by package machine
make: *** [Makefile:20: smoke-test] Error 1
```

git bisect...

```
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect start
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect good release
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect bad dev
Bisecting: 51 revisions left to test after this (roughly 6 steps)
[bcce296ca315d81633802183a8ba8658450f7ee3] transform: optimize reflect.Type Implements() method
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect good
Bisecting: 25 revisions left to test after this (roughly 5 steps)
[0b7957d61249ce4b745f78405e36f07245eb8da3] compiler: optimize string literals and globals
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect bad
Bisecting: 12 revisions left to test after this (roughly 4 steps)
[99a41bec4e8c41c0179a56686bdeed896d6aa5a3] transform: fix bug in interface lowering when signatures are renamed
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect good
Bisecting: 6 revisions left to test after this (roughly 3 steps)
[72acda22b0a8d137405e41e9ed54cbfbcce7b26f] machine: refactor PWM support
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect bad
Bisecting: 2 revisions left to test after this (roughly 2 steps)
[896a848001c511a20ad6ef3fd8175cd18b448fe5] builder: add support for -x flag
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect good
Bisecting: 0 revisions left to test after this (roughly 1 step)
[f880950c3efbd005d2e171a5fd1c879a3221367c] ci: limit number of jobs for assert-test-linux
kenneth@kenneth-ubuntu:~/work/tinygo$ go install
kenneth@kenneth-ubuntu:~/work/tinygo$ git bisect good
72acda22b0a8d137405e41e9ed54cbfbcce7b26f is the first bad commit
commit 72acda22b0a8d137405e41e9ed54cbfbcce7b26f
Author: Ayke van Laethem <aykevanlaethem@gmail.com>
Date:   Tue May 19 22:30:46 2020 +0200

    machine: refactor PWM support
    
    This commit refactors PWM support in the machine package to be more
    flexible. The new API can be used to produce tones at a specific
    frequency and control servos in a portable way, by abstracting over
    counter widths and prescalers.
```

kenbell, 2021-04-16:
Affects multiple tests, I commented these out to have `make test` work in `tinygo-drivers`:
```Makefile
#	tinygo build -size short -o ./build/test.hex -target=itsybitsy-m0 ./examples/apa102/itsybitsy-m0/main.go
#	@md5sum ./build/test.hex
#	tinygo build -size short -o ./build/test.hex -target=arduino-nano33 ./examples/l293x/simple/main.go
#	@md5sum ./build/test.hex
#	tinygo build -size short -o ./build/test.hex -target=arduino-nano33 ./examples/l293x/speed/main.go
#	@md5sum ./build/test.hex
#	tinygo build -size short -o ./build/test.hex -target=arduino-nano33 ./examples/l9110x/simple/main.go
#	@md5sum ./build/test.hex
#	tinygo build -size short -o ./build/test.hex -target=arduino-nano33 ./examples/l9110x/speed/main.go
#	@md5sum ./build/test.hex
```

deadprogram, 2021-04-18:
See https://github.com/tinygo-org/drivers/pull/252

deadprogram, 2021-05-06:
Now that https://github.com/tinygo-org/drivers/pull/252 has been merged I am closing this issue. Thanks @kenbell

closed 2021-05-06 by deadprogram