tinygo-org/tinygo #1818: No line number info in backtraces?

dkegel-fastly, 2021-04-17:
Having line numbers would be handy, but they don't seem to be present.  (And showing source code would be even nicer.)
```
$ cat bad.go
package main

func bad(a []int) {
        println(a[0])
}

func main() {
        bad([]int{})
}

$ tinygo build -opt 0 -o foo

$ lldb ./foo
(lldb) target create "./foo"
Current executable set to '/Users/dkegel/gr/foo' (x86_64).
(lldb) run
Process 14639 launched: '/Users/dkegel/gr/foo' (x86_64)
panic: runtime error: index out of range
Process 14639 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff6d30633a libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff6d30633a <+10>: jae    0x7fff6d306344            ; <+20>
    0x7fff6d30633c <+12>: movq   %rax, %rdi
    0x7fff6d30633f <+15>: jmp    0x7fff6d300629            ; cerror_nocancel
    0x7fff6d306344 <+20>: retq
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff6d30633a libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff6d3c2e60 libsystem_pthread.dylib`pthread_kill + 430
    frame #2: 0x00007fff6d28d808 libsystem_c.dylib`abort + 120
    frame #3: 0x00000001000022a4 foo`runtime.runtimePanic + 52
    frame #4: 0x0000000100003032 foo`runtime.lookupPanic + 18
    frame #5: 0x0000000100003470 foo`main.bad + 32
    frame #6: 0x0000000100003496 foo`main.main + 22
    frame #7: 0x0000000100003210 foo`runtime.run$1 + 16
    frame #8: 0x00000001000031eb foo`runtime.run + 11
    frame #9: 0x00000001000031d6 foo`runtime.runMain + 6
    frame #10: 0x00000001000031b7 foo`main + 23
    frame #11: 0x00007fff6d1becc9 libdyld.dylib`start + 1
    frame #12: 0x00007fff6d1becc9 libdyld.dylib`start + 1
(lldb) f 5
frame #5: 0x0000000100003470 foo`main.bad + 32
foo`main.bad:
->  0x100003470 <+32>: ud2    
    0x100003472 <+34>: nopw   %cs:(%rax,%rax)
    0x10000347c <+44>: nopl   (%rax)

foo`main.main:
    0x100003480 <+0>:  pushq  %rax
(lldb) quit
```

This is with tinygo 0.17-ish on mac.

aykevl, 2021-04-17:
TinyGo normally supports debug information, but apparently something is missing here. Weird.

I wonder if something is missing on MacOS? The last time I tried it worked just fine on Linux and other targets (using GDB).

aykevl, 2021-04-18:
It appears to be working on Linux at least:

```
$ tinygo build -o test.elf ./testdata/alias.go
$ lldb-11 ./test.elf
(lldb) target create "./test.elf"
Current executable set to '/home/ayke/src/github.com/tinygo-org/tinygo/test.elf' (x86_64).
(lldb) b runtime.printnl 
Breakpoint 1: where = test.elf`runtime.printnl + 1 [inlined] runtime.putchar at print.go:286, address = 0x0000000000201751
(lldb) run
Process 22850 launched: '/home/ayke/src/github.com/tinygo-org/tinygo/test.elf' (x86_64)
Process 22850 stopped
* thread #1, name = 'test.elf', stop reason = breakpoint 1.1
    frame #0: 0x0000000000201751 test.elf`runtime.printnl [inlined] runtime.putchar(c='\r') at runtime_unix.go:66:10
   63  	}
   64  	
   65  	func putchar(c byte) {
-> 66  		_putchar(int(c))
   67  	}
   68  	
   69  	const asyncScheduler = false
(lldb) bt
* thread #1, name = 'test.elf', stop reason = breakpoint 1.1
  * frame #0: 0x0000000000201751 test.elf`runtime.printnl [inlined] runtime.putchar(c='\r') at runtime_unix.go:66:10
    frame #1: 0x0000000000201751 test.elf`runtime.printnl at print.go:286
    frame #2: 0x0000000000201788 test.elf`runtime.runMain [inlined] main.main at alias.go:29:9
    frame #3: 0x0000000000201774 test.elf`runtime.runMain [inlined] runtime.run$1 at scheduler_any.go:21
    frame #4: 0x0000000000201774 test.elf`runtime.runMain [inlined] runtime.run at scheduler_any.go:18
    frame #5: 0x0000000000201774 test.elf`runtime.runMain at runtime_unix.go:62
    frame #6: 0x000000000020176f test.elf`main at runtime_unix.go:53:9
    frame #7: 0x00007ffff7df909b libc.so.6`__libc_start_main(main=(test.elf`main at runtime_unix.go), argc=1, argv=0x00007fffffffe328, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffe318) at libc-start.c:308:16
    frame #8: 0x000000000020166a test.elf`_start + 42
```

Unfortunately I do not have access to a Mac so I can't test this.

dkegel-fastly, 2021-04-19:
Line numbers do work better on linux, thanks.

aykevl, 2021-04-20:
I think I see why it doesn't work. In short: on MacOS, it seems that debug information is stored in object files and not in the final binary. TinyGo however creates an object file in a temporary directory and removes it after the build, so that object file (with debug information) is gone after compilation. Therefore `lldb` won't be able to find it.

I see two possible workarounds:

* Try installing GDB and use `tinygo gdb <packagename>`. It runs before the temporary directory is removed.
* Try running on a different target, for example `-target=cortex-m-qemu`. This is maybe not what you're after but might work in some cases.

For a proper fix, I intend to eventually move towards a non-LTO or [ThinLTO](https://clang.llvm.org/docs/ThinLTO.html) compilation model which would keep object files in a cache and thus should allow regular debugging.

aykevl, 2022-11-05:
Note to self: it appears to be more complicated than that (https://github.com/tinygo-org/tinygo/issues/2861#issuecomment-1272103141):

> I did some digging to try and figure out why debug information isn't usable on darwin, and my theory is that it will be fixed with https://github.com/tinygo-org/tinygo/issues/2870. I suspect the issue is that something in the toolchain doesn't know how to deal with multiple compile units in a single file, which is exactly what TinyGo is doing.
> 
> Unfortunately, without proper debug information, fixing these bugs is a lot more complicated.

aykevl, 2023-02-26:
Indeed, my suspicion was (almost) correct! Apart from fixing the multiple translation units in a single module, I also had to add `llvm.ident` which is a very small patch. More information: https://github.com/tinygo-org/tinygo/pull/3489#issuecomment-1445492033

awmorgan, 2023-06-05:
@aykevl  @dkegel-fastly I found that you have to run dsymutil to get the debug information created in foo.dSYM directory. 

this is with 0.27.

observe:

```
/Volumes/git/foo $ tinygo version
tinygo version 0.27.0 darwin/amd64 (using go version go1.20.4 and LLVM version 15.0.0)
/Volumes/git/foo $ ls
go.mod  main.go
/Volumes/git/foo $ cat go.mod 
module foo

go 1.20
/Volumes/git/foo $ cat main.go 
package main

func bad(a []int) {
        println(a[0])
}

func main() {
        bad([]int{})
}
/Volumes/git/foo $ tinygo build -opt=0 .
/Volumes/git/foo $ lldb ./foo 
(lldb) target create "./foo"
Current executable set to '/Volumes/git/foo/foo' (x86_64).
(lldb) r
Process 45385 launched: '/Volumes/git/foo/foo' (x86_64)
panic: runtime error: index out of range
Process 45385 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff2090290e <+10>: jae    0x7fff20902918            ; <+20>
    0x7fff20902910 <+12>: movq   %rax, %rdi
    0x7fff20902913 <+15>: jmp    0x7fff208fcab9            ; cerror_nocancel
    0x7fff20902918 <+20>: retq   
Target 0: (foo) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff209315bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20886406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100000e90 foo`runtime.runtimePanic(msg=<unavailable>) + 48 [opt]
    frame #4: 0x0000000100001882 foo`runtime.lookupPanic + 18 [opt]
    frame #5: 0x0000000100001eda foo`main.bad + 26
    frame #6: 0x0000000100001ef9 foo`main.main + 25 [opt]
    frame #7: 0x0000000100001e9b foo`runtime.run$1 + 11 [opt]
    frame #8: 0x0000000100001b46 foo`<goroutine wrapper> + 6 [opt]
    frame #9: 0x00000001000005f1 foo`tinygo_startTask at task_stack_amd64.S:29
(lldb) f 5
frame #5: 0x0000000100001eda foo`main.bad + 26
foo`main.bad:
->  0x100001eda <+26>: ud2    
    0x100001edc <+28>: nopl   (%rax)

foo`main.main:
    0x100001ee0 <+0>:  pushq  %rax
    0x100001ee1 <+1>:  movl   $0x3, %esi
(lldb) q
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
/Volumes/git/foo $ dsymutil  foo
/Volumes/git/foo $ lldb ./foo 
(lldb) target create "./foo"
Current executable set to '/Volumes/git/foo/foo' (x86_64).
(lldb) r
Process 45442 launched: '/Volumes/git/foo/foo' (x86_64)
panic: runtime error: index out of range
Process 45442 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff2090290e <+10>: jae    0x7fff20902918            ; <+20>
    0x7fff20902910 <+12>: movq   %rax, %rdi
    0x7fff20902913 <+15>: jmp    0x7fff208fcab9            ; cerror_nocancel
    0x7fff20902918 <+20>: retq   
Target 0: (foo) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff209315bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20886406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100000e90 foo`runtime.runtimePanic(msg=<unavailable>) at panic.go:59:7 [opt]
    frame #4: 0x0000000100001882 foo`runtime.lookupPanic at panic.go:132:14 [opt]
    frame #5: 0x0000000100001eda foo`main.bad(a=<unavailable>) at main.go:4:11 [opt]
    frame #6: 0x0000000100001ef9 foo`main.main at main.go:8:5 [opt]
    frame #7: 0x0000000100001e9b foo`runtime.run$1 at scheduler_any.go:25:11 [opt]
    frame #8: 0x0000000100001b46 foo`<goroutine wrapper> at scheduler_any.go:23:2 [opt]
    frame #9: 0x00000001000005f1 foo`tinygo_startTask + 6
(lldb) f 5
foo was compiled with optimization - stepping may behave oddly; variables may not be available.
frame #5: 0x0000000100001eda foo`main.bad(a=<unavailable>) at main.go:4:11 [opt]
   1    package main
   2   
   3    func bad(a []int) {
-> 4            println(a[0])
   5    }
   6   
   7    func main() {
(lldb) q
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
/Volumes/git/foo $ 
```

aykevl, 2023-06-05:
Have you tried the dev branch? https://github.com/tinygo-org/tinygo/pull/3491 should improve debug information on Darwin, but it's not included in TinyGo 0.27.0.

awmorgan, 2023-06-05:
identical behavior is observed on the dev branch (commit: 744574193b9e213bc2d897d5e5f02ec1c256ff7a)

```
/Volumes/git/foo $ tinygo version
tinygo version 0.28.0-dev darwin/amd64 (using go version go1.20.4 and LLVM version 15.0.7)
/Volumes/git/foo $ tinygo build -opt=0 .
/Volumes/git/foo $ lldb ./foo
(lldb) target create "./foo"
Current executable set to '/Volumes/git/foo/foo' (x86_64).
(lldb) r
Process 49560 launched: '/Volumes/git/foo/foo' (x86_64)
panic: runtime error at 0x0000000100001d95: index out of range
Process 49560 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff2090290e <+10>: jae    0x7fff20902918            ; <+20>
    0x7fff20902910 <+12>: movq   %rax, %rdi
    0x7fff20902913 <+15>: jmp    0x7fff208fcab9            ; cerror_nocancel
    0x7fff20902918 <+20>: retq   
Target 0: (foo) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff209315bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20886406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100000ca1 foo`runtime.runtimePanicAt(addr=<unavailable>, msg=<unavailable>) + 81 [opt]
    frame #4: 0x00000001000017b7 foo`runtime.lookupPanic + 23 [opt]
    frame #5: 0x0000000100001d9a foo`main.bad + 26
    frame #6: 0x0000000100001db9 foo`main.main + 25 [opt]
    frame #7: 0x0000000100001d5b foo`runtime.run$1 + 11 [opt]
    frame #8: 0x0000000100001a06 foo`<goroutine wrapper> + 6 [opt]
    frame #9: 0x0000000100000641 foo`tinygo_startTask at task_stack_amd64.S:29
(lldb) f 5
frame #5: 0x0000000100001d9a foo`main.bad + 26
foo`main.bad:
->  0x100001d9a <+26>: ud2    
    0x100001d9c <+28>: nopl   (%rax)

foo`main.main:
    0x100001da0 <+0>:  pushq  %rax
    0x100001da1 <+1>:  movl   $0x3, %esi
(lldb) q
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
/Volumes/git/foo $ dsymutil  foo
/Volumes/git/foo $ lldb ./foo
(lldb) target create "./foo"
Current executable set to '/Volumes/git/foo/foo' (x86_64).
(lldb) r
Process 49599 launched: '/Volumes/git/foo/foo' (x86_64)
panic: runtime error at 0x0000000100001d95: index out of range
Process 49599 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff2090290e <+10>: jae    0x7fff20902918            ; <+20>
    0x7fff20902910 <+12>: movq   %rax, %rdi
    0x7fff20902913 <+15>: jmp    0x7fff208fcab9            ; cerror_nocancel
    0x7fff20902918 <+20>: retq   
Target 0: (foo) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff209315bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20886406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100000ca1 foo`runtime.runtimePanicAt(addr=<unavailable>, msg=<unavailable>) at panic.go:71:7 [opt]
    frame #4: 0x00000001000017b7 foo`runtime.lookupPanic at panic.go:144:16 [opt]
    frame #5: 0x0000000100001d9a foo`main.bad(a=<unavailable>) at main.go:4:11 [opt]
    frame #6: 0x0000000100001db9 foo`main.main at main.go:8:5 [opt]
    frame #7: 0x0000000100001d5b foo`runtime.run$1 at scheduler_any.go:25:11 [opt]
    frame #8: 0x0000000100001a06 foo`<goroutine wrapper> at scheduler_any.go:23:2 [opt]
    frame #9: 0x0000000100000641 foo`tinygo_startTask + 6
(lldb) f 5
foo was compiled with optimization - stepping may behave oddly; variables may not be available.
frame #5: 0x0000000100001d9a foo`main.bad(a=<unavailable>) at main.go:4:11 [opt]
   1    package main
   2   
   3    func bad(a []int) {
-> 4            println(a[0])
   5    }
   6   
   7    func main() {
(lldb) q
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
/Volumes/git/foo $ 
```

aykevl, 2023-06-06:
Is it any better if you use `tinygo gdb` or `tinygo lldb`?

aykevl, 2023-06-07:
I assume this will eventually be solved once we implement real ThinLTO support (instead of our custom LTO setup). See https://github.com/tinygo-org/tinygo/pull/3489 (it will take a while until this is ready for merge).

davecheney, 2026-10-28:
Retested on current TinyGo (0.44.0-dev-eae3f875) on macOS ARM64. LLDB resolves Go function names in the panic backtrace (`main.bad`, `main.main`) but shows no Go source file/line locations; the runtime panic reports only an address. The issue still reproduces.