tinygo-org/tinygo #1825: Can't print local variables, they're always "optimized out"?

dkegel-fastly, 2021-04-19:
On linux on amd64, gdb can do ```info variables``` to see globals, but ```info locals``` usually lists all the variables as optimized out.

This is the case even with 

 tinygo -opt 1

I can't go lower; opt 0 fails with

```
panic: unexpected getFuncPtrCall

goroutine 12 [running]:
github.com/tinygo-org/tinygo/transform.LowerFuncValues(0x7ff9b0002ac0)
	/home/dank/src/tinygo/transform/func-lowering.go:191 +0x11d4
github.com/tinygo-org/tinygo/transform.Optimize(0x7ff9b0002ac0, 0xc0002be0f0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/dank/src/tinygo/transform/optimizer.go:110 +0x1258
github.com/tinygo-org/tinygo/builder.optimizeProgram(0x7ff9b0002ac0, 0xc0002be0f0, 0xc00114f1d0, 0x2)
	/home/dank/src/tinygo/builder/build.go:603 +0x122
github.com/tinygo-org/tinygo/builder.Build.func2(0x0, 0x0)
	/home/dank/src/tinygo/builder/build.go:304 +0x812
github.com/tinygo-org/tinygo/builder.jobWorker(0xc00065b260, 0xc00065b200)
	/home/dank/src/tinygo/builder/jobs.go:153 +0xe2
```

(This is with dev from mar 29, 35bf0746a178c, to work around the crash https://github.com/tinygo-org/tinygo/issues/1805 .)

MikeJansen, 2024-01-15:
I'm running into the same while debugging Raspberry Pi Pico on an Ubuntu 22.04 host.  Combining this with not being able to figure out how to redirect stdout to UART (instead of USB) when using a picoprobe, I'm finding debugging to be more difficult.

deadprogram, 2024-01-17:
@MikeJansen you possibly want the `-serial` flag:

```
  -serial string
        which serial output to use (none, uart, usb)
```

Beware that this can have side effects like making it required to use debugger to reflash the board if you replace USB serial output support.

aykevl, 2024-01-20:
> Beware that this can have side effects like making it required to use debugger to reflash the board if you replace USB serial output support.

You may need to enter the bootloader manually using double tapping or by holding the boot button while resetting, but I don't think you'd need a hardware debugger? The rp2040 chips have the bootloader in hardware.

maks, 2025-03-27:
I was also seeing this using tinygo `0.37.0` but on off chance that something had changed I tried using `-opt 0` and it worked for me with the basic counter example writing to serial and instead of <optimised out> in VSCode I can now see the count variable value and place breakpoints where previously I wasn't able to.

Also it might be useful to add some documentation on how newcomers to tinygo who want to use VSCode could set it up to work with a gdb as a debugger given that the PR from @sago35  https://github.com/tinygo-org/vscode-tinygo/pull/4 is still WIP so I used the information in that PR and the **super** useful gist from @sago35  https://gist.github.com/sago35/fd1b23d4ac9ac689ae84dee0be8c13b1 to figure out how to set it up for myself.

davecheney, 2026-10-28:
Tested with GDB 18.1 on Linux/AMD64 using the current TinyGo checkout. At -opt=1, GDB stopped at main.check and showed input=5, but first and second as <optimized out>. At -opt=0, stopping in runtime.printint64 and moving up to main.check showed second=30; first and input remained optimized out. Running the -opt=0 binary printed 30. This confirms GDB can recover a live local value at -opt=0; the original report used -opt=1, where optimized locals are expected.

closed 2026-10-28 by davecheney