tinygo-org/tinygo #1826: Segfault after sync.Once ?

dkegel-fastly, 2021-04-20:
A project I'm running with tinygo seems to be triggering a crash on the two platforms I've tried (amd64, both linux and mac).

It seems to occur in both 0.17 (prebuilt or self-built) and dev (though I can't quite say for certain my tests are valid - I am not 100% sure I'm cleaning up properly between runs).

As I minimized the test case, the crash wiggled around between sigsegv, sigabort, and null pointer panics.

It appears to happen shortly after a call to sync.Once's Do method, and commenting that out (or using an init function instead) prevents the crash.  It may also require nested functions.

Oddly, in a quick check just now, replacing the testcase with main() also seemed to keep the crash from happening.

Running ```tinygo test``` isn't enough to see the crash; you have to do ```tinygo test -c``` and run the binary yourself.

Here's a test case.  It could probably be reduced further (not sure if goreduce is up to it, I haven't tried it on this hand-reduced file yet).
```
package bug

import (
	"sync"
	"testing"
)

type Foo struct { }
type FooGen func() Foo
type FooGenListHolder struct {
	FooGens     []FooGen
}
type FooGenListHolderMaker func() *FooGenListHolder

func UseFooGenListHolderMakers(fooGenListHolderMakers ...FooGenListHolderMaker) {
	for _, fooGenListHolderMaker := range fooGenListHolderMakers {
		fooGenListHolderMaker()
	}
}
func GenerateAndUseFooGenListHolderMakers() {
	myFooGen := func() (Foo) {
		return Foo{}
	}
	fooGenListHolderMaker := func() *FooGenListHolder {
		println("Commenting out badnessOnce.Do(initBadness) avoids crash")
		badnessOnce.Do(initBadness)

		var fooGens []FooGen
		fooGens = append(fooGens, myFooGen)

		fooGenListHolder := FooGenListHolder{}
		return &fooGenListHolder
	}
	UseFooGenListHolderMakers(fooGenListHolderMaker, fooGenListHolderMaker)
}
func TestBug(t *testing.T) {
	GenerateAndUseFooGenListHolderMakers()
}

var badnessOnce sync.Once

func initBadness() {
}
```

Here's an example stack dump:
```
=== RUN   TestBug

Program received signal SIGSEGV, Segmentation fault.
bug.GenerateAndUseFooGenListHolderMakers$2 () at /home/dank/bug/bug_test.go:29
29			fooGens = append(fooGens, myFooGen)
(gdb) #0  bug.GenerateAndUseFooGenListHolderMakers$2 () at /home/dank/bug/bug_test.go:29
#1  0x0000000000402dfc in (*internal/task.Task).Resume (t=<optimized out>)
    at /home/dank/src/tinygo/src/internal/task/task_coroutine.go:23
#2  0x0000000000404382 in runtime.scheduler () at /home/dank/src/tinygo/src/runtime/scheduler.go:165
``` 

Here's my test script on linux:
```
#!/bin/sh
rm -rf log *.test ~/.caache/tinygo
tinygo test -c -opt 1
gdb ./*.test < $HOME/bug/cmds > log
if grep -q 'SIGSEGV' log && grep -q '#0 .*GenerateAndUseFooGenListHolderMakers$2' log
then
	echo PRESENT
else
	echo ABSENT
fi
```

where ~/bug/cmds is just
```
run
bt
quit
```

aykevl, 2021-04-20:
This sounds a lot like a memory corruption.

Can you try `-gc=conservative`? I've seen crashes before with the default (extalloc) GC and this might be another instance of it.

dkegel-fastly, 2021-04-20:
Update:
- sometimes you have to run the binary a few times before you see the crash
- the crash does not happen with -scheduler=none
- the test does not compile with -scheduler=tasks
- the crash does not seem to happen with -gc=conservative

aykevl, 2021-04-20:
Interesting! In that case, I suspect that `-gc=extalloc` (the default on Linux and MacOS) is to blame.

@niaow can you maybe take a look?

deadprogram, 2021-05-09:
Now that #1860 has been merged, can we close this issue?

niaow, 2021-05-09:
Huh, this is a different bug than the one I found.

niaow, 2021-05-09:
Interestingly, turning gcDebug on causes this to pass.

niaow, 2021-05-09:
Ok so specifically: observing the allocated pointer inside of `runtime.alloc` appears to change the codegen in some meaningful way.

niaow, 2021-05-09:
Or maybe it could just be placing the pointer on the stack I guess?

niaow, 2021-05-16:
Ok so I have narrowed this down a bit: when using coroutines, something is not being tracked properly. This issue can actually be replicated with the conservative collector too by forcing a GC on every allocation and zeroing freed blocks. Similar to #1245 , the extalloc collector may be triggering it more often due to more GC cycles and memory actually being quickly overwritten after freeing.

dkegel-fastly, 2021-07-29:
This is still happening...

dkegel-fastly, 2021-07-29:
The reduced test case seems to be passing now with

This is still happening with an up to date tinygo:
```
commit e834d7887143fc8da542d3d29907e8db7a5b9bb2 (HEAD -> dev, origin/dev)
Author: Federico G. Schwindt <fgsch@lodoss.net>
Date:   Thu Jul 15 19:14:49 2021 +0100
```

(Unfortunately, my larger app still seems to crash without -scheduler=none.  I guess I'll try generating a new minimal testcase.)

fgsch, 2021-07-29:
This still crashes under wasi and tinygo 0.19.0.

peterbourgon, 2021-07-29:
@niaow 

> Ok so I have narrowed this down a bit: when using coroutines, something is not being tracked properly

Just some additional signal on this — we've been experiencing memory corruption issues with `-scheduler=coroutines` since TinyGo v0.18.0. They appear to be loosely correlated to the number of allocations, seem to happen more under `-gc=conservative` than `-gc=leaking`, and _really_ get triggered if allocate a map type at any point. We've so far been unable to produce a reliable reproducer that can be shared here :(

niaow, 2021-07-29:
So there are a few things:
1. There are some bugs when using stack slots GC where things are not tracked properly, and stack slots GC is mostly required when we can't scan the stack (aka wasm).
2. There are some bugs in extalloc which I started to fix but did not finish. We have switched over to conservative where possible. (this does not affect wasm)
3. The coroutines system is really really complicated. I started working on support for asyncify to hopefully eventually replace it.

peterbourgon, 2021-07-29:
We would be happy to use something other than coroutines, but AFAIK it is the only concurrent scheduler that works in Wasm/Wasi-land.

niaow, 2021-12-14:
Right now with the default `conservative` collector and `tasks`, this no longer seems to replicate. I think this may be effectively fixed?

dkegel-fastly, 2021-12-14:
Running our large app with conservative now crashes differently.   It may be a different crash, will investigate.

dkegel-fastly, 2021-12-15:
OK, running with targets/wasi.json's default-stack-size bumped to 10000 fixed the crash.

Closing.

closed 2021-12-15 by dkegel-fastly