tinygo-org/tinygo #1830: Using the regexp package gives either a stack overlow or "panic: expected exactly one index"

arp242, 2021-04-23:
With the following:

```
package main

import "regexp"

var reTest = regexp.MustCompile(`^\w[\w -]+:$`)

func main() {}
```

I get:

```
panic: expected exactly one index

goroutine 6 [running]:
github.com/tinygo-org/tinygo/interp.(*runner).runAtRuntime(0xc0000c1320, 0xc001271900, 0x36, 0x53, 0xc0021352c0, 0x3, 0x3, 0x1f0c0a0, 0xc000344108, 0x11, ...)
	/data/src/tinygo/interp/interpreter.go:940 +0xd25
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc001271900, 0xc002d9c610, 0x8, 0x1f, 0xc0012591d0, 0xc0003441f8, 0x18, 0x0, 0xc002a972c0, ...)
	/data/src/tinygo/interp/interpreter.go:100 +0xa85a
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc0012716d0, 0xc002a40e10, 0x5, 0x7, 0xc000bbfec0, 0xc000e23668, 0x14, 0x0, 0xc002f9d660, ...)
	/data/src/tinygo/interp/interpreter.go:481 +0x6473
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc001271680, 0xc0029fc010, 0x6, 0x7, 0xc0008de420, 0xc00281bf90, 0x10, 0x0, 0xc002872a10, ...)
	/data/src/tinygo/interp/interpreter.go:481 +0x6473
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc001271630, 0xc002801e90, 0x4, 0x7, 0xc0008bf6e0, 0xc0027d4f60, 0xc, 0x0, 0x3ab7f900, ...)
	/data/src/tinygo/interp/interpreter.go:481 +0x6473
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc001270c30, 0xc002801910, 0x4, 0x7, 0xc0008b7200, 0xc0027d4780, 0x8, 0x0, 0x200000003, ...)
	/data/src/tinygo/interp/interpreter.go:481 +0x6473
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0000c1320, 0xc001270b90, 0x0, 0x0, 0x0, 0x0, 0x80f89a, 0x4, 0x0, 0x0, ...)
	/data/src/tinygo/interp/interpreter.go:481 +0x6473
github.com/tinygo-org/tinygo/interp.Run(0xb11710, 0xc00004cd00, 0x0, 0x0)
	/data/src/tinygo/interp/interp.go:112 +0x69f
github.com/tinygo-org/tinygo/builder.optimizeProgram(0xb11710, 0xc00025e0c0, 0xc000e627a0, 0x2)
	/data/src/tinygo/builder/build.go:649 +0x3f
github.com/tinygo-org/tinygo/builder.Build.func2(0xc001c376e0, 0x0, 0x0)
	/data/src/tinygo/builder/build.go:384 +0x7c6
github.com/tinygo-org/tinygo/builder.jobWorker(0xc000bcf680, 0xc000bcf620)
	/data/src/tinygo/builder/jobs.go:165 +0xe2
created by github.com/tinygo-org/tinygo/builder.runJobs
	/data/src/tinygo/builder/jobs.go:84 +0xc5
```

I attempted to reduce the complexity of the regexp to isolate the problem, but with this:

    var reTest = regexp.MustCompile("0")

I get:

```
runtime: goroutine stack exceeds 1000000000-byte limit
runtime: sp=0xc024280380 stack=[0xc024280000, 0xc044280000]
fatal error: stack overflow

runtime stack:
runtime.throw(0x8147dd, 0xe)
	/usr/lib/go/src/runtime/panic.go:1117 +0x72
runtime.newstack()
	/usr/lib/go/src/runtime/stack.go:1069 +0x7ed
runtime.morestack()
	/usr/lib/go/src/runtime/asm_amd64.s:458 +0x8f

goroutine 53 [running]:
runtime.convT64(0x7f23f0002940, 0x0)
	/usr/lib/go/src/runtime/iface.go:360 +0x79 fp=0xc024280390 sp=0xc024280388 pc=0x429779
tinygo.org/x/go-llvm.Module.Context.func1(0x7f23f0002940, 0x42fbc2)
	/home/martin/.cache/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:512 +0x2b fp=0xc0242803c0 sp=0xc024280390 pc=0x688e4b
tinygo.org/x/go-llvm.Module.Context(0x7f23f0002940, 0xc0242804b0)
	/home/martin/.cache/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:512 +0x2b fp=0xc0242803e0 sp=0xc0242803c0 pc=0x67b40b
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc0002bb880, 0x70, 0x70, 0xc0031fb5f0, 0x70, 0xa, 0x884ec0)
	/data/src/tinygo/interp/memory.go:766 +0x5d fp=0xc0242804c0 sp=0xc0242803e0 pc=0x70d03d

[.. trim ..]
```

Full error over here, as it's a bit long: https://gist.github.com/arp242/eb624496c58ea151daa71009cace3d61

    $ tinygo version
    tinygo version 0.18.0-dev linux/amd64 (using go version go1.16.3 and LLVM version 11.0.0)

(this is commit 80caf2d)

arp242, 2021-04-23:
I appreciate that something like the regexp package might not be supported in TinyGo btw, I just wanted to see if my little CLI program works with TinyGo now that os.Args is supported. But panics and stack overflows are not the best way to communicate that 😅

fgsch, 2021-05-03:
Does this work not using globals? See #1306 .

aykevl, 2021-05-03:
This panic is probably solved with #1850.

You will probably run into #1848 after that (the stack overflow). To fix that, you could move the `regexp.MustCompile` out of init code, for example into the `main` function.

> But panics and stack overflows are not the best way to communicate that :sweat_smile:

Well yeah, but these behaviors weren't exactly intended behavior.

deadprogram, 2021-05-04:
So the panic should now be solved in the `dev` branch with #1850 merged, but we should leave open to address the stack overflow part.

dgryski, 2023-01-31:
This particular issue seems to be solved somewhat.
```
~/go/src/github.com/dgryski/bug $ cat bug.go
package main

import "regexp"

var reTest = regexp.MustCompile(`^\w[\w -]+:$`)

func main() {
	println(reTest.MatchString("foobar"))
}
~/go/src/github.com/dgryski/bug $ go run bug.go
false
~/go/src/github.com/dgryski/bug $ tinygo run -target=wasi bug.go
false
~/go/src/github.com/dgryski/bug $ tinygo run bug.go
false
```

I agree a more generic "interp is slow" bug would be useful, but I think we have enough of those already.

closed 2023-02-01 by dgryski