tinygo-org/tinygo #1831: Panic when building/testing a package without any code

arp242, 2021-04-23:
```
$ cat *.go
// mylib is documented here.
package mylib

$ tinygo test .
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x6c2b65]

goroutine 21 [running]:
github.com/tinygo-org/tinygo/compiler.CompilePackage(0xc000262640, 0xb, 0xc00021f2c0, 0xc0000f5c20, 0x12398a0, 0xc00023c0b0, 0x0, 0x0, 0x0, 0x0, ...)
        /data/src/tinygo/compiler/compiler.go:253 +0x105
github.com/tinygo-org/tinygo/builder.Build.func1(0xc0000f5d40, 0x0, 0x0)
        /data/src/tinygo/builder/build.go:199 +0x128
github.com/tinygo-org/tinygo/builder.jobWorker(0xc000282120, 0xc0002820c0)
        /data/src/tinygo/builder/jobs.go:165 +0xe2
created by github.com/tinygo-org/tinygo/builder.runJobs
        /data/src/tinygo/builder/jobs.go:84 +0xc5
```

Same results for `tinygo build`.

Trying to build/test this is a bit silly of course; the case where I ran in to this is that I have a library where all the actual code is in subpackages, with just a `doc.go` and `go.mod` at the top-level. Still, it shouldn't panic.

Looking at `CompilePackage()`, it's `ssaPkg.Prog.ImportedPackage("runtime")` that returns `nil`; I suppose because it doesn't have any code, it doesn't import `runtime`. I'm not sure if adding a `!= nil` check here is the best way to fix it though.

    $ tinygo version
    tinygo version 0.18.0-dev linux/amd64 (using go version go1.16.3 and LLVM version 11.0.0)

(this is commit 80caf2d)

FrankReh, 2021-04-25:
I ran into this too. My dev branch is a few days old though so couldn't say with a latest checkout.

While starting a new project with an empty main. Found I can import something like '_ "time"' and that satisfies whatever build assumptions are being made now. It was not enough to import _ "runtime".

deadprogram, 2021-05-06:
This looks like is has been reported before in https://github.com/tinygo-org/tinygo/issues/186 but was closed too soon.

aykevl, 2021-05-06:
No, #186 is a different issue when the `main` function is missing in the main package. This issue is about `tinygo test`.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

PR #1865 (2021-05-06): main: match `go test` output
This commit makes the output of `tinygo test` similar to that of `go test`. It changes the following things in the process:

  * Running multiple tests in a single command is now possible. They aren't paralellized yet.
  * Packages with no test files won't crash TinyGo, instead it logs it in the same way the Go toolchain does.

This fixes #1831.
closed 2021-05-12 by deadprogram