tinygo-org/tinygo #1839: "unable to locate a serial port" error because of missing "/dev/ttyACM0", even when I don't use it

jypelle, 2021-04-25:
With tinygo 0.17.0 on ubuntu 20.04, If I want to flash my arduino-nano, I should execute as a workaround:

```
cd /dev
sudo ln -s ttyUSB0 ttyACM0
```

Before flashing with

```
tinygo flash -target=arduino-nano ./cmd/myawesomeproject --port=/dev/ttyUSB0
```

Without the workaround, I got a "unable to locate a serial port" error message because tinygo always check the presence of at least one "/dev/ttyACM*" even if it doesn't use it.

gfeun, 2021-04-25:
Hey @jypelle,

Try passing your command parameters before you input and use only one dash.

Like so: `tinygo flash -target=arduino-nano -port=/dev/ttyUSB0 ./cmd/myawesomeproject`

jypelle, 2021-04-25:
Indeed, I should have reread myself, It works as expected

closed 2021-04-25 by jypelle