tinygo-org/tinygo #1843: avr: interfaces can't be compared

soypat, 2021-04-27:
# Problem
Interfaces are not comparable. This problem is aggravated by the fact errors usually need to be compared to control program flow, i.e. io.EOF.

The following program yields an error when flashing/building
```go
package main

type A uint8

func (a A) Error() string {
	return "got error"
}
func main() {
	var a A = 12
	var b error = a
	if b == a {
		println("oh no, an error!")
	}
}
```

*Output*
```log
$ tinygo build -o=.bin -target=arduino-nano enc28j60/test/main.go
/tmp/tinygo304709431/main.o: In function `(reflect.Value).Complex':
/usr/local/lib/tinygo/src/reflect/value.go:271: undefined reference to `__extendsfdf2'
/usr/local/lib/tinygo/src/reflect/value.go:271: undefined reference to `__extendsfdf2'
/tmp/tinygo304709431/main.o: In function `(reflect.Value).Float':
/usr/local/lib/tinygo/src/reflect/value.go:245: undefined reference to `__extendsfdf2'
/tmp/tinygo304709431/main.o: In function `runtime.reflectValueEqual':
/usr/local/lib/tinygo/src/runtime/interface.go:55: undefined reference to `__eqdf2'
/usr/local/lib/tinygo/src/runtime/interface.go:55: undefined reference to `__eqdf2'
/usr/local/lib/tinygo/src/runtime/interface.go:53: undefined reference to `__eqdf2'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo304709431/main: exit status 1
```

Below is a MWE of the error, same error message too:

```go
package main
type Aer interface {
	abc()
}
type A uint8

func (a A) abc() {}

func main() {
	var a Aer = A(12)
	var b Aer = a
	if b == a {
		println("oh no, an error!")
	}
}
```

soypat, 2023-09-05:
@aykevl I believe this is fixed, right?

aykevl, 2025-09-30:
Yes this should be fixed, please reopen if it isn't.

closed 2025-09-30 by aykevl