tinygo-org/tinygo #1847: Goland + TinyGo + Modules

adrianbn, 2021-05-01:
Hi,

I've been trying to set up a dev environment using TinyGo, Goland (or IntelliJ Pro), and go modules. I manage to get the environment working without making use of modules with workarounds as those described here #927. There is mention of support for modules in #107 but not a clear setup guide. Is there a supported way to do go modules in Goland IDE as of today? 

Cheers!

sago35, 2021-05-01:
The latest information can be found below.
For `v0.16.0` and later, only the `GOROOT` and `GOFLAGS` settings are required.

* https://tinygo.org/ide-integration/
* https://tinygo.org/ide-integration/#other-ides

I think many parts of the following articles are still useful today.
The difference between then and now is that you need to configure `GOROOT`.
`GOROOT` can be found by running `tinygo env`.

I use vim, so I don't know much about GoLand.
It would be great if you could write a document on how to set up GoLand!

* https://artspb.me/posts/working-with-tinygo-in-goland/

adrianbn, 2021-05-02:
Hey @sago35 ,

Thanks for chiming in. Unfortunately I tried that and it didn't work. If I set `GOROOT` to Go (not tinygo), the machine package won't resolve. But if I try to set it to tinygo (e.g. `C:\Users\adrian\scoop\apps\tinygo\current`) then I get the following error from Goland:

```The selected directory is not a valid home for Go SDK```. See image below:
[image]

Any idea why tinygo's location is not a valid go sdk but regular go's is? They're both installed with `scoop`.

Cheers!

adrianbn, 2021-05-02:
Hey @sago35,

Thanks for chiming in. I figured out what the issue is. The correct value for `GOROOT` in IntelliJ / Goland is the `cached GOROOT` in the output of `tinygo info -target <your_target>`, not the `GOROOT` value you see when you run `tinygo env`! See image below:

[image]

This issue can be closed!
Thanks!

closed 2021-05-02 by adrianbn

adrianbn, 2021-05-02:
Actually, an update here. When using the cached `GOROOT` Goland seems to think go version is 0.1 and can't build anything with modules. I assume this is a bug with Goland / IntelliJ so I'll bring it up with them. See error below:
```
tinygo flash -target=arduino main.go
error: requires go version 1.11 through 1.16, got go0.1
make: *** [C:/Users/adrian/IdeaProjects/tinygo-with-modules/Makefile:4: flash] Error 1
```

sago35, 2021-05-02:
@adrianbn 

I'm sorry for the wrong information about `GOROOT`.
The information you wrote is correct.

> Actually, an update here. When using the cached GOROOT Goland seems to think go version is 0.1 and can't build anything with modules. I assume this is a bug with Goland / IntelliJ so I'll bring it up with them. See error below:

For `gopls`, you need to set `GOROOT`.
However, you should not set `GOROOT` to `cached GOROOT` when running `tinygo build/flash`.
For example, if you run `set GOROOT=` and then run `tinygo flash`, it should work fine.

aykevl, 2021-05-02:
Indeed, it's the "cached GOROOT" that should be used in IDEs (but note that the value can sometimes change, with Go and TinyGo updates).

> Actually, an update here. When using the cached `GOROOT` Goland seems to think go version is 0.1 and can't build anything with modules.

I suspect the file src/runtime/internal/sys/zversion.go is to blame. Can you try bumping the version in there to some newer Go version?
(A proper fix would need to do something different, but it's a quick hack to test whether that is the issue).

adrianbn, 2021-05-02:
Hey @aykevl,

That is indeed a hack that clears that error. Not sure why that file contains such a weird version value however.

There is yet another issue after that one is fixed:

```
tinygo build -target=arduino -o main main.go
..\..\AppData\Local\tinygo\goroot-go1.16.3-81b2f4ccc9969612f6e9e2ef99ecd6e517427
8a43f0295671d7e54b77f92e29c-syscall\src\device\avr\atmega328p.go:12:2: cannot fi
nd package "unsafe" in any of:
        C:\Users\adrian\AppData\Local\tinygo\goroot-go1.16.3-81b2f4ccc9969612f6e
9e2ef99ecd6e5174278a43f0295671d7e54b77f92e29c-syscall\src\unsafe (from $GOROOT)
        C:\Users\adrian\go\src\unsafe (from $GOPATH)
```

Same error shows up for other packages such as `io` or `time`. From a shell the project builds fine using the same command though, but the shell environment points to the original `GOROOT`, not the cached one. Any idea what's going on there?

Cheers!

P.S. IntelliJ / Goland are working on a tinygo plugin that should make this a lot easier.

aykevl, 2021-05-02:
Good to know! Then this file zversion.go might need to be generated.

> There is yet another issue after that one is fixed:

Weird, just to be sure you could try running `tinygo clean` which clears the cache and might resolve any issues.
If that doesn't solve the issue, can you verify that `C:\Users\adrian\scoop\apps\go\current` exists and contains the `src\time` package?

adrianbn, 2021-05-02:
That was indeed a cache problem. Not sure what happened there but `tinygo clean` solved it. Thanks for all the help!.

artspb, 2021-05-03:
Just in case, since Go 1.17 the version is in `internal/buildcfg/zbootstrap.go`. GoLand supports both the old and the new file (and a few intermediate, actually).

@aykevl Could you please shed a light on why an IDE needs to use `cached GOROOT`? What's the difference with a regular `GOROOT` and what could happen if one uses it instead? Thanks.