tinygo-org/tinygo #1848: stack overflow building wasm

mayocream, 2021-05-01:
```bash
runtime: goroutine stack exceeds 1000000000-byte limit
runtime: sp=0xc0353003b8 stack=[0xc035300000, 0xc055300000]
fatal error: stack overflow

runtime stack:
runtime.throw(0x4e8ceeb, 0xe)
        /usr/local/go/src/runtime/panic.go:1116 +0x72
runtime.newstack()
        /usr/local/go/src/runtime/stack.go:1042 +0x6ce
runtime.morestack()
        /usr/local/go/src/runtime/asm_amd64.s:449 +0x8f

goroutine 12 [running]:
runtime.heapBitsSetType(0xc014c05138, 0x8, 0x8, 0x4e862e0)
        /usr/local/go/src/runtime/mbitmap.go:947 +0xa4d fp=0xc0353003c8 sp=0xc0353003c0 pc=0xae882d
runtime.mallocgc(0x8, 0x4e862e0, 0x1, 0x0)
        /usr/local/go/src/runtime/malloc.go:1066 +0x53e fp=0xc035300468 sp=0xc0353003c8 pc=0xadea0e
runtime.growslice(0x4e862e0, 0x0, 0x0, 0x0, 0x1, 0x0, 0x0, 0xc016c5e300)
        /usr/local/go/src/runtime/slice.go:181 +0x1e2 fp=0xc0353004d0 sp=0xc035300468 pc=0xb1c582
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1061 +0x3e1 fp=0xc0353005b0 sp=0xc0353004d0 pc=0xe35191
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc0353006f8 sp=0xc0353005b0 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc0353007d8 sp=0xc0353006f8 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035300920 sp=0xc0353007d8 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035300a00 sp=0xc035300920 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035300b48 sp=0xc035300a00 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035300c28 sp=0xc035300b48 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035300d70 sp=0xc035300c28 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035300e50 sp=0xc035300d70 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035300f98 sp=0xc035300e50 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035301078 sp=0xc035300f98 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc0353011c0 sp=0xc035301078 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc0353012a0 sp=0xc0353011c0 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc0353013e8 sp=0xc0353012a0 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc0353014c8 sp=0xc0353013e8 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035301610 sp=0xc0353014c8 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc0353016f0 sp=0xc035301610 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035301838 sp=0xc0353016f0 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035301918 sp=0xc035301838 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035301a60 sp=0xc035301918 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035301b40 sp=0xc035301a60 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035301c88 sp=0xc035301b40 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035301d68 sp=0xc035301c88 pc=0xe34f23
github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue(0x4e6f00000040, 0x0, 0xc016c9adb0, 0x7f5650011d90)
        /home/circleci/project/interp/memory.go:521 +0x742 fp=0xc035301eb0 sp=0xc035301d68 pc=0xe32002
github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue(0xc016cad500, 0x70, 0x70, 0xc016c9adb0, 0x70)
        /home/circleci/project/interp/memory.go:1045 +0x173 fp=0xc035301f90 sp=0xc035301eb0 pc=0xe34f23
[... 168 lines cut ...]
        /home/circleci/project/builder/jobs.go:73 +0xc5

goroutine 11 [chan receive]:
github.com/tinygo-org/tinygo/builder.jobWorker(0xc000f5e780, 0xc000f5e720)
        /home/circleci/project/builder/jobs.go:151 +0x91
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:73 +0xc5
```

test code:
```go
package main

import (
	_ "github.com/yuin/goldmark/parser"
)

func main() {
	println("test")
}
```

```bash
tinygo build -o md.wasm -target wasm -no-debug main.go
```

aykevl, 2021-05-02:
I can confirm this issue. In the dev branch, there is also a different bug that triggers before the infinite recursion in this issue which I have a fix for.

dkegel-fastly, 2021-06-25:
This is still happening here on the dev branch (v0.18.0-78-ge02f308), and is blocking compilation of a real app.

dkegel-fastly, 2021-06-25:
Another test case, boiled down from the app I was trying to build:
```
package main

import ac "github.com/cloudflare/ahocorasick"

var foo = ac.NewStringMatcher([]string{
	"x",
})

func main() {
	println("test")
}
```

aykevl, 2021-06-25:
If I had to guess it's this line that is causing trouble:

https://github.com/yuin/goldmark/blob/master/util/util.go#L765

```go
var emailDomainRegexp = regexp.MustCompile(`^[a-zA-Z0-9](?:[a-zA-Z0-9-]{0,61}[a-zA-Z0-9])?(?:\.[a-zA-Z0-9](?:[a-zA-Z0-9-]{0,61}[a-zA-Z0-9])?)*`)
```

And in the case of github.com/cloudflare/ahocorasick, a quick glance at the code of `NewStringMatcher` seems to suggest it indeed creates some sort of tree-like structure.

aykevl, 2021-06-25:
Right now, fixing this is a bit complicated because of the special casing in here:

https://github.com/tinygo-org/tinygo/blob/164632616496e21f5f11098316c650ab5ad9356c/interp/memory.go#L793-L801

But perhaps it's still possible without much more refactoring.

dkegel-fastly, 2021-06-25:
Thanks for the update!

Fortunately, I was able to pin an earlier version of a dependency in my app and avoid the explosion for now, but it's tech debt we'd like to  address sooner or later.

aykevl, 2021-07-09:
I have a fix for this bug that I will submit shortly. It's relatively big.

aykevl, 2021-07-15:
Finishing this fix took a little bit longer than expected. It definitely works, I just need to polish it and add tests etc. so it's reviewable.

aykevl, 2021-07-15:
The PR #2001 fixes the test case in https://github.com/tinygo-org/tinygo/issues/1848#issuecomment-868732147 but sadly doesn't yet fix the issue in the issue description.

dkegel-fastly, 2021-07-23:
Tested PR #2001, verified that it fixes the app I boiled down the ahocorasick test case from.

nirui, 2021-10-03:
Oopie! I don't think this is over just yet :Z

Here is my program:

	$ cat main.go 
	package main

	import (
		"fmt"
		"regexp"
	)

	var (
		nameFormatRegExp = regexp.MustCompile(
			`^([[:alnum:] ]+) ` +
				`([[:digit:]]+\.[[:digit:]]+\.[[:digit:]]+(\-[[:alnum:]]+){0,1}) ` +
				`\(C\) ([[:alnum:] ]+) (\<([[:alnum:]\@\.\_\-]+)\> ){0,1}` +
				`([[:digit:]]+)$`)
	)

	func main() {
		fmt.Println(nameFormatRegExp.String())
	}

Go runs as normal:

	$ go run main.go 
	^([[:alnum:] ]+) ([[:digit:]]+\.[[:digit:]]+\.[[:digit:]]+(\-[[:alnum:]]+){0,1}) \(C\) ([[:alnum:] ]+) (\<([[:alnum:]\@\.\_\-]+)\> ){0,1}([[:digit:]]+)$

But TinyGo does not:

	$ (export $(go env | grep "GOROOT" | tr -d '"'); tinygo version; tinygo run main.go)
	tinygo version 0.20.0 linux/amd64 (using go version go1.17.1 and LLVM version 11.0.0)
	runtime: goroutine stack exceeds 1000000000-byte limit
	runtime: sp=0xc025a00398 stack=[0xc025a00000, 0xc045a00000]
	fatal error: stack overflow

	runtime stack:
	runtime.throw({0x4488e9e, 0x5f0a740})
		/usr/local/go/src/runtime/panic.go:1198 +0x71
	runtime.newstack()
		/usr/local/go/src/runtime/stack.go:1088 +0x5ac
	runtime.morestack()
		/usr/local/go/src/runtime/asm_amd64.s:461 +0x8b

	goroutine 24 [running]:
	github.com/tinygo-org/tinygo/interp.(*memoryView).get(0xc002346900, 0x895)
		/home/circleci/project/interp/memory.go:227 +0x327 fp=0xc025a003a8 sp=0xc025a003a0 pc=0xd25de7
	github.com/tinygo-org/tinygo/interp.pointerValue.llvmValue(...)
		/home/circleci/project/interp/memory.go:518
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0041b5f00}, 0xc002346900)
		/home/circleci/project/interp/memory.go:526 +0x5b fp=0xc025a00568 sp=0xc025a003a8 pc=0xd2775b
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc0039e4e00, 0x895, 0xc025a006e8}}, 0xc002346900)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025a00658 sp=0xc025a00568 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0041b5e00}, 0xc002346900)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025a00818 sp=0xc025a00658 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc0039e4e00, 0x895, 0xc025a00998}}, 0xc002346900)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025a00908 sp=0xc025a00818 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0041b5d00}, 0xc002346900)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025a00ac8 sp=0xc025a00908 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc0039e4e00, 0x895, 0x4f99494b0623b}}, 0xc002346900)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025a00bb8 sp=0xc025a00ac8 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0041b5c00}, 0xc002346900)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025a00d78 sp=0xc025a00bb8 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc0039e4e00, 0x895, 0xc025a00ef8}}, 0xc002346900)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025a00e68 sp=0xc025a00d78 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0041b5b00}, 0xc002346900)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025a01028 sp=0xc025a00e68 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc0039e4e00, 0x895, 0xc025a011a8}}, 0xc002346900)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025a01118 sp=0xc025a01028 pc=0xd28959
[... 280 lines cut ...]
		/usr/local/go/src/runtime/stack.go:1088 +0x5ac
	runtime.morestack()
		/usr/local/go/src/runtime/asm_amd64.s:461 +0x8b

	goroutine 23 [running]:
	github.com/tinygo-org/tinygo/interp.(*memoryView).get(0xc002355860, 0x895)
		/home/circleci/project/interp/memory.go:227 +0x327 fp=0xc0259803a8 sp=0xc0259803a0 pc=0xd25de7
	github.com/tinygo-org/tinygo/interp.pointerValue.llvmValue(...)
		/home/circleci/project/interp/memory.go:518
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0019e3800}, 0xc002355860)
		/home/circleci/project/interp/memory.go:526 +0x5b fp=0xc025980568 sp=0xc0259803a8 pc=0xd2775b
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc00020ce00, 0x895, 0x4fb5caee268d0}}, 0xc002355860)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025980658 sp=0xc025980568 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0019e3700}, 0xc002355860)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025980818 sp=0xc025980658 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc00020ce00, 0x895, 0xc025980998}}, 0xc002355860)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025980908 sp=0xc025980818 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0019e3600}, 0xc002355860)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025980ac8 sp=0xc025980908 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc00020ce00, 0x895, 0xc025980c48}}, 0xc002355860)
		/home/circleci/project/interp/memory.go:802 +0x1d9 fp=0xc025980bb8 sp=0xc025980ac8 pc=0xd28959
	github.com/tinygo-org/tinygo/interp.pointerValue.toLLVMValue({0x0}, {0xc0019e3500}, 0xc002355860)
		/home/circleci/project/interp/memory.go:535 +0x1c5 fp=0xc025980d78 sp=0xc025980bb8 pc=0xd278c5
	github.com/tinygo-org/tinygo/interp.rawValue.rawLLVMValue({{0xc00020ce00, 0x895, 0xc025980ef8}}, 0xc002355860)
	....

fgsch, 2021-10-03:
@nirui is that with the fix in #2001? That is not merged yet.

nirui, 2021-10-04:
@fgsch Thanks for the info. 

So I tried the code in `precise-alloc` branch. It stopped the panic related to `interp.rawValue.rawLLVMValue` it seems, but the program still panics and throwing out:

	fatal error: unexpected signal during runtime execution
	[signal SIGSEGV: segmentation violation code=0x80 addr=0x0 pc=0x3fd5c67]
	
	runtime stack:
	runtime.throw({0x4c12ded, 0x7774449371333600})
		.../go1.17/src/runtime/panic.go:1198 +0x71
	runtime.sigpanic()
		.../go1.17/src/runtime/signal_unix.go:719 +0x396
	
	goroutine 8 [syscall]:
	runtime.cgocall(0xd04200, 0xc0052f5938)
		.../go1.17/src/runtime/cgocall.go:156 +0x5c fp=0xc0052f5910 sp=0xc0052f58d8 pc=0x9db85c
	tinygo.org/x/go-llvm._Cfunc_LLVMRunPassManager(0x7fd4bdebcd50, 0x7fd4bc022440)
		_cgo_gotypes.go:8896 +0x4c fp=0xc0052f5938 sp=0xc0052f5910 pc=0xbbe06c
	tinygo.org/x/go-llvm.PassManager.Run.func1({0xbdc999}, {0xc0052f5998})
		.../go_path/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:1998 +0x70 fp=0xc0052f5978 sp=0xc0052f5938 pc=0xbdac70
	tinygo.org/x/go-llvm.PassManager.Run({0x7fd4bdebcd50}, {0x4c04cdb})
		.../go_path/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:1998 +0x19 fp=0xc0052f5998 sp=0xc0052f5978 pc=0xbdabb9
	github.com/tinygo-org/tinygo/transform.Optimize({0xc003dd3dc8}, 0xc00021e2d0, 0x2, 0x0, 0x5)
		.../tinygo/transform/optimizer.go:64 +0x28c fp=0xc0052f5d90 sp=0xc0052f5998 pc=0xcb592c
	github.com/tinygo-org/tinygo/builder.optimizeProgram({0x4ba3120}, 0xc00021e2d0)
	[..more lines..]

Not sure it's because my setup or something else. I'll test it again after the PR is completed.

Thank you again!

dkegel-fastly, 2021-10-04:
I think precise-alloc has regressed.  

Try the old version,
```
    git clone https://github.com/tinygo-org/tinygo && \
    cd tinygo && \
    git fetch origin e1a1f78735b08764dde096dd1483889596164ea3 && \
    git reset --hard FETCH_HEAD && \
    git submodule update --init --recursive
```

dkegel-fastly, 2022-01-03:
Try current dev branch, the regression was fixed... that said, initializing a complex regexp at init time
rather than runtime is likely to cause trouble for now.

codefromthecrypt, 2022-09-07:
@nirui can you close this issue? It looks like you were able to succeed (due to 👍 ) and I also just tried code in your comment and was able to run https://github.com/tinygo-org/tinygo/issues/1848#issuecomment-932891188

nirui, 2022-09-07:
Hi @codefromthecrypt, I can confirm the code is working. But I'm not really sure how to close this issue (I don't have permission to do so).

Maybe give @mayocream the raiser a ping would help?

codefromthecrypt, 2022-09-07:
indeed you are right @nirui! @mayocream do you mind closing this out?

mayocream, 2022-09-07:
> indeed you are right @nirui! @mayocream do you mind closing this out?

Thanks for the follow up, i'll close this issue as it was resolved. 😀

PR #1850 (2021-05-04): interp: support extractvalue/insertvalue with multiple operands
TinyGo doesn't emit these instructions, but they can occur as a result of optimizations.

I have found this bug while investigating #1848. This PR doesn't fix #1848, however.
closed 2022-09-07 by mayocream