tinygo-org/tinygo #1851: panic: runtime error with i2c and bluepill

ofauchon, 2021-05-02:
Building simple I2C code on bluepill fails with "panic: runtime error: nil pointer dereference"

Step to reproduce: 

* Edit  main.go 
```
func main() {
	machine.I2C0.Configure(machine.I2CConfig{
		Frequency: machine.TWI_FREQ_400KHZ,
	})
}
```
* Build and flash

```
$ tinygo build -size=full -target=bluepill -opt=1 -o main.elf main.go
```

* Run it

It 'll fail with error: 
```
panic: runtime error: nil pointer dereference"
```

It could find i2c.Bus is undefined in I2C Configure() function:

https://github.com/tinygo-org/tinygo/blob/1f73941c43dac9bc91ca651479655fea5c01c538/src/machine/machine_stm32_i2c_reva.go#L120-L124

* Fix Proposal  

Here is the fix I use to make I2C work again (tested on bluepill + HD44780_I2C)

https://github.com/tinygo-org/tinygo/compare/dev...ofauchon:i2c_fix

@kenbell : Do this patch makes sense to  you ? 

Thx

kenbell, 2021-05-02:
Yeah, that declaration looks a bit odd.  I guess i missed testing it when i added the 'revb' i2c implementation.

I think the change is good, you'll raise a PR?

ofauchon, 2021-05-02:
Thx for the feedback!
Yes I'll PR this tomorrow

ofauchon, 2021-05-03:
You'll find the PR for the fix here: 

https://github.com/tinygo-org/tinygo/pull/1852

Thanks

deadprogram, 2021-05-05:
#1852 was merged, so marking this issue to be closed on next release.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram