tinygo-org/tinygo #1855: Strange behaviour with timers and i2c on bluepill

ofauchon, 2021-05-03:
Hi all, 

I noticed some kind of unexpected behaviour while playing with i2c and timers on BluePill board.
It seems i2c Tx interferes with timers : 

In the following code, there are three 2ms pulses (low/high). 

Note there  is an extra  I2C Tx  communiction before the second pulse: 
```
package main

import (
	"machine"
	"time"
)

func main() {

	testpin := machine.PA0
	testpin.Configure(machine.PinConfig{Mode: machine.PinOutput})

	testpin.Low()
	time.Sleep(time.Microsecond * 2000)
	testpin.High()

	time.Sleep(time.Second)

	machine.I2C1.Tx(uint16(0x20), []uint8{0xFF}, nil)
	machine.I2C1.Tx(uint16(0x20), []uint8{0xFF}, nil)
	machine.I2C1.Tx(uint16(0x20), []uint8{0xFF}, nil)
	machine.I2C1.Tx(uint16(0x20), []uint8{0xFF}, nil)

	testpin.Low()
	time.Sleep(time.Microsecond * 2000)
	testpin.High()

	time.Sleep(time.Second)

	testpin.Low()
	time.Sleep(time.Microsecond * 2000)
	testpin.High()

}}
```

The scope show the three pulses: 

[image]

... but the second pulse width is wrong:

1st pulse  width : 2,009 ms 
2nd pulse width : 1,102 ms 
3rd pulse width : 2,001 ms 

I'm still analyzing this behaviour, but I guess I2C communications interferes with timers. 
This bad timing cause errors with "hd44780spi" driver. 

Any idea ? 

Thx  

Olivier

kenbell, 2021-05-04:
I do see slightly different timing for the middle pulse, but not quite the same as your values (note: had to add a line to config I2C1 to get that).  Given this is 'bit-banging' based on timer, I suspect it's due to the tick resolution being 1 millisec.

The implementation of sleep is this:
```golang
// Pause the current task for a given time.
//go:linkname sleep time.Sleep
func sleep(duration int64) {
	addSleepTask(task.Current(), nanosecondsToTicks(duration))
	task.Pause()
}
```

I.e. the max resolution of sleep is actually ticks.  Sleeping for 2000us will be translated into a sleep for 2 ticks.  I suspect the i2c code is adding a delay of almost 1 tick, so rounding in the scheduler is causing the actual sleep to be just more than 1 tick.  I doubt the i2c code is causing it as such - any code that delayed by just-less than 1000us would have the same effect.

If sleep accuracy is needed down to sub 1000us, you could try increasing the `TICK_RATE` constant in `runtime_stm32f103.go` to, say 10Khz as a test.

kenbell, 2021-05-04:
Note, this code i think is why it returns early (in scheduler.go):
```go
		if sleepQueue != nil && now-sleepQueueBaseTime >= timeUnit(sleepQueue.Data) {
```
The choice here is to possibly sleep up-to 1 tick extra, or to possibly return up-to 1 tick early from sleep.  This code looks like it's choosing to return from sleep up-to 1 tick early given it uses `>=`.

ofauchon, 2021-05-04:
Some results while playing with TICK_RATE: 

* with TICK_RATE=10000

Pulse1 width = 2,009
Pulse2 width = 1.921
Pulse3 width = 2.001

* with TICK_RATE=100000

Pulse1 width = 2,009
Pulse2 width = 2.012
Pulse3 width = 2.011

=> We can confirm the Sleep Precision is related to Tick rate. 

Do that mean we don't have sub-ms timers out-of-the-box ?
Is that stm32 specific ? 

If I'm right, why not implementing a blocking timer for short durations ? 

Many drivers uses sub-ms Sleeps : 

Olivier

ofauchon, 2021-05-06:
The problem is realted to STM32's tick precision. 
This should be solved by Optimisation in Pull Request https://github.com/tinygo-org/tinygo/pull/1866

deadprogram, 2021-05-30:
Now that #1867 has been merged, can we mark this issue to be closed on next release?

ofauchon, 2021-05-30:
Yes, you can mark it as closed.

Thanks!

closed 2021-07-01 by deadprogram