tinygo-org/tinygo #1857: "make llvm-build" fails with big sur commandline tools?

dkegel-fastly, 2021-05-05:
I updated to Big Sur, and brew forced me to updated commandline tools...
and now "make llvm-build" fails with 
```
ld: unknown option: --color-diagnostics
```
right after 
```
-- Performing Test CXX_SUPPORTS_NO_NESTED_ANON_TYPES_FLAG - Success
```
Others have complained about this in the past in other contexts... e.g. https://github.com/NASA-SW-VnV/ikos/issues/48

The hack
```
--- a/Makefile
+++ b/Makefile
@@ -154,6 +154,8 @@ gen-device-stm32: build/gen-device-svd
 # Get LLVM sources.
 $(LLVM_PROJECTDIR)/llvm:
        git clone -b xtensa_release_11.0.0 --depth=1 https://github.com/tinygo-org/llvm-project $(LLVM_PROJECTDIR)
+       echo "Working around Big Sur problem by disabling unsupported linker color diagnostics flag."
+       sed -i.bak 's,append_if(LINKER_SUPPORTS_COLOR_DIAGNOSTICS,append_if(LINKER_SUPPORTS_COLOR_DIAGNOSTICS_NOPITY_NOPE_NOPE,' $(LLVM_PROJECTDIR)/llvm/cmake/modules/HandleLLVMOptions.cmake
 llvm-source: $(LLVM_PROJECTDIR)/llvm
 # Configure LLVM.
```
seems to get past the problem, but there's probably a better way.

aykevl, 2021-05-05:
Maybe the current working directory needs to be changed? Just a guess.

dankegel, 2021-05-05:
That's not it, I think. 

This is not really a tinygo bug, may as well close or ignore until someone figures out what's going on.

deadprogram, 2021-05-09:
In that case I will close. Please reopen if needed. Thanks!

closed 2021-05-09 by deadprogram