tinygo-org/tinygo #1868: build wasm  data section unimplemented: alignment of type

AdwindOne, 2021-05-08:
wasm convert  wat  data section :

(data (;0;) (i32.const 65536) "\unimplemented: alignment of type \01\01\00 \00\00\00\00\00\00\00\00\00\00\00unimplemented: (reflect.Type).Elem()\00\00\00\00P\01\01\00$\00\00\00Field\00\00\00\00\00\00\00\00\00\00\00reflect: field index out of range\00\00\00\00\00\00\00\90\01\01\00!\00\00\00<unimplemented>LenNumField\00\00\00\00\00\00unimplemented: size of type\00\00\00\00\00\e0\01\01\00\1b\00\00\00BoolComplexFloat\00\00\00\00\00\00\00\00reflect: slice index out of range\00\00\00\00\00\00\00 \02\01\00!\00\00\00reflect: string index out of range\00\00\00\00\00\00P\02\01\00\22\00\00\00IndexInt\00\00\00\00\00\00\00\00unimplemented: (reflect.Value).Pointer()\90\02\01\00(\00\00\00Pointer<T>UintEOF\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00bytes.Buffer: too largeAtoiParseIntParseUint0123456789abcdef\00\00\00\00value out of rangeinvalid syntax0123456789abcdefghijklmnopqrstuvwxyz\00\00\00\00\00\00\00\00\00\00\00\0000010203040506070809101112131415161718192021222324252627282930313233343536373839404142434445464748495051525354555657585960616263646566676869707172737475767778798081828384858687888990919293949596979899panic: out of memory\00\00\00\00type assert failed\00\00\00\00\00\00\00\00\00\00\00\00\00\00index out of range\00\00\00\00\00\00\00\00\00\00\00\00\00\00nil pointer dereference\00\00\00\00\00\00\00\00\00comparing un-comparable type\00\00\00\00panic: runtime error: \00\00\00\00\00\00\00\00\00\00slice out of rangeErrorResultvalue_ptrGetStateLenGetStatekeyfieldvaluePutStateSuccessResultnum1\00failed to get num1num2\00\00\00\00\00\00\00\00\00\00failed to get num2\00\00\00\00\00\00\00\00\00\00\00\00\00\00Result:==============__context_ptr__\00\00\00\00\00\00\00\00\00\00\00\00failed to get ctx ptrget ptr err: ctx_ptrv1.1.0\00\e9\05\01\00\06\00\00\00versionmethod\00\00\00\00\00\00\00\00\00\00\00call increase, put state from empty\00\00\00\00\00\00\00\00\00\00\00\00\00call increase, put state from existCounterheartbeatcall querycounterr\00\00\00\00\00\00\00\00\00\00\00failed to query state\00\00\00\00\00\00\00\00\00\00\00call query, heartbeat:strconv.parsing : \00\00\00\00\00\00\00\00reflect: call of reflect.Type.\00\00reflect: call of reflect.Value. on invalid type\00\00\00\00\00\01\00\00\00\0a\00\00\00\0a\00\00\00\0e\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\03\04\22\00\04\22\05\043\09\01\04\22\0d\01\00\22\14\00\00\00\00\00\00\00\00\00\00\00\00\00\00\04Func\03Num\03Err\06Method\01s")

aykevl, 2021-05-08:
How did you get this error?

aykevl, 2021-05-08:
This looks like the data section of a WebAssembly file, I don't see anything special about it.

AdwindOne, 2021-05-10:
This error is an error message in the generated WASM after compilation

AdwindOne, 2021-05-13:
> This looks like the data section of a WebAssembly file, I don't see anything special about it.

data[0] section   always see  "index out of range... "  and  calculate two numbers over a billion  return panic unreachable

aykevl, 2021-05-13:
> This error is an error message in the generated WASM after compilation

This is normal. These errors do not happen in correct programs.

Do you get an error when you run the wasm binary?

AdwindOne, 2021-05-13:
> > This error is an error message in the generated WASM after compilation
> 
> This is normal. These errors do not happen in correct programs.
> 
> Do you get an error when you run the wasm binary?

TinyGo did not got this error when compiling a Golang file into WASM, but when I converted WASM into WAT, I found that the data segment got this error  ，I don't know if it effect me from doing over a billion number addition operations .Because I got an error message "Panic unreachable" when adding more than 1 billion numbers

AdwindOne, 2021-08-18:
> program
Unreachable happens when I execute the loop

davecheney, 2026-10-26:
Sorry we did not get this resolved. The diagnostic strings in the WASM data section do not indicate that those errors occurred. Is the runtime trap still happening with TinyGo 0.43 or newer? If so, please share a minimal program, build command, WASM runtime and version, and the full error output.