tinygo-org/tinygo #1869: wasm js.FuncOf - 'panic: sync: unlock of unlocked Mutex'

WhiteHexagon, 2021-05-08:
Using:

tinygo version 0.15.0 darwin/amd64 (using go version go1.15.3 and LLVM version 10.0.1)
macos 10.12.4
firefox 68.12

Compiles into WASM, but at runtime generates the following error on the browser console:

`panic: sync: unlock of unlocked Mutex`

code:

```
package main

import "syscall/js"

var imageOnLoadCallback js.Func = js.FuncOf(func(this js.Value, args []js.Value) interface{} {
	return nil
})

func main() {
}
```

Not sure if it something I am doing wrong... but the same code inside main() compiles and runs fine.  I have been using tinygo wasm for about 12 months on and off, after getting frustrated with the wasm binary size on regular Go.  I have to say I am very impressed and very happy that you are working on this project.  Thank you very much!

As an aside, is there any way to trace back such panics to the root cause?  My code base is quite large now, and it took a couple hours to isolate the problem, since I was in the middle of a refactor.  Normally I debug with fmt.Println :) but in this case the wasm fails before any printlns can execute.

FrankReh, 2021-05-08:
Some of us have experienced that tinygo can't do everything at init time that it can do at runtime. While you're using an older version of tinygo than many by now, it is possible that the call to js.FuncOf isn't well supported at init time, or more specifically, that the sync.Mutex isn't well supported at init time. One of the small prices we pay for using tinygo to compile to wasm rather than the go compiler itself.

If you can leave the variable defined globally but init to nil and then run js.FuncOf at runtime to set it, at the start of main or somewhere on the call stack that main has started, you might get past this problem.

WhiteHexagon, 2021-05-08:
Thanks!  good to know, since I remember seeing something similar for another init vs runtime issue around the concurrency stuff some time ago.  Although I tend to avoid all the concurrency features of Go now, since I guess it is all single threaded once it is within wasm/javascript sandbox?  Although just looking at the source for FuncOf makes me wonder if my assumption is correct (the thread safe increment)... and the talk about callbacks from the JS world arriving on different go routines...

Is there a way I could have tracked down easier which line of my code was triggering the panic?  Quite often I find myself trying to guess which line I added that is suddenly causing a panic, usually something stupid like fmt.Print on a struct :)

Apologies as well for not testing with latest versions.  I got burned when Go changed the memory semantics for passing data between Go and JS and only just getting over that experience.  I think Go will drop my OS version soon anyway, so hopefully I get this project completed before all the generics stuff arrives.  Thanks again for the answer.

FrankReh, 2021-05-09:
To answer your question about how to track down the line of code. All I can offer is that I've gotten very used to bisecting what I've changed since the last time it compiled and ran. Also, often I just write a loop on the side that keeps recompiling as the shell sees a file get rewritten. Another big step in my understanding was when I figured out I could change any of the files in the tinygo/src directory and for those not in that tree, I could change the source in the go tree and see the results in the Wasm - neither the tinygo tool nor the go tool needed to be recompiled. Sometimes I look at the IR, using -printir on the tinygo command line. Sometimes I look at the Wasm itself with a wasm2wat tool, and sometimes I add println statements to the source, or put in panics to learn more about what points are being reached.

I've also wondered why the mutex in the func.go file. I've wondered if the original author was guarding against an allocation causing a GC, or a memory grow, which somehow might cause another entry into the table. I don't know. So far, the code has always worked if I run without that Mutex too.

WhiteHexagon, 2021-05-09:
Great to know I am not alone in this challenge :) and thanks for the tips! especially modifying the src folders, I shall investigate that further.  I was also tempted to change some stuff in the .js bridge file, since there seems to be quite some overhead with Funcs in both directions,  but so far just added a couple of my own js functions in my webpage for passing Uint8 arrays back and forth.

Anyway I am back up and running again :) and back to 60fps after my refactoring (I moved my websocket stuff into a webworker).  Also been trying to reduce allocations, and striped out all my go routines and channels now.  Hence I was in a bit of a mess when that panic caught me out :)  lesson learned, small refactorings!  thanks again, this can be closed if you like.

davecheney, 2026-10-24:
Thanks for the update. Creating `js.FuncOf` at package scope also works with current TinyGo. This example checks initialization, invocation, and release:

```go
package main

import "syscall/js"

var callback = js.FuncOf(func(this js.Value, args []js.Value) interface{} {
    return 42
})

func main() {
    if callback.Invoke().Int() != 42 {
        panic("unexpected callback result")
    }
    callback.Release()
    println("package-level js.FuncOf: OK")
}
```

```text
$ tinygo run -target=wasm main.go
package-level js.FuncOf: OK
```

Checked with TinyGo 0.44.0-dev-eae3f875, Go 1.27.3, and Node v26.12.1.

Closing as suggested above. Please open a new issue with a minimal reproducer if you encounter further problems.

closed 2026-10-24 by davecheney