tinygo-org/tinygo #1884: array indexes ignored when initializing pointers to array elements at package level

knieriem, 2021-05-15:
When I'm initializing a pointer to an array element at package level the following way,

```go
var a = [4]int{1, 2, 3, 4}

var p = &a[2]
```
`p` will point to a[0] instead of a[2] starting with tinygo v0.17.0. The array index, in this example: 2, appears to be ignored, and replaced by zero.

I am observing this for initializations at package level and within init, but not for pointers initialized within functions. I'm using the following tinygo versions:

* 0.18.0 linux/amd64 (using go version go1.16 and LLVM version 11.1.0)
* 0.17.0 linux/amd64 (using go version go1.16 and LLVM version 11.1.0)
* 0.16.0 linux/amd64 (using go version go1.15.6 and LLVM version 10.0.1)

Could there be a problem with my LLVM packages? I'm using Debian packages from apt.llvm.org/buster/.
I can see the behaviour when running this program. It initializes pointer variables at
various levels to the same array element, and checks whether each variable contains the correct pointer:

```go
package main

var (
	a = [4]int{1, 2, 3, 4}

	ptop = &a[2]

	pinit *int

	pfunc *int
)

func init() {
	pinit = &a[2]
}

func f() {
	pfunc = &a[2]
}

func main() {
	plocal := &a[2]
	f()
	test(ptop, "package")
	test(pinit, "init")
	test(pfunc, "function")
	test(plocal, "local")
	test(&a[2], "value")
}

func test(p *int, level string) {
	status := "OK"
	if p == &a[0] {
		status = "ERR"
	}
	println(p, *p, level, status)
```

With tinygo v0.16 it prints the value of the third position, "3", and "OK" in all cases, with tinygo v0.17 and v0.18 the output is:

    0x0000000000404068 1 package ERR
    0x0000000000404068 1 init ERR
    0x0000000000404078 3 function OK
    0x0000000000404078 3 local OK
    0x0000000000404078 3 value OK

knieriem, 2021-05-16:
btw, the first revision, where the new behaviour starts, appears to be 30df912565c2.

knieriem, 2021-05-23:
A partial fix appears to be to change `llvm.Type{}` to `llvmType` in interp/memory.go:912. Possibly a typo.

```go
	case llvm.PointerTypeKind:
		if v.buf[0] > 255 {
			// This is a regular pointer.
			llvmValue, err := pointerValue{v.buf[0]}.toLLVMValue(llvm.Type{}, mem)
                                                                                ^^^^
```

Because of the empty type, the array offset stored within the pointerValue got ignored, so that the index value was missing from the GEP operation. After replacing `llvm.Type{}` with `llvmType` (the actual type), the correct GEP expression is constructed, and my example code above runs as expected.

But a more complex example, where pointers to array elements are stored into elements of a slice of structs, still does not work:
```
package main

type B struct {
	p *int
}

var (
	a = [4]int{1, 2, 3, 4}

	b = []B{
		{p: &a[1]},
		{p: &a[2]},
	}
	p = &a[2]
)

func main() {
	println(*p, *b[1].p)
}
```
This should print `3 3`, but prints `3 1`.

aykevl, 2021-06-14:
Thank you for the investigation! I believe the bug is here:

https://github.com/tinygo-org/tinygo/blob/77647970613e6f4115c93521342c891aa2bba02c/interp/memory.go#L574-L576

It should still do the GEP, even if the offset isn't known.

antichris, 2021-06-29:
Could also be the cause why the following code outputs "foo bar" when it clearly [should produce "bar bar"][goplay]?

```golang
package main

func Bar() string { return "foobar"[3:6] }

var bar = Bar()

func main() { println(bar, Bar()) }
```

[goplay]: https://play.golang.org/p/RrGlEI9xceC

knieriem, 2021-06-29:
Yes, this effect has the same cause. I suppose, when the value of the package level variable `bar` is computed, function `Bar()` gets inlined, so that the actual code is: `var bar = "foobar"[3:6]`. Apparently the offset [3], but not the slice length, gets ignored.

 It appears to be fixed by the change mentioned at the top of [my comment above](#issuecomment-846646208)

knieriem, 2021-06-29:
@aykevl, referring to your comment 

> It should still do the GEP, even if the offset isn't known.

-- to specify a GEP, isn't it necessary to know the offsets, to be able to compute correct index operands?

On the other hand, wouldn't the compiler be able to always know the offsets, even in more complex cases like the initialization of a struct field with a pointer to an array element (like in my code snippet above)? -- so that a nil llvmType could be avoided?

I've experimented with replacing `llvm.Type{}` in https://github.com/tinygo-org/tinygo/blob/77647970613e6f4115c93521342c891aa2bba02c/interp/memory.go#L787-L791
with a `llvmType` value derived from `v.buf[i]`, but it's not working yet. I think I need to understand llvm concepts, and how interp/memory.go works, better first.

aykevl, 2021-06-30:
> Could this also be the cause why the following code outputs "foo bar" when it clearly [should produce "bar bar"](https://play.golang.org/p/RrGlEI9xceC)?

Yes, most likely.

> I suppose, when the value of the package level variable `bar` is computed, function `Bar()` gets inlined, so that the actual code is: `var bar = "foobar"[3:6]`.

Not exactly. The interp package literally runs the code as a kind of interpeter, storing heap allocations as global variables.

> > It should still do the GEP, even if the offset isn't known.
> 
> -- to specify a GEP, isn't it necessary to know the offsets, to be able to compute correct index operands?

Sorry I made a mistake there. I meant to say: "It should still do the GEP, even if the type isn't known".

> On the other hand, wouldn't the compiler be able to always know the offsets, even in more complex cases like the initialization of a struct field with a pointer to an array element (like in my code snippet above)? -- so that a nil llvmType could be avoided?

Unfortunately, no. Heap allocations are untyped so `rawLLVMValue` makes up a type as it creates the value. Hopefully this becomes much less of a problem once LLVM converts entirely to opaque pointers (where all pointers in IR are of the same type, they don't have a LLVM type anymore).

knieriem, 2021-07-08:
As an experiment, I have modified the [`if llvmType.IsNil()`](https://github.com/tinygo-org/tinygo/blob/77647970613e6f4115c93521342c891aa2bba02c/interp/memory.go#L574-L576) block @aykevl [mentioned,](#issuecomment-860754431) so that a GEP is created also if the type is unknown. With this change I have got the correct results for [my example program.](#issuecomment-846646208) The remaining problem is that I had to divide the array element _byte offset_ by a hardcoded array element size (that matches the element size used in my example) to get the array element _index_.

For example, when initializing a struct field with `&a[2]`, a pointer refering to the second element of an int32 array,
the compiler might create a _store_ instruction with an operand `<1+8>`, with 1 refering to the array `a`, and 8 being the byte offset `sizeof(int32) * 2`.
When trying to create a gep in lines 574.. of interp/memory.go then, it seems I have to provide the array index 2 instead of the byte offset 8. Thus, I need to divide by the element size. I haven't found a way yet to extract this element size, though.

In this context, where does a _store_ instruction know how wide the value is that it is storing?  
For instance, `a = [4]int32{1, 2, 3, 4}` appears to be translated into

    store: 1 <1>
    store: 2 <1+4>
    store: 3 <1+8>
    store: 4 <1+12>

while, similarly, `a = [4]int16{1, 2, 3, 4}` is compiled to

    store: 1 <1>
    store: 2 <1+2>
    store: 3 <1+4>
    store: 4 <1+6>

The byte offsets are adjusted accordingly, but where does each single _store_ instruction know that the element size is 2 resp. 4 bytes?

aykevl, 2021-07-08:
I have made a fix here: https://github.com/tinygo-org/tinygo/pull/1986
I simply cast the type to a *i8 so that I don't have to worry about element size (this includes structs and the like!).

knieriem, 2021-07-09:
Thank you for the fix. I can confirm that it also fixes my original, more complex program.

Do you think it would still be a good idea to also make the change in line [interp/memory.go:923](https://github.com/tinygo-org/tinygo/blob/8dc9d3eb24909c180bd7d14cbf40cd2cb186f887/interp/memory.go#L923), as suggested [in an earlier comment](#issuecomment-846646208)? It appears that it is not strictly necessary, because it will be taken care of by your fix too. But, since llvmType is non-Nil (llvm.PointerTypeKind) in that case, because it stems from the call `obj.buffer.toLLVMValue(obj.llvmGlobal.Type().ElementType(), &mem)` in [interp.go:242,](https://github.com/tinygo-org/tinygo/blob/8dc9d3eb24909c180bd7d14cbf40cd2cb186f887/interp/interp.go#L242) it possibly wouldn't hurt to use it. If needed, I could provide a PR.

aykevl, 2021-07-09:
> Do you think it would still be a good idea to also make the change in line [interp/memory.go:923](https://github.com/tinygo-org/tinygo/blob/8dc9d3eb24909c180bd7d14cbf40cd2cb186f887/interp/memory.go#L923), as suggested [in an earlier comment](#issuecomment-846646208)?

Looking at the code, I can't really think of a good reason not to do that. So yes, if you make a PR that would be great.

> The byte offsets are adjusted accordingly, but where does each single _store_ instruction know that the element size is 2 resp. 4 bytes?

From the stored value. It stores either a 2-byte value or a 4-byte value in your example. It's the `v.len` method here for example (`v` is the value to be stored):

https://github.com/tinygo-org/tinygo/blob/0565b7c0e050f0cccbccdcbc1a9cee43462603d9/interp/memory.go#L302

closed 2021-09-21 by aykevl