tinygo-org/tinygo #1885: interp panic during compilation

kenbell, 2021-05-15:
branch: fork off dev
build command: tinygo build -target stm32f4disco-1 -o test.elf --opt=0 ./src/examples/pwm/

panic:
```
panic: interp: unknown type kind in markExternalValue

goroutine 51 [running]:
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb4434f00, 0x7f3fb4434f02)
        /home/kenneth/work/tinygo/interp/memory.go:200 +0x7e5
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb424d628, 0x7f3fb424d602)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb41a1248, 0x7f3fb41a1202)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb43a7568, 0x7f3fb43a7502)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb46db168, 0x7f3fb46db102)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb43f63b8, 0x7f3fb43f6302)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternal(0xc000d5b860, 0x7f3fb46db318, 0x7f3fb444b602)
        /home/kenneth/work/tinygo/interp/memory.go:168 +0x372
github.com/tinygo-org/tinygo/interp.(*memoryView).markExternalStore(...)
        /home/kenneth/work/tinygo/interp/memory.go:111
github.com/tinygo-org/tinygo/interp.(*runner).runAtRuntime(0xc002dc4000, 0xc0028b1090, 0x2d, 0x8, 0xc001a2a900, 0x6, 0x8, 0x7f3fb496e7a0, 0x0, 0x0, ...)
        /home/kenneth/work/tinygo/interp/interpreter.go:914 +0x350
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc002dc4000, 0xc0028b1090, 0x0, 0x0, 0x0, 0x0, 0x808c1a, 0x4, 0x0, 0x0, ...)
        /home/kenneth/work/tinygo/interp/interpreter.go:492 +0x66c5
github.com/tinygo-org/tinygo/interp.Run(0x7f3fb41a7db0, 0xc0033dfd00, 0x0, 0x0)
        /home/kenneth/work/tinygo/interp/interp.go:112 +0x69f
github.com/tinygo-org/tinygo/builder.optimizeProgram(0x7f3fb41a7db0, 0xc0002aa360, 0xc0034128a0, 0x2)
        /home/kenneth/work/tinygo/builder/build.go:656 +0x3f
github.com/tinygo-org/tinygo/builder.Build.func2(0xc0005f1aa0, 0x0, 0x0)
        /home/kenneth/work/tinygo/builder/build.go:390 +0x7c6
github.com/tinygo-org/tinygo/builder.jobWorker(0xc0020f3800, 0xc0020f37a0)
        /home/kenneth/work/tinygo/builder/jobs.go:165 +0xe2
created by github.com/tinygo-org/tinygo/builder.runJobs
```

Adding diagnostic output to the panic, I see:
```
llvmValue.Dump: i32 ()* asm sideeffect "\0A\09\09mrs $0, PRIMASK\0A\09\09cpsid i\0A\09", "=&r"
llvmType.TypeKind: PointerTypeKind
llvmType: PointerType(FunctionType():IntegerType(32 bits
```

It appears to be the implementation of device/arm/arm.go, DisableInterrupts

kenbell, 2021-05-15:
As a temporary fix, i added this case to the switch statement:
```golang
		case llvm.PointerTypeKind:
			// Do nothing
```

This appears to generate functional binaries.

deadprogram, 2021-05-26:
@kenbell can you please verify if #1891 addressed this issue? Thank you.

kenbell, 2021-05-26:
It did, thanks

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

closed 2021-07-01 by deadprogram