tinygo-org/tinygo #1886: Getuid not declared by package os

iarkaroy, 2021-05-16:
I am working on an wasm project. When building with normal go command like:
```
GOARCH=wasm GOOS=js go build -o app.wasm
```
it is working perfectly, but when building with tinygo:
```
tinygo build -o app.wasm -target=wasm
```
it is showing errors:
```
../go/pkg/mod/github.com/google/uuid@v1.2.0/dce.go:47:42: Getuid not declared by package os
../go/pkg/mod/github.com/google/uuid@v1.2.0/dce.go:55:41: Getgid not declared by package os
```
Go version: `1.16.4`

Any suggestion?

aykevl, 2021-05-19:
I have made a patch which should fix this: https://github.com/tinygo-org/tinygo/pull/1892

deadprogram, 2021-05-21:
@iarkaroy can you please check if this corrects your issue? Thank you.

gedw99, 2021-05-21:
@iarkaroy did you have any success ?
 I am also Interested is go-app project being able to compile with tinygo

iarkaroy, 2021-05-22:
When building, I am getting this error:
```
cgo/libclang.go:19:10: fatal error: clang-c/Index.h: No such file or directory
   19 | #include <clang-c/Index.h> // if this fails, install libclang-10-dev
```
I have already installed `libclang-10-dev`. Is there anything else I need to do?

skhwhu, 2021-05-23:
I am also Interested is go-app project being able to compile with tinygo, can you build a fixed tinygo.exe for windows

aykevl, 2021-05-23:
@iarkaroy that error message is probably outdated, you may need to install libclang-11-dev instead. For details, see https://tinygo.org/docs/guides/build/#with-system-installed-llvm.

aykevl, 2021-05-23:
@skhwhu you can download a build from here: https://dev.azure.com/tinygo/tinygo/_build/results?buildId=3391&view=artifacts&pathAsName=false&type=publishedArtifacts

skhwhu, 2021-05-24:
@aykevl thank you very much for you propmt reply. I download tinygo.exe from the above link, but I find "Getuid" doen;t fixed!

C:\Users\SKH\go\pkg\mod\github.com\google\uuid@v1.2.0\dce.go:47:42: Getuid not declared by package os
C:\Users\SKH\go\pkg\mod\github.com\google\uuid@v1.2.0\dce.go:55:41: Getgid not declared by package os

aykevl, 2021-06-24:
> @aykevl thank you very much for you propmt reply. I download tinygo.exe from the above link, but I find "Getuid" doen;t fixed!

You don't need to download the tinygo.exe file, you need to download the entire build and extract it. The tinygo.exe file is just the compiler, it doesn't contain the fixes in #1892.

thepudds, 2021-09-17:
Hi there 👋 I’m a bit curious what the latest here might be, especially at the level of something like “this looks like a tractable problem to solve“ vs. “it might be a long time before this is solved“?

Thanks for all your work on TinyGo!

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram