tinygo-org/tinygo #1887: Compiling ANTLR with tinygo returning Function return type does not match operand type of return inst!

jjeb, 2021-05-17:
Hi, I am trying to compile [ANTLR4](https://github.com/antlr/antlr4/blob/master/doc/index.md) using tinyGo, I am fully aware that there is some challenges so I am not really looking for a solution but to understand a little bit more about the error so I can try to figure it out: 

```
▶ tinygo version
tinygo version 0.18.0 darwin/amd64 (using go version go1.16 and LLVM version 11.0.0)

github.com/tinygo-antlr  tinygo18_single_import ✗   
▶ tinygo build -o fe-test/public/main.wasm -target wasm main.go
Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ATN.187"* %3, !dbg !27382
 %"github.com/antlr/antlr4/runtime/Go/antlr.ATN"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ATN.187"* %5, !dbg !27382
 %"github.com/antlr/antlr4/runtime/Go/antlr.ATN"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ATN.187"* %9, !dbg !27382
 %"github.com/antlr/antlr4/runtime/Go/antlr.ATN"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator.206"* %3, !dbg !27311
 %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator.206"* %5, !dbg !27311
 %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator.206"* %9, !dbg !27311
 %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator"*Function return type does not match operand type of return inst!
  ret %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator.206"* %10, !dbg !27311
 %"github.com/antlr/antlr4/runtime/Go/antlr.ParserATNSimulator"*error: optimizations caused a verification failure

```

aykevl, 2021-05-19:
That certainly looks like a bug in TinyGo. Can you share a small code sample that reproduces this problem?

aykevl, 2021-05-20:
Thank you! Now I can reproduce the issue!

aykevl, 2021-05-20:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/1898

deadprogram, 2021-05-21:
Hello @jjeb can you please check to see if this fix works for you. Thanks.

jjeb, 2021-05-25:
Hello @deadprogram / @aykevl  

It does compiles but it does not work, I am getting and error in web console, when I tried to execute it: 

```
panic: runtime error: nil pointer dereference
main.wasm:0x6505 Uncaught (in promise) RuntimeError: unreachable
    at runtime.runtimePanic (http://localhost:9000/tinygo/main.wasm:wasm-function[100]:0x6505)
    at runtime.nilPanic (http://localhost:9000/tinygo/main.wasm:wasm-function[28]:0x1b1d)
    at (github.com/antlr/antlr4/runtime/Go/antlr.TokenStream).LT (http://localhost:9000/tinygo/main.wasm:wasm-function[863]:0x545c6)
    at (*github.com/antlr/antlr4/runtime/Go/antlr.BaseParser).EnterRule (http://localhost:9000/tinygo/main.wasm:wasm-function[870]:0x55c41)
    at (*github.com/ql/parser.QlParser).Query (http://localhost:9000/tinygo/main.wasm:wasm-function[954]:0x5d660)
    at getCompletionItems (http://localhost:9000/tinygo/main.wasm:wasm-function[252]:0x1294f)
    at _start (http://localhost:9000/tinygo/main.wasm:wasm-function[249]:0x11cf4)
    at global.Go.run (http://localhost:9000/tinygo/wasm_exec.js:478:24)
    at Goinit (http://localhost:9000/tinygo/:11:12)
runtime.runtimePanic @ main.wasm:0x6505
runtime.nilPanic @ main.wasm:0x1b1d
(github.com/antlr/antlr4/runtime/Go/antlr.TokenStream).LT @ main.wasm:0x545c6
(*github.com/antlr/antlr4/runtime/Go/antlr.BaseParser).EnterRule @ main.wasm:0x55c41
(*github.com/ql/parser.QlParser).Query @ main.wasm:0x5d660
getCompletionItems @ main.wasm:0x1294f
_start @ main.wasm:0x11cf4
run @ wasm_exec.js:478
Goinit @ (index):11
async function (async)
Goinit @ (index):10
(anonymous) @ (index):13
favicon.ico:1 
```

I updated the [repo-example](https://github.com/jjeb/ql/commit/d4e5d24f5ae6cfff12c2d63a43f72d3717fa6f0a) so there is a consistent way to reproduce the issue and you can see the error. 

The tinygo version after build from the branch with the fix is: 

``` ~/src/github.com/tinygo-org/tinygo/tinygo version
tinygo version 0.19.0-dev darwin/amd64 (using go version go1.16.3 and LLVM version 11.1.0)
```

Thank you

bluezheng, 2021-08-09:
Hello,

I met the same problem.
Is there any update on this issue?

Thank you

deadprogram, 2021-11-18:
I think the request was fulfilled in the new release, so closing. Please reopen if needed. Thanks!

closed 2021-11-18 by deadprogram