tinygo-org/tinygo #1888: wasm will report an error when using the crypto/x509 package

vicnoah, 2021-05-17:
/snap/go/current/src/net/pipe.go:156:16: ErrDeadlineExceeded not declared by package os
/snap/go/current/src/net/pipe.go:169:16: ErrDeadlineExceeded not declared by package os
/snap/go/current/src/net/pipe.go:188:16: ErrDeadlineExceeded not declared by package os
/snap/go/current/src/net/pipe.go:204:17: ErrDeadlineExceeded not declared by package os

aykevl, 2021-05-19:
This would be best solved by replacing the net package entirely. See: https://github.com/tinygo-org/tinygo/pull/1881

fgsch, 2021-05-19:
Actually, there is another issue that needs to be addressed on top of #1881 - I will open a PR shortly.

fgsch, 2021-05-20:
@vicnoah both PRs are merged now. Can you give it another try using the dev branch?

vicnoah, 2021-05-20:
> @vicnoah both PRs are merged now. Can you give it another try using the dev branch?

Ok, I will try later

vicnoah, 2021-05-24:
> @vicnoah both PRs are merged now. Can you give it another try using the dev branch?

I still get an error compiling with the dev branch.

/home/wwg/golib/pkg/sec/pay/v1/wasm/main.go:21: blocking operation in exported function: clientpay

traceback:
clientpay
        /home/wwg/golib/pkg/sec/pay/v1/wasm/main.go:22:23
github.com/vicnoah/golib/pkg/sec/pay/v1.Decrypt
        /home/wwg/golib/pkg/sec/pay/v1/decrypt.go:71:30
github.com/vicnoah/golib/pkg/sec/pay/v1.decryptCipher
        /home/wwg/golib/pkg/sec/pay/v1/decrypt.go:112:33
github.com/vicnoah/golib/crypto/vrsa.DecryptOAEP
        /home/wwg/golib/crypto/vrsa/rsa_oaep.go:35:35
crypto/rsa.DecryptOAEP
        /snap/go/7360/src/crypto/rsa/rsa.go:608:19
crypto/rsa.decrypt
        /snap/go/7360/src/crypto/rsa/rsa.go:501:25
crypto/internal/randutil.MaybeReadByte
        /snap/go/7360/src/crypto/internal/randutil/randutil.go:25:6
runtime.chanSelect
        /home/wwg/tinygo/src/runtime/chan.go:635:12

fgsch, 2021-05-25:
This is different though.  As the error says, `clientpay` will block and is exported.

vicnoah, 2021-05-28:
> This is different though.  As the error says, `clientpay` will block and is exported.

I don't understand what this means. It didn't generate the wasm file. I will try again.

closed 2021-06-10 by vicnoah

aykevl, 2021-06-13:
> I don't understand what this means.

It means that you do some sort of blocking operation in an exported function. In this case, it appears that the `Decrypt` call in the exported `clientpay` function indirectly tries to generate a random number which in turn has a `select` block, which is a blocking operation. This is not allowed in TinyGo and I don't think it will be in the future because JavaScript itself is non-blocking.