tinygo-org/tinygo #1889: Panic messages are not always printed out

ysoldak, 2021-05-18:
Sometimes my program panics and crashes.

Tinygo runtime tries its best to notify over serial console where exactly problem was, but this message usually fails to be delivered. Probably system dies too soon, could we make sure it prints out the message fully before halting?

An example of a message would be really nice to see in the console: "goroutine stack overflow"
See https://github.com/tinygo-org/tinygo/blob/release/src/internal/task/task_stack.go#L43

I use Arduino Nano 33 IoT board.

ysoldak, 2021-05-27:
@deadprogram Shall I close it myself or it's a part of the release process?

deadprogram, 2021-05-27:
Marking for next release.

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

PR #1908 (2021-05-27): Do not disable interrupts on abort
Allows panic messages to be printed fully into serial console.

I've been testing this tiny fix for several days now.
So far panic messages been delivered reliably to serial console w/o any adverse side effects.

Thanks Ayke for the hint where to look.

Fixes #1889
closed 2021-07-01 by deadprogram