tinygo-org/tinygo #1900: Int is being parsed as int32

hamidrezakks, 2021-05-22:
Hi,

Is there anyway to change default architecture to 64bit,
TinyGO always parse int asa int32,
Even in JS OS package,

aykevl, 2021-05-22:
The `int` type depends on the architecture. So it is `int64` on amd64 and aarch64, and `int32` on all others, including WebAssembly which is a 32-bit system (but includes 64-bit integer types).

If you need to use an integer type that can hold values bigger than an `int32` can hold, you should use `int64` or `uint64` instead of `int`.

hamidrezakks, 2021-05-22:
Well golang wasm and tinygo wasm should have the same logic on WASM right?

In golang wasm int is being parsed as int64,
In fact golang is using int64 bud tinygo is using int32 when using int primitive,

I think it would be good if you use a flag for default int bits.

aykevl, 2021-05-22:
> In golang wasm int is being parsed as int64,
> In fact golang is using int64 bud tinygo is using int32 when using int primitive,

Oh, I didn't know that.

> Well golang wasm and tinygo wasm should have the same logic on WASM right?

Not really. The size of `int` is implementation defined, and TinyGo chooses to follow the pointer size of the underlying architecture which is 32-bits in the case of WebAssembly.

> I think it would be good if you use a flag for default int bits.

This would introduce a lot of complexity especially since it affects the calling convention. It's unlikely I'd want to add it to TinyGo unless there is a really strong reason to do so.

But the most important question is: why do you need `int` to be 64-bits? If you need it to be 64-bits, you should use `int64` instead.

hamidrezakks, 2021-05-22:
Well we are using a cryptography SDK for our products, which might use int, I didn't dig much..., so you are suggesting to port all modules?

All I know is that I get different result using tinygo.

aykevl, 2021-05-22:
> Well we are using a cryptography SDK for our products, which might use int, I didn't dig much..., so you are suggesting to port all modules?
> 
> All I know is that I get different result using tinygo.

In that case you should investigate why you're getting different results. It might be the different `int` type (which would be a bug in the SDK), or it might be a bug in TinyGo.

hamidrezakks, 2021-05-22:
Ok then,
Thanks

closed 2021-05-22 by hamidrezakks