tinygo-org/tinygo #1901: Calls to fmt.Sprintf in my main.go makes tinygo panic at compile time

thomas-maurice, 2021-05-22:
# What ?
I am observing a panic when I am calling the `fmt.Sprintf` function

TLDR
```
$ tinygo build -size=short -target=arduino-nano-new -o firmwares/arduino-nano-new.hex
panic: compiler: could not store type code number inside interface type code

[... stack trace ...]
```

# How to reproduce
I am using tinygo 0.18.0, and go `go version go1.15.6 linux/amd64`

```go
package main

import (
	"fmt"
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	serial := machine.UART0
	serial.Configure(machine.UARTConfig{
		BaudRate: 9600,
	})
	cnt := 0
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
		serial.Write([]byte(fmt.Sprintf("hello from go -> %d\n", cnt)))
		cnt++
	}
}
```

I am using the arduino-nano-new target, and I compile with the following command and output
```
$ tinygo build -size=short -target=arduino-nano-new -o firmwares/arduino-nano-new.hex
panic: compiler: could not store type code number inside interface type code

goroutine 15 [running]:
github.com/tinygo-org/tinygo/transform.LowerReflect(0x7fcf1c02b2c0)
        /home/circleci/project/transform/reflect.go:185 +0xc78
github.com/tinygo-org/tinygo/transform.Optimize(0x7fcf1c02b2c0, 0xc00022e600, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
        /home/circleci/project/transform/optimizer.go:91 +0x478
github.com/tinygo-org/tinygo/builder.optimizeProgram(0x7fcf1c02b2c0, 0xc00022e600, 0xc000e465a0, 0x2)
        /home/circleci/project/builder/build.go:698 +0x13d
github.com/tinygo-org/tinygo/builder.Build.func2(0xc002cb79e0, 0x0, 0x0)
        /home/circleci/project/builder/build.go:390 +0x812
github.com/tinygo-org/tinygo/builder.jobWorker(0xc0007af560, 0xc0007af500)
        /home/circleci/project/builder/jobs.go:165 +0xe0
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:84 +0xc5
```

Let me know if you need more info !

deadprogram, 2021-05-22:
Try not using the `fmt` package on avr processors. It uses reflection which uses too much memory on these small processors.

thomas-maurice, 2021-05-23:
@deadprogram right, thank you for the answer, I will close the issue now !

closed 2021-05-23 by thomas-maurice