tinygo-org/tinygo #1904: I2C Peripheral Mode doesn't work on AVRs

pojntfx, 2021-05-23:
Hi! I'm trying to use an Arduino Uno as an I2C peripheral instead of an I2C controller, but it doesn't seem to be possible to do so using the relevant machine package. To work around this, I though I might use either `device/avr` or CGo, however neither of them seem to work. For the following snippets, I've created minimal, reproducible examples here: [growlapse subdir](https://github.com/pojntfx/growlapse/tree/main/i2c-testing).

## C Sanity Check (`avr-gcc`/`clang`)

So, the first example, which works - written in plain C:

```c
#include <avr/interrupt.h>
#include <util/twi.h>

#define I2C_ADDRESS 0x10

int main() {
  asm volatile("cli" ::: "memory");

  TWAR = I2C_ADDRESS << 1;
  TWCR |= 1 << TWIE;
  TWCR |= 1 << TWEA;
  TWCR |= 1 << TWINT;
  TWCR |= 1 << TWEN;

  asm volatile("sei" ::: "memory");

  while (1) {
  }
}
```

When I run `i2cdetect -y 1`, it shows up at the relevant I2C address:

```shell
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

So far so good; I've tested this example program using both `avr-gcc` and `clang`, both work; to replicate, clone the growlapse repo and use the relevant [Makefile](https://github.com/pojntfx/growlapse/blob/main/i2c-testing/c/Makefile) with `make install` (or `make install-clang` for clang).

## Idea 1: Using CGo

My initial idea of porting this to TinyGo was to use CGo; basically by extracting the `main` function to `i2c_init` ([i2c.c](https://github.com/pojntfx/growlapse/blob/main/i2c-testing/tinygo-cgo/i2c.c)) and then calling it from Go:

```go
package main
/*#include "i2c.h"#cgo CFLAGS: -I/usr/avr/include*/import "C"
func main() {	C.i2c_init()
	for {	}}
```

This however does not work, and the Arduino does not show up using `i2cdetect -y 1`:

```shell
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

I'm using the latest TinyGo release:

```shell
[pojntfx@fedora c]$ tinygo version
tinygo version 0.18.0 linux/amd64 (using go version go1.16.3 and LLVM version 11.0.0)
```

Extracting the `for` loop into C was also tried, with no change to behaviour. To reproduce the issue, use the relevant [Makefile](https://github.com/pojntfx/growlapse/blob/main/i2c-testing/tinygo-cgo/Makefile) and run `make install`.

## Idea 2: Using `device/avr`

My last idea was to use the `device/avr` package ([main.go](https://github.com/pojntfx/growlapse/blob/main/i2c-testing/tinygo/main.go)):

```go
package main

import (
	"device/avr"
)

const (
	i2cAddress = 0x10
)

func main() {
	avr.AsmFull("cli", map[string]interface{}{})

	avr.TWAR.Set(i2cAddress)
	avr.TWCR.SetBits(avr.TWCR_TWIE)
	avr.TWCR.SetBits(avr.TWCR_TWEA)
	avr.TWCR.SetBits(avr.TWCR_TWINT)
	avr.TWCR.SetBits(avr.TWCR_TWEN)

	avr.AsmFull("sei", map[string]interface{}{})

	for {
	}
}
```

This is basically a straight-forward port of the C code. It yields however the same result as the CGo idea, with it not showing up in `i2cdetect`. To reproduce this issue, use the [relevant Makefile](https://github.com/pojntfx/growlapse/blob/main/i2c-testing/tinygo/Makefile) and run `make install`.

Does anyone have an idea what is going on here? I've arrived at a point where I don't know where to continue debugging - my low-level knowledge here is very limited, so I'd be glad for any help :)

deadprogram, 2021-05-24:
The correct way to handle this requires extension to the I2C interface as currently defined. As you have already noticed. 

Let me give it a little thought. Perhaps @aykevl or others want to chime in as well.

deadprogram, 2021-11-22:
Here is a concept bit of code, for how extending the I2C interface in TinyGo to support peripheral code might look:

```go
package main

import (
	"machine"
	"time"
)

var input [16]byte

func main() {
	machine.I2C0.Configure(machine.I2CConfig{
		// valid options are machine.I2CController or machine.I2CPeripheral
		Mode:    machine.I2CPeripheral,
		Address: 0x84,
	})

	for {
		if machine.I2C0.Buffered() {
			n, _ := machine.I2C0.Read(input[:]) 	
			
			machine.I2C0.Write([]byte("hello, "))
			machine.I2C0.Write(input[:n])
			machine.I2C0.Write([]byte("\r\n"))
		}
		
		time.Sleep(10 * time.Millisecond)
	}
}
```

The implication of this code is that calling `Configure()` is enough to start things off, which seems to fit other interfaces. Since the peripheral waits until some I2C controllers talks to it, I thought that borrowing the `UART` interface might work well. One difference is that the i2c peripheral waits to receive some data, and then would generally just respond right away.

Thoughts, anyone?

pojntfx, 2021-12-02:
I like it very much! Feels similar to the C I wrote for now so it should be easy to pick up 👍

soypat, 2022-09-27:
Do most i2c peripheral modes work with a buffer as opposed to some immediate response mechanism, i.e. interrupts?

deadprogram, 2022-09-27:
> Do most i2c peripheral modes work with a buffer as opposed to some immediate response mechanism, i.e. interrupts?

I think that is an implementation detail, probably. We use interrupts in a mostly transparent way for the various I2C controller implementations now.

kenbell, 2022-11-20:
@deadprogram - biggest thing I see with this API above, is that I2C is transaction-oriented.  The controller could terminate one transaction and start another - but the peripheral would not know, so could be doing 'writes' as a response the wrong transaction.  I think it needs some way to synchronize the target logic to the wire state.

In the API above, the controller could be trying to read 1 register or 256 registers, but the target logic wouldn't know which (no i2c target does until the ack appears on the wire), so the target would have to write all 256 to the wire in case?

There's also the possibility of i2c devices that are 'register triggered'.

(I'm thinking here about how to emulate a BME280, for example, where in single-capture mode the sensor is triggered by register access over i2c)

I think the API above could work great for bespoke comms over i2c where both parties are implemented in Tinygo, I'm pretty sure it could be implemented on top of the 'lower-level' API I've proposed in #3301.

aykevl, 2022-12-08:
Agree with @kenbell here, a byte oriented protocol doesn't describe I2C very well. I2C is not like UART.
Properly designing this would require looking into how peripheral mode is implemented in various chips (like nrf, avr, sam, etc) and finding the commonalities to design an API around. These are not unimportant implementation details. I suspect (without looking at the datasheets) that we'll need a protocol where we supply the I2C peripheral with a buffer to use (the size is the maximum possible transaction size) and use DMA to receive the message where available. Then, once it is received, we call a callback with the buffer and can send the response inside this callback.

deadprogram, 2023-08-18:
Since "target" mode was introduced a couple of releases ago, now closing this issue. Thank you everyone!

closed 2023-08-18 by deadprogram