tinygo-org/tinygo #1906: <undefined reference> problem while compiling a binary for Arm from Linux

cheer-Yuan, 2021-05-26:
Hello,

I am trying to compile a program with tinygo from Linux, when I enter the following command : 
```
GOARCH=arm tinygo build -o <OutputName> ./main.go
```
I got :
```
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: /tmp/tinygo859602615/main.o: in function `syscall.Seek':
/usr/local/go/src/syscall/syscall_linux_arm.go:60: undefined reference to `syscall.seek'
```
The relevant codes are :
```
// Underlying system call writes to newoffset via pointer.
// Implemented in assembly to avoid allocation.
func seek(fd int, offset int64, whence int) (newoffset int64, err Errno)

func Seek(fd int, offset int64, whence int) (newoffset int64, err error) {
	newoffset, errno := seek(fd, offset, whence)
	if errno != 0 {
		return 0, errno
	}
	return newoffset, nil
}
```
I have just use
```
sudo apt-get install arm-linux-gnueabihf
```
to install this package.

However,
```
GOARCH=arm go build -o <OutputName> ./main.go
```
works without any problem.
Using CGO_ENABLED="0" flag doesn't change anything.
I have searched on the Internet and found no similar issues. Thank you for any possible information.

Environment variables :
```
CGO_ENABLED="1"
go version go1.16.3 linux/amd64
GOROOT="/usr/local/go"
TINYGOROOT="/usr/local/lib/tinygo"
```

dkegel-fastly, 2022-01-03:
Also seems to affect i386.  See log
https://github.com/tinygo-org/tinygo/runs/4687146838?check_suite_focus=true
from https://github.com/tinygo-org/tinygo/pull/2457

aykevl, 2022-01-03:
I have a local branch to implement Go assembly. It's not complete (only works on amd64 and arm64 now) but once finished that branch should fix this issue.

dkegel-fastly, 2022-03-22:
May be fixed already by https://github.com/tinygo-org/tinygo/pull/2571 in 0.22?

Please try 0.22 or dev and see if that works better.

aykevl, 2022-08-28:
This is fixed now, see #2571.

closed 2022-08-28 by aykevl