tinygo-org/tinygo #1910: windows issue: unabe to flash arduino nano33

loreaman, 2021-05-28:
``` 
tinygo flash -target arduino-nano33 .\step0

bossac: extra arguments found
Try 'bossac -h' or 'bossac --help' for more information
error: failed to flash C:\Users\Lorenzo Amante\AppData\Local\Temp\tinygo006274015\main.bin: exit status 1
```

sago35, 2021-05-29:
If you add the `-x` option as follows, how does it appear?

```
$ tinygo flash -target arduino-nano33 -x .\step0
```

loreaman, 2021-06-01:
Hi, thank for the reply. Unfortunately I am getting again the error. I will paste the output here:
```
C:\Users\Lorenzo Amante\Documents\PROJECTS\TINYGO\gophercon-2019\sensor\arduino>tinygo flash -target arduino-nano33 -x .\step0
ld.lld --emit-relocs --gc-sections -L C:\Users\Lorenzo Amante\scoop\apps\tinygo\current -T targets/atsamd21.ld -o C:\Users\Lorenzo Amante\AppData\Local\Temp\tinygo270044323\main C:\Users\Lorenzo Amante\AppData\Local\Temp\tinygo270044323\main.o C:\Users\Lorenzo Amante\scoop\apps\tinygo\current\pkg\armv6m-none-eabi\compiler-rt.a C:\Users\Lorenzo Amante\scoop\apps\tinygo\current\pkg\armv6m-none-eabi\picolibc.a C:\Users\Lorenzo Amante\AppData\Local\tinygo\obj-9ae6912157365e41ef4601bc6ddfde4fb334865ec0b3445a142a6c0f.o C:\Users\Lorenzo Amante\AppData\Local\tinygo\obj-7d1f71c6e0b8e3e91c6e0bc9d1b3b8b6257cb2b7ccddcb662876d57a.o C:\Users\Lorenzo Amante\AppData\Local\tinygo\obj-789b0940e03ceb173eba2c87acf8891d566caff16c9621f0ebdf6957.o C:\Users\Lorenzo Amante\AppData\Local\tinygo\obj-029fbf7e06de2115289d368f10f043505bb949b6f059b5f5b18312c9.o
bossac -i -e -w -v -R -U --port=COM12 --offset=0x2000 C:\Users\Lorenzo Amante\AppData\Local\Temp\tinygo270044323\main.bin
bossac: extra arguments found
Try 'bossac -h' or 'bossac --help' for more information
error: failed to flash C:\Users\Lorenzo Amante\AppData\Local\Temp\tinygo270044323\main.bin: exit status 1
```

deadprogram, 2021-06-01:
What does `bossac.exe -h` by itself show?

loreaman, 2021-06-01:
`
C:\Users\Lorenzo Amante>bossac.exe -h
Usage: bossac.exe [OPTION...] [FILE]
Basic Open Source SAM-BA Application (BOSSA) Version 1.9.1
Flash programmer for Atmel SAM devices.
Copyright (c) 2011-2018 ShumaTech (http://www.shumatech.com)

Examples:
  bossac -e -w -v -b image.bin   # Erase flash, write flash with image.bin,
                                 # verify the write, and set boot from flash
  bossac -r0x10000 image.bin     # Read 64KB from flash and store in image.bin

Options:
  -e, --erase           erase the entire flash starting at the offset
  -w, --write           write FILE to the flash; accelerated when
                        combined with erase option
  -r, --read[=SIZE]     read SIZE from flash and store in FILE;
                        read entire flash if SIZE not specified
  -v, --verify          verify FILE matches flash contents
  -o, --offset=OFFSET   start erase/write/read/verify operation at flash OFFSET;
                        OFFSET must be aligned to a flash page boundary
  -p, --port=PORT       use serial PORT to communicate to device;
                        default behavior is to use first serial port
  -b, --boot[=BOOL]     boot from ROM if BOOL is 0;
                        boot from FLASH if BOOL is 1 [default];
                        option is ignored on unsupported devices
  -c, --bod[=BOOL]      no brownout detection if BOOL is 0;
                        brownout detection is on if BOOL is 1 [default]
  -t, --bor[=BOOL]      no brownout reset if BOOL is 0;
                        brownout reset is on if BOOL is 1 [default]
  -l, --lock[=REGION]   lock the flash REGION as a comma-separated list;
                        lock all if not given [default]
  -u, --unlock[=REGION] unlock the flash REGION as a comma-separated list;
                        unlock all if not given [default]
  -s, --security        set the flash security flag
  -i, --info            display device information
  -d, --debug           print debug messages
  -h, --help            display this help text
  -U, --usb-port[=BOOL] force serial port detection to USB if BOOL is 1 [default]
                        or to RS-232 if BOOL is 0
  -R, --reset           reset CPU (if supported)
  -a, --arduino-erase   erase and reset via Arduino 1200 baud hack

Report bugs to <bugs@shumatech.com>`

aykevl, 2021-06-01:
I suspect the issue is improper escaping of the command, because the path contains a space.

sago35, 2021-06-01:
As a workaround, I think you can set TMP to write as follows

```
$ mkdir C:\tinygo\tmp
$ set TMP=C:\tinygo\tmp
$ tinygo flash -target arduino-nano33 -x .\step0
```

aykevl, 2021-06-02:
This PR should fix it: https://github.com/tinygo-org/tinygo/pull/1927

closed 2021-07-01 by deadprogram