tinygo-org/tinygo #1918: Can NEOPIXEL(S) name be unified?

sago35, 2021-05-30:
`NEOPIXEL(S)` is now defined with the following name
[drivers/examples/ws2812](https://github.com/tinygo-org/drivers/blob/release/examples/ws2812/others.go#L9) expects the name `NEOPIXELS`.
So we would like to unify these names.
Even if there is only one `NEOPIXEL`, we want to use the name `NEOPIXELS`.

If you want to ignore compatibility, you can change them all to `NEOPIXELS`.
If you want to keep compatibility, you can set an additional name `NEOPIXELS`.

If anyone has a better idea, please let me know.

```
$ git grep NEOPIXEL
src/machine/board_circuitplay_bluefruit.go:40:  NEOPIXELS = D8
src/machine/board_circuitplay_express.go:43:    NEOPIXELS = D8
src/machine/board_clue_alpha.go:74:     NEOPIXEL = D18
src/machine/board_feather-m4-can.go:46: NEOPIXELS = D8
src/machine/board_feather-nrf52840.go:61:       NEOPIXEL = D8
src/machine/board_feather-stm32f405.go:81:      NUM_BOARD_NEOPIXEL = 1
src/machine/board_feather-stm32f405.go:84:      LED_NEOPIXEL = D8
src/machine/board_grandcentral-m4.go:135:       NEOPIXEL_PIN    = D88 // (PC24)
src/machine/board_grandcentral-m4.go:141:       NEOPIXEL = NEOPIXEL_PIN
src/machine/board_matrixportal-m4.go:77:        NEOPIXEL = D4
src/machine/board_pybadge.go:42:        NEOPIXELS = D8
src/machine/board_pygamer.go:44:        NEOPIXELS = D8
src/machine/board_pyportal.go:70:       NEOPIXEL = D2
src/machine/board_wioterminal.go:34:    PIN_NEOPIXEL = PA15
```

deadprogram, 2021-05-30:
> If you want to ignore compatibility, you can change them all to NEOPIXELS

I vote for this one. It is a one time small pain, but does not leave any extra cruft around. 

Maybe someone else has a different opinion? Do please let us know.

aykevl, 2021-06-01:
Yes, using the same name everywhere seems like a good idea.

However, I wonder whether we should maybe use something other than NeoPixel, which is an AdaFruit brand name for the WS2812. We could just call it `WS2812` everywhere and perhaps add an alias `NEOPIXEL` or `NEOPIXELS` on boards from AdaFruit.

sago35, 2021-06-06:
If the board has more than one WS2812, just add the name WS2812 to the board.

sago35, 2021-06-07:
I will change the following targets.

circuitplay-bluefruit
circuitplay-express
clue-alpha
clue
feather-m4-can
feather-m4
feather-nrf52840
feather-stm32f405
grand-central-m4
matrixportal-m4
metro-m4-airlift
pybadge
pygamer
pyportal
qtpy

The wioterminal defines the name PIN_NEOPIXEL, but ws2812 is not equipped with it.
PIN_NEOPIXEL exists because it implements the name of the arduino source.
https://github.com/Seeed-Studio/ArduinoCore-samd/blob/master/variants/wio_terminal/variant.h#L86

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

closed 2021-07-01 by deadprogram