tinygo-org/tinygo #1939: Recent STM32WL's TIM structs changes.

ofauchon, 2021-06-07:
Hi, 

As I was trying to finish my STM32WL port, I found out that STM32WL's TIM Struct differs from older STM32: 

STM32WL TIM Structs: 

```

// Low-power timer
type LPTIM_Type struct {
	ISR       volatile.Register32 // 0x0
	ICR       volatile.Register32 // 0x4
	IER       volatile.Register32 // 0x8
	CFGR      volatile.Register32 // 0xC
	CR        volatile.Register32 // 0x10
	CMP       volatile.Register32 // 0x14
	ARR       volatile.Register32 // 0x18
	CNT       volatile.Register32 // 0x1C
	LPTIM1_OR volatile.Register32 // 0x20
	_         [4]byte
	RCR       volatile.Register32 // 0x28
}

// General-purpose timers
type TIM_Type struct {
	CR1          volatile.Register32 // 0x0
	CR2          volatile.Register32 // 0x4
	_            [4]byte
	DIER         volatile.Register32 // 0xC
	SR           volatile.Register32 // 0x10
	EGR          volatile.Register32 // 0x14
	CCMR1_Output volatile.Register32 // 0x18
	_            [4]byte
	CCER         volatile.Register32 // 0x20
	CNT          volatile.Register32 // 0x24
	PSC          volatile.Register32 // 0x28
	ARR          volatile.Register32 // 0x2C
	RCR          volatile.Register32 // 0x30
	CCR1         volatile.Register32 // 0x34
	_            [12]byte
	BDTR         volatile.Register32 // 0x44
	DCR          volatile.Register32 // 0x48
	DMAR         volatile.Register32 // 0x4C
	TIM16_OR1    volatile.Register32 // 0x50
	_            [12]byte
	TIM16_AF1    volatile.Register32 // 0x60
	_            [4]byte
	TIM16_TISEL  volatile.Register32 // 0x68
}

// Advanced-control timers
type AdavanceTIM_Type struct {
	CR1                    volatile.Register32 // 0x0
	CR2                    volatile.Register32 // 0x4
	SMCR                   volatile.Register32 // 0x8
	DIER                   volatile.Register32 // 0xC
	SR                     volatile.Register32 // 0x10
	EGR                    volatile.Register32 // 0x14
	CCMR1_Output           volatile.Register32 // 0x18
	CCMR2_Output           volatile.Register32 // 0x1C
	CCER                   volatile.Register32 // 0x20
	CNT                    volatile.Register32 // 0x24
	PSC                    volatile.Register32 // 0x28
	ARR                    volatile.Register32 // 0x2C
	RCR                    volatile.Register32 // 0x30
	CCR1                   volatile.Register32 // 0x34
	CCR2                   volatile.Register32 // 0x38
	CCR3                   volatile.Register32 // 0x3C
	CCR4                   volatile.Register32 // 0x40
	BDTR                   volatile.Register32 // 0x44
	DCR                    volatile.Register32 // 0x48
	DMAR                   volatile.Register32 // 0x4C
	OR1                    volatile.Register32 // 0x50
	CCMR3OutputComparemode volatile.Register32 // 0x54
	CCR5                   volatile.Register32 // 0x58
	CCR6                   volatile.Register32 // 0x5C
	AF1                    volatile.Register32 // 0x60
	AF2                    volatile.Register32 // 0x64
	TISEL                  volatile.Register32 // 0x68
}
```

  * STM32F4

```
// Advanced-timers
type TIM_Type struct {
	CR1          volatile.Register32 // 0x0
	CR2          volatile.Register32 // 0x4
	SMCR         volatile.Register32 // 0x8
	DIER         volatile.Register32 // 0xC
	SR           volatile.Register32 // 0x10
	EGR          volatile.Register32 // 0x14
	CCMR1_Output volatile.Register32 // 0x18
	CCMR2_Output volatile.Register32 // 0x1C
	CCER         volatile.Register32 // 0x20
	CNT          volatile.Register32 // 0x24
	PSC          volatile.Register32 // 0x28
	ARR          volatile.Register32 // 0x2C
	RCR          volatile.Register32 // 0x30
	CCR1         volatile.Register32 // 0x34
	CCR2         volatile.Register32 // 0x38
	CCR3         volatile.Register32 // 0x3C
	CCR4         volatile.Register32 // 0x40
	BDTR         volatile.Register32 // 0x44
	DCR          volatile.Register32 // 0x48
	DMAR         volatile.Register32 // 0x4C
}
```

This cause many errors when trying to use STM32WL's timers with current machine_stm32_tim implementation. 

@kenbell : Did you encounter this problem when you worked on the STM32 TIM and PWM implementation ?
How would you solve that ? 
A solution would be to 'fork' machine_stm32_tim.go timer code (one for 'old' STM32's timers, and one for 'newest' STM32) 

Thanks 

Olivier

kenbell, 2021-06-08:
This might need some PRs against the stm32-rs project

For other chips, I think `TIM_Type` and `AdvanceTIM_Type` typically get merged into a single type `TIM_Type` - it looks like `AdavanceTIM_Type` is a superset of `TIM_Type`, so this is probably the first thing to investigate.  It might be some SVD patches would make sense to the `stm32-rs` project to make it like the other ST chips.   (there's a typo in the name of `AdavanceTIM_Type` so it would be good to at least get that patched).

If that doesn't happen, an unsafe cast might fix it, since the underlying data-structures look compatible?

For the low-power timers, I've ignored them on the other boards.  To support them, they look different enough they're probably worth treating like another class of peripheral (that also does PWM, with it's own implementation).

kenbell, 2021-06-08:
You can try changing the `stm32wle5.yaml` file to have this content:
```yaml
_svd: ../svd/stm32wle5.svd

# Remove the bogus _CM4 from the name
# and put all timers (except low-power timers) in
# the TIM group, consistent with other chips.
_modify:
  name: STM32WLE5
  TIM*:
    groupName: TIM
```

You'll then need to do a `make` in `lib/stm32-svd/` and then do `make gen-device-stm32`.  There's a bit more of a write-up here: https://github.com/tinygo-org/tinygo/wiki/Adding-support-for-an-STM32-MCU

ofauchon, 2021-06-09:
Thanks @kenbell

ofauchon, 2021-08-20:
Hi @kenbell,

Still stuck with this issue. 
No success with patching stm32-rs or device/stm32/stm32wle5.go file. 

Could you please make some quick tests on your side ?
I'm affraid stm32wle5's timers differs a lot from other stm32.

Thanks

Olivier

kenbell, 2021-08-21:
Yeah - I'll take a look.  I'm working on some improvements to the tool that processes SVD files to help improve some of this, but it's slow going.

kenbell, 2021-11-09:
Sorry - i've been busy for the longest time :(     I've just raised a PR to move all the timers into the TIM group in stm32-rs.

https://github.com/stm32-rs/stm32-rs/pull/657

ofauchon, 2021-11-09:
Hi @kenbell,

Thank you very much for your help on this issue.

Olivier

kenbell, 2021-11-10:
The change has been merged in stm32-rs, this PR will update the repo of fully-patched SVD files: https://github.com/tinygo-org/stm32-svd/pull/4

kenbell, 2021-11-11:
@ofauchon - I've borrowed a bunch of your code and raised a PR, so i think we're in good shape: https://github.com/tinygo-org/tinygo/pull/2248

If we go with my PR, I cut out a bunch of stuff i couldn't verify like SPI (most of my kit is in storage).  The PLL stuff wasn't working - but not sure if my fault, so it's just running at the default 4Mhz from reset.  Could you raise a PR to add in all the stuff I've cut out.

If you'd sooner raise a PR from your branch with complete support and merge some of my changes (mostly adding more timers), please do so and I'll cancel my PR.

ofauchon, 2021-11-11:
Thanks @kenbell , let's go with your PR, you helped so much !

I'll PR over your code to add extra features !

Thx

Olivier

ofauchon, 2021-11-21:
Solved by latest commits in dev branch.

deadprogram, 2021-11-22:
Reopening until this code is part of next release.

closed 2022-01-26 by deadprogram