tinygo-org/tinygo #1943: wasm cgo problem

rtttech, 2021-06-08:
I used cgo in my code as following, compile success but wasm load failed: "Failed to load Wasm module due to a missing import: env.my_time"

/*
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <time.h>

int64_t my_time() {
return time(NULL);
}
*/
import "C"

...

fsyyft, 2021-06-09:
I also encountered the same problem, hope to get help!

aykevl, 2021-06-09:
Yes, it's not currently supported to define functions in the header. What you can do instead is move the code (in this case the function `my_time`) into a separate C file and declare it above `import "C"` instead.

rtttech, 2021-06-09:
How to compile the .go file with the added c file?

aykevl, 2021-06-09:
You can put them in the same package. The C file will be compiled together with the Go files in the package. Please see https://golang.org/doc/code for how code is normally organized in Go.

closed 2021-06-11 by rtttech