tinygo-org/tinygo #1949: panic: runtime error: goroutine stack overflow on pca10056-s140v7

andrewrynhard, 2021-06-13:
I get this error with the following:

```
import (
	"fmt"
	"machine"
	"math/rand"
	"time"

	"tinygo.org/x/bluetooth"
)

var heartRate uint8 = 75
var manufacturer = "Test"
var localName = "Test"
var modelNumber = "1"

func main() {
	fmt.Printf("%s %s\r\n", localName, modelNumber)

	heartRateMeasurement := setup()

	go start_button_loop()
	start_hr_loop(heartRateMeasurement)
}

func setup() *bluetooth.Characteristic {
	adapter := bluetooth.DefaultAdapter

	if err := adapter.Enable(); err != nil {
		panic("failed to enable BLE")
	}

	fmt.Printf("Enabled BLE\r\n")

	adv := adapter.DefaultAdvertisement()

	options := bluetooth.AdvertisementOptions{
		LocalName:    localName,
		ServiceUUIDs: []bluetooth.UUID{bluetooth.ServiceUUIDHeartRate},
	}

	if err := adv.Configure(options); err != nil {
		panic("failed to configure heart rate advertisment")
	}

	adapter.SetConnectHandler(func(addresser bluetooth.Addresser, connected bool) {
		if addresser != nil {
			fmt.Printf("%s %t\r\n", addresser.String(), connected)
		}
	})

	if err := adv.Start(); err != nil {
		panic("failed to start advertisement")
	}

	fmt.Printf("Started advertisement\r\n")

	var heartRateMeasurement bluetooth.Characteristic

	service := &bluetooth.Service{
		UUID: bluetooth.ServiceUUIDHeartRate,
		Characteristics: []bluetooth.CharacteristicConfig{
			{
				UUID:  bluetooth.CharacteristicUUIDManufacturerNameString,
				Value: []byte(manufacturer),
				Flags: bluetooth.CharacteristicNotifyPermission | bluetooth.CharacteristicReadPermission,
			},
			{
				UUID:  bluetooth.CharacteristicUUIDModelNumberString,
				Value: []byte(modelNumber),
				Flags: bluetooth.CharacteristicNotifyPermission | bluetooth.CharacteristicReadPermission,
			},
			{
				Handle: &heartRateMeasurement,
				UUID:   bluetooth.CharacteristicUUIDHeartRateMeasurement,
				Flags:  bluetooth.CharacteristicNotifyPermission | bluetooth.CharacteristicReadPermission,
			},
		},
	}

	if err := adapter.AddService(service); err != nil {
		panic("faild to create BLE service")
	}

	fmt.Printf("Added service\r\n")

	return &heartRateMeasurement
}

func start_hr_loop(hr *bluetooth.Characteristic) {
	led1 := machine.LED1
	led1.Configure(machine.PinConfig{Mode: machine.PinOutput})

	nextBeat := time.Now()

	for {
		led1.High()

		// Mock data.
		nextBeat = nextBeat.Add(time.Minute / time.Duration(heartRate))
		time.Sleep(nextBeat.Sub(time.Now()))
		heartRate = randomInt(65, 85)

		hr.Write([]byte{0, heartRate})

		led1.Low()
		time.Sleep(10 * time.Millisecond)
	}
}

func start_button_loop() {
	led2 := machine.LED2
	led2.Configure(machine.PinConfig{Mode: machine.PinOutput})

	button1 := machine.BUTTON1
	button1.Configure(machine.PinConfig{Mode: machine.PinInputPullup})

	for {
		if button1.Get() {
			led2.Low()
		} else {
			led2.High()
		}

		time.Sleep(100 * time.Millisecond)
	}
}

func randomInt(min, max int) uint8 {
	return uint8(min + rand.Intn(max-min))
}
```
Changing the `scheduler` to `coroutines` seems to fix it.

oflebbe, 2021-06-17:
Maybe you can try to add (add a comma to the previous line to get syntax correct)
```
"default-stack-size": 4096
```
to `targets/pca10059-s140v7.json`

Maybe the automatic guessing of the stack doesn't fit here.

aykevl, 2021-06-18:
I've seen this a number of times now, I think it makes sense to increase the default stack size. See this PR: https://github.com/tinygo-org/tinygo/pull/1960

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

andrewrynhard, 2021-07-01:
I still seem to be hitting this with 0.19.0.

oflebbe, 2021-07-02:
I don't have a pca10056 but tried it on a pca10049-s132v6 

I can repro the problem even on this board...

Starts normal (until Added service) and than the blinking stops and panic appears.
May need up to 30sec.
Installed softdevice as recommended in bluetooth repo
command `tinygo flash --build pca10040-s132v6`

```
Enabled BLE                                                                     
Started advertisement                                                           
Added service                                                                   
panic: runtime error: goroutine stack overflow   
```                               

Not sure how to proceed , tried gdb on it, it looked like the stack overflow happend in the goroutine start_button_loop at time.Sleep()

aykevl, 2021-07-02:
> Not sure how to proceed , tried gdb on it, it looked like the stack overflow happend in the goroutine start_button_loop at time.Sleep()

This would be important to know for sure. You can break on `runtime.runtimePanic` and then get a backtrace to know it. Compiling with `-opt=1` can make it easier to debug.
Also, what's the output of compiling with `-print-stacks`? It prints the calculated stack sizes. For `start_button_loop`, I would expect it to know exactly how big the stack should get (although there can of course be bugs).

oflebbe, 2021-07-03:
Have a lot of datapoints, which might be not at all relevant to you, but I wasn't sure about memory layout of tinygo. I only replicating in cause you need it as well ...

But first let me tell that the stack overflow will happen at some random point in time, sometimes after a few seconds or even after 10 minutes (!) 

Ok, breakpoint on `runtime.runtimePanic` : Panic happens in Sleep (actually Pause) 

```
Breakpoint 1, runtime.runtimePanic (msg=...) at /Users/olaf/Devel/oss/tinygo/src/runtime/panic.go:19
19		println(msg)
(gdb) where
#0  runtime.runtimePanic (msg=...) at /Users/olaf/Devel/oss/tinygo/src/runtime/panic.go:19
#1  0x00026d06 in internal/task.Pause () at /Users/olaf/Devel/oss/tinygo/src/internal/task/task_stack.go:43
#2  0x0002d8ee in main.start_button_loop () at /Users/olaf/tmp/main.go:125
#3  0x0002d550 in <goroutine wrapper> () at /Users/olaf/tmp/main.go:22
#4  0x0002e352 in tinygo_startTask () at /Users/olaf/Devel/oss/tinygo/src/internal/task/task_stack_cortexm.S:26
```

Stacksizes
```
olaf@silversun tmp % /usr/local/tinygo/bin/tinygo build -o p.hex -print-stacks --target pca10040-s132v6 main.go 
function                         stack usage (in bytes)
Reset_Handler                    324
main.start_button_loop           108
runtime.run$1                    recursive, (*fmt.pp).printArg may call itself
##
```

Looking at the code something else seem to have run into the stack of the goroutine , if I understand correctly.
(Checking canaryptr) unfortunately currentTask is optimized out even for opt=1 :(

Registers at panic look like this:
```
(gdb) info reg
r0             0x2e8d9             190681
r1             0x1                 1
r2             0x16                22
r3             0x4000211c          1073750300
r4             0x18                24
r5             0x2e781             190337
r6             0x40000             262144
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x200070b8          0x200070b8
lr             0x16                22
pc             0x26d9e             0x26d9e <runtime.runtimePanic+20>
xPSR           0x61000000          1627389952
fpscr          0x0                 0
msp            0x2000496c          0x2000496c
psp            0x200070b8          0x200070b8
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x2                 2
```

Not too familiar with ARM assembler but SP seems to be stackpointer at 0x200070b8

According to the layout 
```
tinygo build -o a.elf --target pca10040-s132v6 main.go
arm-none-eabi-nm a.elf | sort -n
...
200039c0 A __app_ram_base
200049c0 B _stack_top
200049c0 D _globals_start
200049c0 D _sdata
...
0006428 B _ebss
20006428 B _globals_end
20006428 B _heap_start
20010000 A _heap_end
```
goroutine stack is located in heap -- makes sense since it is allocated with make()

I checked if the crashes are related to garbage collections, but it seems that GC are not correlated to the stack issue.
(I printed logs to detect when an GC happend)

I checked the `__app_ram_base` with values from the nRF SDK for this softdevice , we are reserving a way larger amount of ram for the softdevice, so we should be save . 

What I am unsure if we have an unexpecteed interaction of the softdevice code with tinygo runtime.

deadprogram, 2021-07-03:
Perhaps this is related to https://github.com/tinygo-org/bluetooth/issues/63?

aykevl, 2021-07-04:
> Perhaps this is related to [tinygo-org/bluetooth#63](https://github.com/tinygo-org/bluetooth/issues/63)?

Probably not, otherwise you would probably have gotten a message telling there is a stack overflow.

> But first let me tell that the stack overflow will happen at some random point in time, sometimes after a few seconds or even after 10 minutes (!)

This strongly suggests it's an interrupt that is interfering. It probably only happens when the interrupt happens just at the wrong moment.

> Looking at the code something else seem to have run into the stack of the goroutine , if I understand correctly.

Yes. It's a safety check. Once it is triggered, a stack overflow has happened sometime in the past. It's a valuable check because memory corruption (caused by a stack overflow) is incredibly hard to debug.

Thank you for the debug messages!

oflebbe, 2021-07-04:
I have an additional -- but very weak -- data point . I retried the test with tinygo-0.17 . I was not able to repro ... (after tinygo clean) But maybe I only did not wait long enough.

oflebbe, 2021-07-04:
> > Perhaps this is related to [tinygo-org/bluetooth#63](https://github.com/tinygo-org/bluetooth/issues/63)?
> 
> Probably not, otherwise you would probably have gotten a message telling there is a stack overflow.
> 
But we get the message there is a stack overflow  (shown in my first message on this thread)

aykevl, 2021-07-04:
I think I know what's going on, now I need to figure out a way to properly fix it. In essence, the stack of new goroutines is misaligned and because of the STKALIGN flag (set by default) it will be realigned on interrupt entry. This realignment causes a one-word stack overflow.

If you want a quick and dirty fix, here it is:

```diff
diff --git a/src/internal/task/task_stack.go b/src/internal/task/task_stack.go
index a703d10a..758690a6 100644
--- a/src/internal/task/task_stack.go
+++ b/src/internal/task/task_stack.go
@@ -61,7 +61,8 @@ func (t *Task) Resume() {
 // initialize the state and prepare to call the specified function with the specified argument bundle.
 func (s *state) initialize(fn uintptr, args unsafe.Pointer, stackSize uintptr) {
        // Create a stack.
-       stack := make([]uintptr, stackSize/unsafe.Sizeof(uintptr(0)))
+       stack := make([]uintptr, stackSize/unsafe.Sizeof(uintptr(0))+1)
 
        // Set up the stack canary, a random number that should be checked when
        // switching from the task back to the scheduler. The stack canary pointer
```

I'll now think about how to properly fix this.

> But we get the message there is a stack overflow (shown in my first message on this thread)

Not in tinygo-org/bluetooth#63. That is probably an unrelated issue.

oflebbe, 2021-07-04:
Looking at the assembler of internal/task.Pause with breakpoint at the canary check.

```
(gdb) x/10 0x00026cd8
   0x26cd8 <internal/task.Pause>:	push	{r4, lr}
   0x26cda <internal/task.Pause+2>:	movw	r4, #25216	; 0x6280
   0x26cde <internal/task.Pause+6>:	movt	r4, #8192	; 0x2000
   0x26ce2 <internal/task.Pause+10>:	ldr	r0, [r4, #0]
   0x26ce4 <internal/task.Pause+12>:	cbz	r0, 0x26d14 <internal/task.Pause+60>
   0x26ce6 <internal/task.Pause+14>:	ldr	r1, [r0, #20]
=> 0x26ce8 <internal/task.Pause+16>:	cbz	r1, 0x26d14 <internal/task.Pause+60>
   0x26cea <internal/task.Pause+18>:	ldr	r1, [r1, #0]
   0x26cec <internal/task.Pause+20>:	movw	r2, #62837	; 0xf575
   0x26cf0 <internal/task.Pause+24>:	movt	r2, #47163	; 0xb83b
(gdb) 
```

right at 0x26ce8 register  r1 should be the canaryPtr, right ? 
```
(gdb) info reg
...
r1             0x20007058          536899672
...
```

The Canary is there
```
(gdb) x/1x 0x20007058
0x20007058:	0xb83bf575
```

I was happy to observe a panic shortly after

(gdb) c
Continuing.

Breakpoint 1, runtime.runtimePanic (msg=...) at /Users/olaf/Devel/oss/tinygo/src/runtime/panic.go:19
19		println(msg)

(gdb) x/1x 0x20007058
0x20007058:	0x7d784000

So not only a word has been overwritten, the whole Canary is dead.  Looks like even more around that memory has changed , btw.

aykevl, 2021-07-05:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/1984
While the fix itself is not very difficult, it took me a while to convince myself this is indeed the correct fix.

> So not only a word has been overwritten, the whole Canary is dead. Looks like even more around that memory has changed , btw.

I'm not sure what you mean? As far as I can see, only the canary will be overwritten, nothing else. And with this fix, I no longer see a stack overflow in the example from @andrewrynhard while I did see one before.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram