tinygo-org/tinygo #1954: WASM: lib WASI update

hamidrezakks, 2021-06-16:
Hi,
Thanks for the great project,
github.com/WebAssembly/wasi-libc has been updated recently, can you please update tinygo wasm to the latest wasi-lib?

Bests,
KK

aykevl, 2021-06-17:
Is there anything that we need from the new libc version? Better features or bug fixes relevant to TinyGo?

I'm not necessarily against updating the library, but I think it needs a bit more justification than that it is the newest version.

hamidrezakks, 2021-06-19:
They have updated the MUSL

**MUSL release note:**
1.2.2 release notes

major changes:
- child restrictions lifted after fork of multithreaded parent

new features:
- _Fork function (POSIX-future)
- reallocarray function (extension from OpenBSD, now widespread)
- gettid function (kernel tid as supported concept)
- SIGEV_THREAD_ID sigevent API (Linux extension)
- tcgetwinsize and tcsetwinsize functions (POSIX-future)

performance:
- faster software sqrt on archs without native sqrt instruction

compatibility:
- realpath no longer depends on procfs availability & accuracy
- time zone parser now always prefers 64-bit tables if present
- crypt_blowfish now supports $2b$ prefix
- res_query now reports errors via h_errno
- set*id and setrlimit are now safe in vforked/cloned child
- setgroups now applies to all threads
- dlopen debugger notification is improved, should work with lldb
- setrlimit no longer needs __synccall broadcast on linux 2.6.36+
- faccessat with AT_EACCESS no longer needs child process on linux 5.8+

bugs fixed:
- buffer overflow and infinite loop errors in wcsnrtombs (CVE-2020-28928)
- sem_close unmapped still-referenced semaphores
- fork of process with active aio could deadlock or crash paren
- pthread_cond_wait was broken with priority-inheritance mutex
- getgrouplist wrongly failed when nscd reported an empty list
- abort could leak modified SIGABRT disposition to fork or posix_spawn child
- regression with mallocng: malloc_usable_size(0) crashed
- readlink wrongly gave EINVAL on zero length dest buffer
- sqrtl was severely inaccurate (not correctly rounded) on ldquad archs
- assert failure wrongly flushed stdio (possible deadlock)
- MUSL_LOCPATH search was broken with multiple components
- missing newline in herror output
- possible deadlock in pthread_exit with pshared mutex or barrier usage
- pthread_mutexattr_getprotocol didn't read back protocol
- v4l2 ioctl translation for pre-time64 kernels didn't work

arch-specific bugs fixed:
- x86_64 longjmp failed to handle 0 argument reliably
- i386 __set_thread_area fallback for pre-2.6 kernels didn't work
- missing O_LARGEFILE macro value on x86_64, x32, mips64
- unpredictable s390x breakage from failure to preserve call-saved registers

dgryski, 2021-11-18:
wasi-libc has been updated as part of https://github.com/tinygo-org/tinygo/pull/2224; closing this.

closed 2021-11-18 by dgryski