tinygo-org/tinygo #1955: Why does adding port operations improve performance?

sago35, 2021-06-17:
In my efforts to improve the performance of ili9341, I found a strange behavior.

The following two branches have very different speeds.
The x02 branch is nearly 10 percent faster, despite having one more port operation.

```
tinygo flash --target feather-m4-can --size short --programmer cmsis-dap --scheduler tasks ./examples/ili9341/pyportal_boing/
```

* with `dbg5.Low()` : 96 fps
    * https://github.com/tinygo-org/drivers/tree/x02
    * https://github.com/tinygo-org/drivers/blob/x02/examples/ili9341/pyportal_boing/main.go#L170

[image]

* without `dbg5.Low()` : 85 fps
    * https://github.com/tinygo-org/drivers/tree/x03
    * https://github.com/tinygo-org/drivers/blob/x03/examples/ili9341/pyportal_boing/main.go#L170

[image]

The diff of *.ll is as shown in the screenshot below.
I've compared them with the debug information removed.

```
tinygo build -o xxx.ll --target feather-m4-can --size short --scheduler tasks ./examples/ili9341/pyportal_boing/
```

[image]

Is there any way to find out what is causing this?
Is it an alignment issue?
Or is there some other reason?

sago35, 2021-06-17:
x02's log:
```
width, height == 240 320
128  fps
102  fps
96  fps
102  fps
98  fps
96  fps
94  fps
97  fps
96  fps
94  fps
93  fps
96  fps
```

x03's log:
```
width, height == 240 320
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
85  fps
```

aykevl, 2021-06-17:
You could disassemble the binary. One IR instruction could lead to a big difference in the generated assembly.

Also, if you are optimizing for performance you should use `-opt=2` (which is like `-O2` in GCC), the default is to optimize for code size. It may make a difference.

sago35, 2021-06-26:
When I measured again, I got the following results.
I used the latest version of tinygo compiler.

* x02 (with dbg5.Low())
    * w/o opt : 96 fps
    * opt=2 : 108 fps
* x03 (without dbg5.Low())
    * w/o opt : 104 fps
    * opt=2 : 103 fps

```
$ tinygo version
tinygo version 0.19.0-dev-0e267dd2 windows/amd64 (using go version go1.16 and LLVM version 11.0.0)
```

deadprogram, 2023-05-23:
I think this was actually solved by https://github.com/tinygo-org/tinygo/pull/3492

Should we close it on next release?

closed 2023-06-14 by deadprogram