tinygo-org/tinygo #1957: size of int on darwin inconsistent

oflebbe, 2021-06-17:
Hi, I was wondering about a test failure on my mac (testdata/filesystem.go) failed with "panic: should be non exist error"

Digging deeper I found that comparison of the fd filedescriptor in Open https://github.com/tinygo-org/tinygo/blob/dev/src/syscall/syscall_libc.go#L47 does not fail because the "int" datatype in go (and tinygo) is a 64 bit datatype , rather in the "C" binding is a 32bit data type. 

For example in go on GOARCH=amd64 GOOS=darwin
`package main

import "unsafe"
import "C"

func main() {
	var a int = 1
	println(unsafe.Sizeof(a))
	var b C.int = 2
	println(unsafe.Sizeof(b))

}`

outputs 
`4
8`

Looks like the tinygo promise to have an easy interop between C and tinygo might not work on this platform:
https://tinygo.org/docs/reference/lang-support/

I am not too sure how to proceed from here:
* Make tinygo int 32bit on darwin ?
* Supply a different implementation for darwin  libc bindings, explicit done as int32
* introduce a C.int datatype like in go for interop 

Did I miss something along the way ?

oflebbe, 2021-06-17:
seems like @aykevl  recommended to use int32 in #1669 . I didn't recognized this accepting the code now in tinygo.

aykevl, 2021-06-17:
Yes, `int32` would be the correct choice (because practically all modern OSes nowadays uses 32-bit int). Despite the name, C int and Go int are two different types and you shouldn't assume they are the same size.

aykevl, 2021-06-17:
Here is a patch: https://github.com/tinygo-org/tinygo/pull/1958
Can you test whether it fixes the bug on your system?

oflebbe, 2021-06-18:
Alright! the tinygo tests are now running flawless on macos 11.4 with xcode 12.5 .

deadprogram, 2021-07-01:
This has been released with v0.19.0 so now closing. Thanks everyone!

closed 2021-07-01 by deadprogram