tinygo-org/tinygo #1961: todo: go on interface call

fgsch, 2021-06-18:
This was discussed on slack a while back, but for reference, the following code fails with `todo: go on interface call`:

```go
package main

import (
	"fmt"
	"net/http"
)

func main() {
	fmt.Println(http.Cookie{})
}
```

Tinygo version:

`tinygo version 0.19.0-dev-1913cb76 darwin/amd64 (using go version go1.16.5 and LLVM version 11.0.0)`

aykevl, 2021-06-21:
I looked into this before and I believe my conclusion was that it's easier to finish refactoring the reflect package first.

dgryski, 2021-11-18:
Support for "go on interfaces" as been merged and tinygo 0.21 was released with this fix.  Closing this bug.  Please re-open if you're still encountering problems.

closed 2021-11-18 by dgryski