tinygo-org/tinygo #1963: Error building for wasi target

fgsch, 2021-06-21:
Also discussed recently on slack, but for the record, the code below fails to build with:

```
# internal/syscall/unix
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:44:52: Stat_t not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at_sysnum_fstatat64_linux.go:11:37: SYS_FSTATAT64 not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at_sysnum_linux.go:9:38: SYS_UNLINKAT not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at_sysnum_linux.go:10:36: SYS_OPENAT not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/nonblocking.go:13:36: SYS_FCNTL not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:16:20: BytePtrFromString not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:21:25: Syscall not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:31:20: BytePtrFromString not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:36:26: Syscall6 not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:46:20: BytePtrFromString not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/at.go:51:25: Syscall6 not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/copy_file_range_linux.go:13:26: Syscall6 not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/fcntl_linux_32bit.go:15:25: SYS_FCNTL64 not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/getrandom_linux.go:35:26: Syscall not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/nonblocking.go:16:25: Syscall not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/nonblocking.go:16:76: F_GETFL not declared by package syscall
/usr/local/Cellar/go/1.16.5/libexec/src/internal/syscall/unix/nonblocking.go:20:22: O_NONBLOCK not declared by package syscall
```

```go
package main

import (
	"fmt"
	"net/http"
)

func main() {
	fmt.Println(http.Cookie{})
}
```

TinyGo version:

`tinygo version 0.19.0-dev-e107efa6 darwin/amd64 (using go version go1.16.5 and LLVM version 11.0.0)`

Invocation:

`tinygo build -target wasi -o main.wasm main.go`

fgsch, 2021-06-21:
The `internal/syscall/unix` error should be addressed by https://github.com/tinygo-org/tinygo/pull/1962.

fgsch, 2021-06-21:
This is also related to #1961 .

fgsch, 2021-06-21:
Applying #1962 and ignoring #1961 now fails with:

```
# internal/cpu
/usr/local/Cellar/go/1.16.5/libexec/src/internal/cpu/cpu_wasm.go:7:7: CacheLinePadSize redeclared in this block
/usr/local/Cellar/go/1.16.5/libexec/src/internal/cpu/cpu_arm.go:7:7: 	other declaration of CacheLinePadSize
/usr/local/Cellar/go/1.16.5/libexec/src/internal/cpu/cpu_wasm.go:9:6: doinit redeclared in this block
/usr/local/Cellar/go/1.16.5/libexec/src/internal/cpu/cpu_arm.go:21:6: 	other declaration of doinit
```

Depending on how things progress, I will eventually move this out to its own issue, but adding it here so it's documented before I forget.

fgsch, 2021-06-21:
The above should be fixed by #1964.

fgsch, 2021-06-21:
I can confirm that after applying #1964 and commenting out the offending line to workaround #1961, this compiles, and a slightly modified version of this (one that doesn't depend on reflection) runs successfully.

fgsch, 2021-06-22:
@deadprogram since this only applies to dev we can probably close it.

deadprogram, 2021-06-22:
Ah, ok. Thanks for letting me know. Now closing.

closed 2021-06-22 by deadprogram