tinygo-org/tinygo #1966: [DEV] memory corruption?

fgsch, 2021-06-21:
While testing #1964 with the code further down, I got:

```
021/06/21 23:32:49 net/http: invalid Cookie.Domain "<lots of garbage>"; dropping domain attribute
"foo="
```

The same code compiled with `-gc leaking` or `-scheduler none` works as expected (just prints `"foo="`).

Code:

```go
package main

import (
        "fmt"
        "net/http"
)

func main() {
        c := http.Cookie{Name:"foo"}
        fmt.Printf("%q\n", c.String())
}
```

Command line:

`tinygo build -target wasi -o test.wasm test.go`

fgsch, 2021-06-22:
I can also repro this issue with 0.18.0.

fgsch, 2021-06-22:
Reduced repro:

```go
package main

import (
	"fmt"
	"log"
	"strings"
)

type fubar struct {
	foo string
	bar string
}

func (f *fubar) String() string {
	var b strings.Builder
	b.Grow(1000)
	log.Printf("Below should be empty")
	b.WriteString(f.foo)
	return b.String()
}

func main() {
	c := fubar{}
	fmt.Printf("%q\n", c.String())
}
```

fgsch, 2021-07-01:
Fails with 0.17.0 and 0.18.0 as well.

aykevl, 2021-07-10:
I have further narrowed down this bug to this:

```go
package main

import (
	"os"
	"time"
)

var std = &Logger{out: os.Stderr}

type foo struct {
	a int
	b int
	c int
}

type Writer interface {
	Write([]byte) (int, error)
}

type Logger struct {
	out Writer
}

func main() {
	f(&foo{})
}

func f(f *foo) {
	println("f:", f.a, f.b, f.c)
	time.Time{}.Location().String()
	std.out.Write(nil)
	println("f:", f.a, f.b, f.c)
}
```

It prints the following:

```
$ tinygo run -target=wasi .
f: 0 0 0
f: 0 2 0
```

But clearly it should print the following:

```
$ go run .
f: 0 0 0
f: 0 0 0
```

If you want to help, it would be great if you could break this issue down even further. My strategy so far is inlining functions from packages, seeing whether the copy still reproduces the issue, and then incrementally removing pieces that don't matter while keeping the memory corruption.

fgsch, 2021-07-11:
I've fiddled with it a bit more and it looks like locking the mutex, and thus interacting with the scheduler, is related. 
Updated code below.

```
package main

import (
	"os"
	"sync"
)

var std = &Logger{out: os.Stderr}

type foo struct {
	a int
	b int
	c int
}

type Writer interface {
	Write([]byte) (int, error)
}

type Logger struct {
	out Writer
}

func main() {
	f(&foo{})
}

func f(f *foo) {
	var m sync.Mutex
	println("f:", f.a, f.b, f.c)
	m.Lock()
	std.out.Write(nil)
	println("f:", f.a, f.b, f.c)
}
```

 To be continued.

fgsch, 2021-07-11:
This might be the same as in #1981.

aykevl, 2021-07-11:
Maybe, although my first guess is that they're unrelated. WebAssembly does some things very differently from other instruction sets (such as amd64) and it requires some ugly workarounds in TinyGo that I suspect cause this bug.

aykevl, 2021-07-14:
You were right after all, https://github.com/tinygo-org/tinygo/pull/1999 also fixes this issue.

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

closed 2021-09-21 by deadprogram