tinygo-org/tinygo #1972: Extreme high memory usage when building wasm

zluo01, 2021-06-28:
When try to build my code into a wasm with tinygo which command `tinygo build -o main.wasm -target wasm ./main.go` . It yields memory usage over 100GB and long build time (eventually kill due to insufficient memory). [source](https://github.com/zluo01/color-thief/blob/master/wasm/main.go) code here.

[image]

zluo01, 2021-07-01:
try with the latest 0.19 release, yet the problem still exists.  I find out this [file](https://github.com/zluo01/color-thief/blob/master/wsm/wsm.go) cause the problem. Yet the only external thing I use is `math.Sqrt`. I wonder what can be the reason that cause tinygo to fail the build ? Thank you.

Inkeliz, 2021-09-13:
I have the same issue compiling the Gio app (https://todo.sr.ht/~eliasnaur/gio/151). It eats all the memory and then leave an "LLVM ERROR: out of memory". :\

gedw99, 2021-09-18:
@Inkeliz Does this help ? https://github.com/tinygo-org/tinygo/issues/2108#issuecomment-918700845

It reservees the memory and i think will unblock you.. Let me know

Inkeliz, 2021-09-18:
I didn't try because I can't compile, and it doesn't seem to be related.

The #2108 is about to limit the RAM on the runtime. It says, "I build this file by the command `tinygo build -o a.wasm -target=wasi a.go`", so there's no issue with the compilation. The question is how to limit the memory during the runtime. 

However, the issue that @zluo01 and I are facing is on the compilation, not the runtime. We can't compile at all. The compiler (tinygo) uses all the memory available and then crashes with out-of-memory, even with 256GB of RAM. 

When we run `tinygo build`, it starts to eat all the memory, and after hours it crashes.

gedw99, 2021-09-21:
Does it work with the gio kitchen example ?

inkeliz, 2023-04-14:
I suspect that this bug is from  `wasm-opt` rather than TinyGo. 

If you still have the code, can you compile with GoCompiler (`GOARCH=wasm GOOS=js go build ./main.go` and then use `wasm-opt -O3 main.wasm -o main.wasm` or `wasm-opt -Oz main.wasm -o main.wasm`?

In my case, `wasm-opt` uses all memory and then crash. The same exact code which `TinyGo` can't compile (and TinyGo uses wasm-opt). **I'm not sure if that is reproducible on all cases.**

closed 2023-04-14 by aykevl